openwrt/openwrt #14258: IEI PUZZLE-M902 does not support 5Gbps out of the box

julmarqu, 2023-12-21:
### Describe the bug

The IEI PUZZLE-M902 router has three mGig ports that should auto-negotiate to 0.1/1/2.5/5/10 Gbps but the 5Gbps capability is not being advertised out of the box for those ports as can be seen via the `ethtool` command output:

```
 Supported link modes:   100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                10000baseKR/Full 
                                2500baseT/Full
```

I have an ATT modem that has a 5Gig port (supports up to 5gbps link) and it auto-negotiates to 2.5Gbps with the m902. This is a problem because a 5 Gbps service would not be supported.

I am able to advertise the capability manually using ethtool, as can be seen in the output below:

```
root@quetzalcoatl:~# ethtool eth6
Settings for eth6:
        Supported ports: [ ]
        Supported link modes:   100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                10000baseKR/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                10000baseKR/Full 
                                2500baseT/Full 
                                5000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Half 100baseT/Full 
                                             1000baseT/Half 1000baseT/Full 
                                             10000baseT/Full 
                                             2500baseT/Full 
                                             5000baseT/Full 
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 10000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 2
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: yes
```

The port will establish the link at 5 Gbps if I do this, but I have strange behavior where download speeds work just fine, but uploads are capped at around 1Gbps (tested with Ookla servers). If I pass the link through an intermediate switch that negotiates the 5 Gbps with the modem and 10 Gbps with the m902, I get the full bandwidth both for upstream and downstream. I also don't have problems if I have the m902 auto-negotiate at 2.5 Gbps with the modem, but again this means 5 Gbps cannot be reached.

I was looking at the driver code and it is missing the 5 Gbps profile, so not sure why it was skipped since the router should support it according to official documentation (at least in the QNAP version of the software) https://www.qnap.com/en-us/product/qhora-322/specs/hardware

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mvebu/cortexa72

### Device

iEi Puzzle-M902

### Image kind

Official downloaded image

### Steps to reproduce

Plug in a device that should auto-negotiate to 5 Gbps to one of the 10G mgig ports (6, 7, 8).

### Actual behaviour

Link auto-negotiates at 2.5 Gbps, the next-best available speed.

### Expected behaviour

Link should auto-negotiate to 5Gbps

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-24:
-KR type should not be supported, it is for PCB-like connectors, not rj45 or optical SFP sockets.
i.e try advertise map without -KX.
Aquantia AQR113 description says 5gbps yes, 10kr no.
Likely somebody wrote over wrong bit while phy programming

julmarqu, 2024-01-15:
Just for kickers I tried the advertise map without the KX setting. Didn't work, which makes sense to me since even though it is available, it is not what is being negotiated so would not be the profile used. This is how it looks
```
Settings for eth6:
        Supported ports: [ ]
        Supported link modes:   100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                10000baseKR/Full 
                                2500baseT/Full 
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  100baseT/Half 100baseT/Full 
                                1000baseT/Full 
                                10000baseT/Full 
                                2500baseT/Full 
                                5000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Half 100baseT/Full 
                                             1000baseT/Half 1000baseT/Full 
                                             2500baseT/Full 
                                             5000baseT/Full 
        Link partner advertised pause frame use: Symmetric
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 5000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 2
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Link detected: yes
```
While on 5000 Mb/s speed, uploads speeds are below 1Gbps. When I re-introduce the intermediate media converter to negotiate the link at 10G, I have full upload bandwidth again.

brada4, 2024-01-15:
Understand your frustation, driver itself just reads mii register without conversion. 
Can you check
```
mii-tool -v eth0 eth1 wan lan2 | grep product
  product info: vendor 00:00:00, model 0 rev 0
```
What seems that particular phy advertises 5gbps mode using wrong bit in mask.
What other cable end reports when you get 1gbps ipo 5gbps?
If ports are different types try other just in case.

julmarqu, 2024-01-21:
Unfortunately there is only one multi-gig port on the modem (the others are 1G only). This is how it looks on the modem end. It looks the same both with the media converter in between, and when connected directly to the m902
```
Connection Speed 	5000Mbps fullduplex

  	Port 1 	Port 2 	Port 3 	Port 4
State 	up 	down 	down 	down
Transmit Speed 	5000000000 	0 	0 	0
Transmit Packets 	789906913 	0 	0 	0
Transmit Bytes 	1998148143 	0 	0 	0
Transmit Unicast 	784671938 	0 	0 	0
Transmit Multicast 	514373 	0 	0 	0
Transmit Dropped 	65 	0 	0 	0
Transmit Errors 	0 	0 	0 	0
Receive Packets 	2291900977 	0 	0 	0
Receive Bytes 	992641609 	0 	0 	0
Receive Unicast 	2289704063 	0 	0 	0
Receive Multicast 	2193341 	0 	0 	0
Receive Dropped 	134917 	0 	0 	0
Receive Errors 	0 	0 	0 	0
```
It doesn't matter which 10G port I use on the m902, all three have the same behavior.

This is the non-invalid output of the above command:
```
product info: vendor 0c:70:c7, model 17 rev 2
```

brada4, 2024-01-21:
Obviously creds of 'broken' port are needed.