openwrt/openwrt #14262: Kernel 6.6.x LTS instead of 6.1 LTS?

Neustradamus, 2023-12-23:
### Describe the bug

Dear @OpenWrt team,

I think it is time to switch 6.1.x LTS to 6.6.x LTS in master.

Thanks in advance.

### OpenWrt version

snapshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

*

### Device

*

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

closed 2023-12-23 by ynezz

oliv3r, 2024-02-23:
Just for passers by dropping this one: http://lists.openwrt.org/pipermail/openwrt-devel/2024-February/042204.html

Neustradamus, 2024-02-23:
@oliv3r: Thanks for your important comment!

@hauke: Thanks to have looked my ticket and to plan with the latest LTS Kernel: 6.6.x.

And please like LibreOffice, plan RC versions with the real official month release date.
LibreOffice 24.2 has been released the last day of January: 2024-01-31:
- https://blog.documentfoundation.org/blog/2024/01/31/libreoffice-24-2/