openwrt/openwrt #14271: Build openwrt for xiaomi miwifi mini error when compile batman adv 2003.3

InBomBi, 2023-12-24:
### Describe the bug

Hi,i got these error when trying to build openwrt for xiaomi miwifi mini device - mt7620 chip
I enable mesh wifi with batman adv 2003.3 in config, then when compile, i have got error as bellow, pls suggest me to fix them.
Thanks
`make[3]: Entering directory '/workdir/openwrt/feeds/routing/batman-adv'
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/.built_check
make  -C "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.144" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Dec 23 07:52:05 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/symvers/mac80211.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/symvers/mt76.symvers" CC="mipsel-openwrt-linux-musl-gcc" KERNELRELEASE=5.15.144 M="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=n CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/lib/gcc/mipsel-openwrt-linux-musl/12.3.0/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/net/batman-adv -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2023.3-openwrt-2\\\"" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.144'
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/net/batman-adv/bat_algo.o
In file included from <command-line>:
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/compat-hacks.h:38:35: error: redefinition of 'skb_vlan_eth_hdr'
   38 | static inline struct vlan_ethhdr *skb_vlan_eth_hdr(const struct sk_buff *skb)
      |                                   ^~~~~~~~~~~~~~~~
In file included from /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/compat-hacks.h:33:
./include/linux/if_vlan.h:66:35: note: previous definition of 'skb_vlan_eth_hdr' with type 'struct vlan_ethhdr *(const struct sk_buff *)'
   66 | static inline struct vlan_ethhdr *skb_vlan_eth_hdr(const struct sk_buff *skb)
      |                                   ^~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:289: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/net/batman-adv/bat_algo.o] Error 1
make[4]: *** [Makefile:1908: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/net/batman-adv] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.15.144'
make[3]: *** [Makefile:104: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/batman-adv-2023.3/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#0.77#0.29#0.97
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:128: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:122: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.`

### OpenWrt version

25.05 shanpshot

### OpenWrt release

snapshot

### OpenWrt target/subtarget

ramips/mt7620

### Device

xiaomi miwifi mini

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

InBomBi, 2023-12-24:
> github

yes, i build from clean repository

brada4, 2023-12-24:
it was fixed yesterday https://github.com/openwrt/routing/commit/96b5fa139c59baeb163b7b1d85bfe992c5b21133
You can close this issue as it duplicates https://github.com/openwrt/routing/issues/1036 in proper repo.

InBomBi, 2023-12-25:
> it was fixed yesterday https://github.com/openwrt/routing/commit/96b5fa139c59baeb163b7b1d85bfe992c5b21133
> 
> You can close this issue as it duplicates https://github.com/openwrt/routing/issues/1036 in proper repo.

Thank you many much

closed 2023-12-25 by InBomBi