openwrt/openwrt #14279: How to have a night LED mode (timer) for router's led to turn off and on all leds at particular times daily on openwrt firmware?

blue-pine, 2023-12-25:
### Describe the bug

I tried myself and also searched everywhere for setting night led mode activated for OpenWrt firmware but all effort went to no avail for such a simple feature. This is already avilable on the stock firmware of my wifi router TP-Link Archer C6 AC1200 v3.2 which is itself based on OpenWrt and for which I have provide all related files from my research here in my repository : https://github.com/blue-pine/openwrt-tplink_archer_c6_v3.2 . Still I have failed to join the led_night_mode library (from stock router firmware) to the OpenWrt firmware for use on the router. So, I ask for everyone's help whatever they can provide for getting this last and only thing lacking on all OpenWrt firmwares. If you need any help then please ask me, you are always welcome.
Thank you

### OpenWrt version

all versions

### OpenWrt release

all releases

### OpenWrt target/subtarget

all targets

### Device

TP-Link Archer C6 AC1200 v3.2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-25:
start with 'uci export' and flipp trigger to off in all 'lonfig led' sections. Then trigger activity via main crontab.
not a bug like to change code. Likely someone smart on forums can teach you script tricks

blue-pine, 2023-12-26:
> start with 'uci export' and flipp trigger to off in all 'lonfig led' sections. Then trigger activity via main crontab. not a bug like to change code. Likely someone smart on forums can teach you script tricks

Thank you, but I am a beginner and dont know how to go forward, so how can I integrate this timer with openwrt ?

brada4, 2023-12-26:
Yes, just present your idea in forums, it is not defect per se, it is a quite reasonable wishlist item for new or changing existing wifischedule (similar in the way that it shuts down then restores radios on schedules) package

blue-pine, 2023-12-26:
> Yes, just present your idea in forums, it is not defect per se, it is a quite reasonable wishlist item for new or changing existing wifischedule (similar in the way that it shuts down then restores radios on schedules) package

Yes, grateful for your help
Thank you

raenye, 2023-12-26:
`wifischedule` package does something similar for WLANs.
Maybe it is time for a more comprehensive "event scheduler" OpenWrt package.

blue-pine, 2023-12-26:
> `wifischedule` package does something similar for WLANs. Maybe it is time for a more comprehensive "event scheduler" OpenWrt package.

Yes, but wifischeduler sets timer only for the wifi radio not for led indicators. So features like led night timer highlight the limitation and complications of ths package with respect to use of openwrt on wifi routers

raenye, 2023-12-26:
I'm not saying it does what you want. But (a) it has similar functionality if you want to learn about using the cron scheduler; and (b) someone could extend it to a general scheduler framework.

blue-pine, 2023-12-27:
> I'm not saying it does what you want. But (a) it has similar functionality if you want to learn about using the cron scheduler; and (b) someone could extend it to a general scheduler framework.

Right. Thanks for your explanation. I appreciate it