openwrt/openwrt #14283: ARV752DPW Lantiq XWay Major Version Upgrade to 23 bricks device

Yemtex, 2023-12-26:
### Describe the bug

Hi,

after upgrading from version 22.03.6 to 23.05 my Arcadyan ARV752DPW (Easybox 802) was bricked.
https://openwrt.org/toh/arcadyan/arv752dpw

I connected a serial cable to it.
When booting the new OpenWrt Image (23.05.*), I get the following output. The devices just hangs and does nothing else, like it froze. No serial input accepted. Needs full device restart.
```
U-Boot 2010.03 (Apr 09 2012 - 01:10:47)

Board: ARV752DPW
SoC: Danube/Twinpass/Vinax-VE V1.5, DDR Speed 166 MHz, CPU Speed 333 MHz
DRAM:  64 MB
Flash:  8 MB
*** Warning - bad CRC, using default environment

Net:   
searching for rtl8306 switch ... found
Reset Hard
Done
Reset Soft
Success

lq_cpe_eth
Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at b0020000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.137
   Created:      2023-11-14  13:38:11 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2726841 Bytes =  2.6 MB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.137 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 14 13:38:11 2023
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Arcor 802
[    0.000000] Initrd not found or empty - disabling initrd
```

Can you please help me, getting the new firmware to run.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

lantiq/xway

### Device

Arcor 802

### Image kind

Official downloaded image

### Steps to reproduce

1. Copy image via tftpboot to RAM
2. Erase flash memory
3. Copy image from RAM to flash storage
4. Reset u-boot
5. Let OpenWrt start

### Actual behaviour

1. Device starts normally.
2. Bootloader start OpenWrt Image
3. Device freezes

### Expected behaviour

OpenWrt boots up.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-26:
Please get back to v22, issue is already documented: 
https://openwrt.org/toh/arcadyan/arv752dpw

Yemtex, 2023-12-26:
I did that. I updated the documentation for other people, so they don't have to struggle.

action-server, 2023-12-27:
I have a spare [Arcadyan ARV7506PW11](https://openwrt.org/toh/arcadyan/arv7506pw11) that doesn't work with openwrt version 23 and up.
Serial console with u-boot shows following message:

> *** Warning - bad CRC, using default environment
> ...
> ...
> ...
> Starting kernel ...
> 
> [    0.000000] Linux version 5.15.145 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24698-b799dd3c70) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Dec 27 15:01:30 2023
> [    0.000000] SoC: Danube rev 1.5
> [    0.000000] printk: bootconsole [early0] enabled
> [    0.000000] CPU0 revision is: xxxxxxx (MIPS 24KEc)
> [    0.000000] MIPS: machine is Alice/O2 IAD 4421
> [    0.000000] Initrd not found or empty - disabling initrd

Will be using lower versions in the meantime

brada4, 2023-12-27:
check in v22 if 4x64k growth in v23 depletes rootfs_data to under 320k meaning likely first boot config cannot be written. Though should not die thst early.

Yemtex, 2023-12-27:
Just for your info. I also have a https://openwrt.org/toh/t-com/speedport_w_504v with the same architecture.

Yemtex, 2023-12-27:
> check in v22 if 4x64k growth in v23 depletes rootfs_data to under 320k meaning likely first boot config cannot be written. Though should not die thst early.

Im sorry, but could you explain what you mean with 4x64k?
What should I do?

brada4, 2023-12-28:
cat /proc/mtd
i believe flash eraseblocks are 64k in size, so check 256k firmware growth still leaves 320k free for jffs overlay.  I.e. if jffs /overlay is at least 576k in size in v22.

Yemtex, 2023-12-28:
okey.

```
root@EasyBox802:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00010000 00002000 "uboot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 007d0000 00010000 "firmware"
mtd3: 002633e1 00010000 "kernel"
mtd4: 0056cc1f 00010000 "rootfs"
mtd5: 00110000 00010000 "rootfs_data"
mtd6: 00010000 00010000 "board_config"
root@EasyBox802:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.5M      4.5M         0 100% /rom
tmpfs                    27.7M    236.0K     27.4M   1% /tmp
/dev/mtdblock5            1.1M    252.0K    836.0K  23% /overlay
overlayfs:/overlay        1.1M    252.0K    836.0K  23% /
tmpfs                   512.0K         0    512.0K   0% /dev
root@EasyBox802:~#
```

I want to learn more about this. What documentation could help me understand more about what you are talking?

brada4, 2023-12-28:
linux-mtd project etc...
64k erase block size., and in this regard  enough space for upgrade.

Yemtex, 2023-12-28:
Okey, nice to know.
So how can I proceed to debug this and maybe give this issue enough info?

Yemtex, 2023-12-28:
Related issues: #13934 and #13891

brada4, 2023-12-28:
seems about right, neither case overfills flash space leading to freeze or no conf persistance later.

Yemtex, 2023-12-28:
What could I try next, do you have and ideas?
Thanks in advance.

brada4, 2023-12-28:
should be smp and ram detection right after

abajk, 2023-12-30:
Reverting[ this commit](https://github.com/openwrt/openwrt/commit/084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26) fixes tem problem?

zuzia-dev, 2024-01-01:
> Reverting [smp patch](https://github.com/openwrt/openwrt/commit/084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26) fixes this issue?

Yes. The image uploads properly.
```
ROM VER: 1.0.3
CFG 01
Read EEPROMX
 X

U-Boot 2010.03 (May 11 2012 - 02:54:15)

Board: ARV4518PW
SoC: Danube/Twinpass/Vinax-VE V1.5, DDR Speed 166 MHz, CPU Speed 333 MHz
DRAM:  64 MB
Flash:  8 MB
*** Warning - bad CRC, using default environment

Net:   
searching for rtl8306 switch ... found
lq_cpe_eth
Hit any key to stop autoboot:  0 
## Booting kernel from Legacy Image at b0020000 ...
   Image Name:   MIPS OpenWrt Linux-5.15.139
   Created:      2023-12-28  13:27:49 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2586472 Bytes =  2.5 MB
   Load Address: 80002000
   Entry Point:  80002000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.15.139 (EASYBOX@zuzia-dev) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23674-60e49cf170) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Thu Dec 28 13:27:49 2023
[    0.000000] SoC: Danube rev 1.5
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019641 (MIPS 24KEc)
[    0.000000] MIPS: machine is Arcor 802
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 16kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 16kB, 4-way, VIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyLTQ0,115200
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=0006bf50
[    0.000000] Readback ErrCtl register=0006bf50
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55148K/65536K available (6324K kernel code, 604K rwdata, 1356K rodata, 1172K init, 213K bss, 10388K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] CPU Clock: 333MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 11467562657 ns
[    0.000004] sched_clock: 32 bits at 166MHz, resolution 6ns, wraps every 12884901885ns
[    0.008134] Calibrating delay loop... 221.18 BogoMIPS (lpj=1105920)
[    0.094350] pid_max: default: 32768 minimum: 301
[    0.101848] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.109127] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.140367] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.150175] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.157382] pinctrl core: initialized pinctrl subsystem
[    0.166927] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.200072] pinctrl-xway 1e100b10.pinmux: Init done
[    0.214976] dma-xway 1e104100.dma: Init done - hw rev: 3, ports: 5, channels: 20
[... 143 lines cut ...]
[  223.998129] br-lan: port 1(eth0.1) entered forwarding state
[  224.914916] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  231.054194] done.
[  231.054734] jffs2: notice: (1542) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[  231.900997] overlayfs: upper fs does not support tmpfile.
```
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23674-60e49cf170'
DISTRIB_TARGET='lantiq/xway'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23674-60e49cf170'
DISTRIB_TAINTS='no-all busybox'
```
```
dev:    size   erasesize  name
mtd0: 00010000 00002000 "uboot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 007d0000 00010000 "firmware"
mtd3: 002777a8 00010000 "kernel"
mtd4: 00558858 00010000 "rootfs"
mtd5: 00130000 00010000 "rootfs_data"
mtd6: 00010000 00010000 "board_config"
```

action-server, 2024-01-01:
> Reverting [smp patch](https://github.com/openwrt/openwrt/commit/084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26) fixes this issue?

I cant also confirm reverting that patch fixed the issue for me.

Will someone merge it into snapshot?

action-server, 2024-01-01:
Or rather backport it to stable, cause it bricks aforementioned devices.

abajk, 2024-01-04:
Will prepare Patch.  From what I remember, this change also caused performance degradation on ARX100.

commit 64d8c277 (2024-01-16): Revert "lantiq: xway: kernel: enable SMP support"

Enabling SMP on the xway target results in two issues:

* some danube chipset-based devices fail on boot,
* on devices based on the arx100 chipset, enabling smp
results in a degradation of NAT performance.

After these two issues are fixed, SMP can be re-enabled.

This reverts commit 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26.

Fixes: #13934
Fixes: #14283
commit 65fb97b4 (2024-01-22): lantiq: fix boot isues on danube boards

Enabling SMP on Danube[1] is incompatible with a patch that
adds support for interrupt handling on all cores on other
platforms[2]. This patch fixes the mentioned issue.

1. 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26 ("lantiq: xway: kernel: enable SMP support ")
2. fbd33d61648ae8982fbada7ad3b6d8222b367ab5 ("lantiq: enable interrupts on second VPEs")

Fixes: #13934
Fixes: #14283
PR #14388 (2024-01-22): lantiq: fix boot isues on danube boards
Enabling SMP on Danube[1] is incompatible with a patch that adds support for interrupt handling on all cores on other  platforms[2]. This patch fixes the mentioned issue.

1. 084c20f6c54180f2dc6b3efc1c5ba9b62afbaf26 ("lantiq: xway: kernel: enable SMP support ")
2. fbd33d61648ae8982fbada7ad3b6d8222b367ab5 ("lantiq: enable interrupts on second VPEs")

Fixes: #13934
Fixes: #14283
closed 2024-01-22 by openwrt-bot