openwrt/openwrt #14286: [dropbear]: default RecvWindowSize limits performance

sudden6, 2023-12-27:
### Describe the bug

The default receive window size of 24576 bytes limits the file transfer performance to an unnecessarily slow speed. My test when using rsync over the internet, with the router as target of an rsync transfer results in a limit of ~800KBytes/s. This limit is not caused by the CPU or storage, since setting `        option RecvWindowSize '524288' resulted in 2.5MBytes/s (connection maximum).

I propose to increase the default value for `RecvWindowSize` for all targets with enough RAM, so that file transfers are faster.
I don't know what a good default would be, I selected the new value for my device randomly, since it has lots of RAM.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

Use rsync and ssh to transfer files to a device running OpenWrt. Target is an external USB HDD with ext4 attached to the device.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-27:
yes, 2nd rsync transfers less data. that is by design.

if you dont like upstream defaults you need to petition them.

brada4, 2023-12-27:
sftp would use 32k, rsync 128k blocks over ssh serial abstraction. try something like 135168, more will not gain.

sudden6, 2023-12-28:
> yes, 2nd rsync transfers less data. that is by design.

I don't understand what you mean by that, can you rephrase?
 
> if you dont like upstream defaults you need to petition them.

I'm not sure if it makes sense to petition a default value for all dropbear users, since they might have more resource constrained devices in mind. I think it makes sense though to ship a `/etc/config/dropbear` config file with an increased limit for OpenWrt, since routers nowadays have plenty of RAM.

> sftp would use 32k, rsync 128k blocks over ssh serial abstraction. try something like 135168, more will not gain.

Good info, I need to test that.

brada4, 2023-12-28:
it seems more like add a line to the wiki page. Dropbear is per arch not per device package, like one can let aarch64 and x86 loose but for the rest still wiki is needed.

brada4, 2023-12-30:
Fixed for future releases by @blocktrron
https://github.com/openwrt/openwrt/commit/f95eecfb21ff08662e022accd30e8254028ff63b
You can set config value in the meantime to 262144 to match future default.

sudden6, 2024-01-01:
Thank you @blocktrron and @brada4 I think with this it's closed now.

closed 2024-01-01 by sudden6