openwrt/openwrt #14294: WiFi Crashing

john08burke, 2023-12-28:
### Describe the bug

I am using a pretty minimalist config, but am experiencing issues with the WiFi after 10-30mins of traffic. All devices are disconnected, as the WiFi is no longer being broadcast.  This affects all radios it seems.  I can connect to the router and internet traffic over WAN is working as expected. The issues is temporarily fixed by rebooting the router.

The dmesg logs are cluttered with snippets like this:
```
[ 1506.186742] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[ 1506.190475] ath11k c000000.wifi: failed to send pdev bss chan info request
[ 1506.198088] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
[ 1506.204686] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 0
[ 1506.211207] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
[ 1506.217688] ath11k c000000.wifi: failed to submit beacon template command: -108
[ 1506.223864] ath11k c000000.wifi: failed to update bcn template: -108
[ 1506.231070] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
[ 1506.237750] ath11k c000000.wifi: failed to set BA BUFFER SIZE 256 for vdev: 0
[ 1506.244368] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
```

And possibly a crash:
```
[  583.085894] ------------[ cut here ]------------
[  583.093470] WARNING: CPU: 1 PID: 1812 at sta_set_sinfo+0xca4/0xd54 [mac80211]
[  583.098248] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath11k_ahb ath11k pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 cfg80211 spi_gpio spi_bitbang slhc qrtr_smd qrtr qmi_helpers nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c gpio_74x164 compat sha512_generic seqiv jitterentropy_rng drbg michael_mic hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug ext4 mbcache jbd2 aquantia hwmon crc_ccitt crc32c_generic
[  583.150925] CPU: 1 PID: 1812 Comm: hostapd Not tainted 6.1.69 #0
[  583.173160] Hardware name: Netgear RAX120v2 (DT)
[  583.179407] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  583.184012] pc : sta_set_sinfo+0xca4/0xd54 [mac80211]
[  583.190693] lr : sta_set_sinfo+0xbe4/0xd54 [mac80211]
[  583.195899] sp : ffffffc00be5b870
[  583.200931] x29: ffffffc00be5b870 x28: ffffff8002faec00 x27: ffffffc00be5bdc8
[  583.204237] x26: ffffff80020ad880 x25: ffffffc008cca400 x24: ffffff8003d40940
[  583.211356] x23: ffffff8004459ab8 x22: ffffff8003d40940 x21: ffffff80063388a0
[  583.218474] x20: 0000000000000001 x19: ffffff8004459000 x18: 00000000000002d3
[  583.225591] x17: 31333a31633a3761 x16: 3a6363203a6e6f69 x15: ffffffc008bc7328
[  583.232710] x14: 0000000000000879 x13: 00000000000002d3 x12: 00000000ffffffea
[  583.239828] x11: 00000000ffffefff x10: ffffffc008c1f328 x9 : ffffffc008bc72d0
[  583.246947] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : 0000000000000001
[  583.254064] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffff8002faec00
[  583.261182] x2 : 0000000000000000 x1 : ffffff8002faec00 x0 : 00000000ffffff94
[  583.268300] Call trace:
[  583.275409]  sta_set_sinfo+0xca4/0xd54 [mac80211]
[  583.277671]  sta_info_destroy_addr_bss+0x54/0x80 [mac80211]
[  583.282533]  ieee80211_color_change_finish+0x1c68/0x1f80 [mac80211]
[  583.287917]  cfg80211_check_station_change+0x1268/0x3530 [cfg80211]
[  583.294166]  genl_family_rcv_msg_doit+0xb8/0x11c
[  583.300414]  genl_rcv_msg+0x108/0x230
[  583.305272]  netlink_rcv_skb+0x5c/0x12c
[  583.308832]  genl_rcv+0x38/0x50
[  583.312476]  netlink_unicast+0x1e8/0x2d4
[  583.315602]  netlink_sendmsg+0x1a0/0x3d0
[  583.319770]  ____sys_sendmsg+0x1c8/0x270
[  583.323676]  ___sys_sendmsg+0x7c/0xc0
[  583.327582]  __sys_sendmsg+0x48/0xb0
[  583.331141]  __arm64_sys_sendmsg+0x24/0x30
[  583.334787]  invoke_syscall.constprop.0+0x5c/0x104
[  583.338695]  do_el0_svc+0x58/0x17c
[  583.343466]  el0_svc+0x18/0x54
[  583.346850]  el0t_64_sync_handler+0xf4/0x120
[  583.349892]  el0t_64_sync+0x174/0x178
[  583.354318] ---[ end trace 0000000000000000 ]---
[  583.357980] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
[  583.362574] ath11k c000000.wifi: failed to delete peer vdev_id 0 addr cc:a7:c1:31:06:8f ret -108
[  583.368585] ath11k c000000.wifi: Failed to delete peer: cc:a7:c1:31:06:8f for VDEV: 0
[  583.377607] ath11k c000000.wifi: Found peer entry 00:03:7f:12:88:bf n vdev 0 after it was supposedly removed
```

Please let me know if you need further information, thanks!

### OpenWrt version

r24697-485adc9d3c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Netgear RAX120v2

### Image kind

Official downloaded image

### Steps to reproduce

Crash is a bit sporadic, after a few minutes the WiFi crashes.  It can be reproduced by restarting the router and waiting a few minutes.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

john08burke, 2023-12-28:
Here is a complete dump from dmesg for the WiFI crash.

[dmesg_openwrt.log](https://github.com/openwrt/openwrt/files/13787729/dmesg_openwrt.log)

Here is the interesting part:
```
[ 2533.930575] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
[ 2533.930575] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1
[ 2533.930575] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
[ 2533.930575]
[ 2533.930575] NOC_error.c:474 NOCError: FATAL ERRORparam0 :zero, param1 :zero, param2 :zero.
[ 2533.930575] Thread ID      : 0x00000070  Thread name    : ABTIST182  Process ID     : 0
[ 2533.930575] Register:
[ 2533.930575] SP : 0x4b5d4cd8
[ 2533.930575] FP : 0x4b5d4ce0
[ 2533.930575] PC : 0x4b1080c4
[ 2533.930575] SSR : 0x00000008
[ 2533.930575] BADVA : 0x00020000
[ 2533.930575] LR : 0x4b107860
[ 2533.930575]
[ 2533.930575] Stack Dump
[ 2533.930575] from : 0x4b5d4cd8
[ 2533.930575] to   : 0x4b5d4db0
[ 2533.930575]
[ 2533.977615] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
[ 2533.999811] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
[ 2534.007936] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
[ 2534.040530] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
```

kirdesde, 2023-12-29:
That is a crash of the remoteproc. It runs a QCA firmware blob (as part of the wifi firmware)
Since this is closed source, it's not possible to further analyse that crash (by us)

Have a look in logread (grep hostapd) and check what is happening at the time of the crash, maybe a specific device is triggering that crash

pwned-pixel, 2023-12-29:
The latest ath11k firmware blobs are not so good. 
https://forum.openwrt.org/t/xiaomi-ax3600-ath11k-firmware-crash-qcom-q6v5-wcss-pil-cd00000-q6v5-wcss-fatal-error-received/162603/66

The only thing you can do about is to adjust IRQ's manually or by a script. Don't use irqbalance for it. For my devices (AX3600 and NBG7815) this works around this specific crashes.

john08burke, 2023-12-29:
Thanks for the script! I had to tweak the `kill_irqbalance` func a bit to first check if `irqbalance` existed as the cmd was failing since my device didn't have `irqbalance`.

The router has already been up longer than any other attempts :smile:. I will update/close if it continues to be stable.

john08burke, 2023-12-29:
Dang, spoke too soon. This is the full `logread` output up until the crash. It doesn't seem that any `hostapd` logs correlate with the crash.  Maybe a tcpdump would be of use?

```
Fri Dec 29 11:36:14 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 11:40:59 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 11:42:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.117 28:ec:9a:16:61:ee
Fri Dec 29 11:42:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.117 28:ec:9a:16:61:ee Ring-1661ee
Fri Dec 29 11:44:34 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 11:54:11 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 11:57:15 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 38:ca:84:01:dd:f4
Fri Dec 29 11:57:15 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 IEEE 802.11: disassociated due to inactivity
Fri Dec 29 11:57:16 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Dec 29 11:57:17 2023 daemon.notice hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 IEEE 802.11: did not acknowledge authentication response
Fri Dec 29 11:57:30 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 IEEE 802.11: authenticated
Fri Dec 29 11:57:30 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 IEEE 802.11: associated (aid 3)
Fri Dec 29 11:57:30 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 38:ca:84:01:dd:f4 auth_alg=open
Fri Dec 29 11:57:30 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 RADIUS: starting accounting session 2305B95689E81BAC
Fri Dec 29 11:57:30 2023 daemon.info hostapd: phy1-ap0: STA 38:ca:84:01:dd:f4 WPA: pairwise key handshake completed (RSN)
Fri Dec 29 11:57:30 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 38:ca:84:01:dd:f4
Fri Dec 29 11:57:31 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.188 38:ca:84:01:dd:f4
Fri Dec 29 11:57:31 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.188 38:ca:84:01:dd:f4 HP01DDF2
Fri Dec 29 11:57:32 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 11:57:32 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:01:28 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:07:07 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:17:04 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:24:24 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:29:50 2023 daemon.warn odhcpd[2008]: A default route is present but there is no public prefix on lan thus we don't announce a default route by overriding ra_lifetime!
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535]
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] NOC_error.c:474 NOCError: FATAL ERRORparam0 :zero, param1 :zero, param2 :zero.
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] Thread ID      : 0x00000070  Thread name    : ABTIST182  Process ID     : 0
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] Register:
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] SP : 0x4b5d4cd8
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] FP : 0x4b5d4ce0
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] PC : 0x4b1080c4
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] SSR : 0x00000008
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] BADVA : 0x00020000
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] LR : 0x4b107860
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535]
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] Stack Dump
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] from : 0x4b5d4cd8
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535] to   : 0x4b5d4db0
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.315535]
Fri Dec 29 12:30:19 2023 kern.err kernel: [ 4867.362444] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
[... cut ...]
Fri Dec 29 12:30:19 2023 kern.warn kernel: [ 4867.753544] ath11k c000000.wifi: Failed to set dtim period for VDEV 0: -108
Fri Dec 29 12:30:19 2023 kern.warn kernel: [ 4867.760329] ath11k c000000.wifi: WMI vdev 0 failed to send FILS discovery enable/disable command
Fri Dec 29 12:30:24 2023 kern.err kernel: [ 4872.737702] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
Fri Dec 29 12:30:24 2023 kern.err kernel: [ 4872.737754] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
```

pwned-pixel, 2023-12-29:
Well then someone else have to have a look on this and give advise howto debug it and decide if it is worth to debug it. Given the firmware blob is closed source. But given what was written in the thread I've linked I doubt it makes sense.

What you could do is trying different clients and/or different use cases (e. g. ax vs ac or streaming vs. lan traffic only). It would be nice to find a trigger. I didn't dig further because the script was doing it for me.

To debug further you have to make your own builds. Hostapd is quite mute. For this you have to delete the patch: package/network/services/hostapd/patches/410-limit_debug_messages.patch
and to enable it within /etc/init.d/wpad script adding the option "-dd" to the procd command line.

john08burke, 2023-12-30:
So I think I have a stable setup now. I moved the 5G from AX to AC and has been stable since.

closed 2023-12-30 by john08burke

pwned-pixel, 2023-12-31:
This is just a workaround and you should not have closed it because it is a valid issue for this architecture.

I had a similar observation about ax vs. ac. Which brought me to remember that bss_coloring was causing crashes in the past during development phase. But bss_coloring is now enabled by default and cannot be disabled. On top of that it is set to 1 by default which does not make any sense if you are running multiple APs in your area. Because it has to be uniqe respectivley different for each device to have any effect. And as far I've read about this specific setting you have to enable Spatial Reuse Operation to have any speed gain. But it is not set anywhere or available as an option within hostapd.

Anyway I just can observe things and we are one step further as we know there is a difference between AC and AX.

TiagoSocialpal, 2024-01-18:
For IPQ8074 you need a different blob i'd say, check my last reply on #9555 , not having issues for some time now.

`diff --git a/package/firmware/ath11k-firmware/Makefile b/package/firmware/ath11k-firmware/Makefile
index ea4977aa68..75b65787d8 100644
--- a/package/firmware/ath11k-firmware/Makefile
+++ b/package/firmware/ath11k-firmware/Makefile
@@ -60,14 +60,14 @@ $(eval $(call Download,qcn9074-board))
 define Package/ath11k-firmware-ipq8074/install
        $(INSTALL_DIR) $(1)/lib/firmware/IPQ8074
        $(INSTALL_DATA) \
-               $(PKG_BUILD_DIR)/ath11k-firmware/IPQ8074/hw2.0/2.9.0.1/WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1/* \
+               $(PKG_BUILD_DIR)/ath11k-firmware/IPQ8074/hw2.0/testing/2.9.0.1/WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1/* \
                $(1)/lib/firmware/IPQ8074/
 endef

 define Package/ath11k-firmware-qcn9074/install
        $(INSTALL_DIR) $(1)/lib/firmware/ath11k/QCN9074/hw1.0
        $(INSTALL_DATA) \
-               $(PKG_BUILD_DIR)/ath11k-firmware/QCN9074/hw1.0/2.9.0.1/WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1/* \
+               $(PKG_BUILD_DIR)/ath11k-firmware/QCN9074/hw1.0/testing/2.9.0.1/WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1/* \
                $(1)/lib/firmware/ath11k/QCN9074/hw1.0/
        $(INSTALL_BIN) \
                $(DL_DIR)/$(QCN9074_BOARD_FILE) $(1)/lib/firmware/ath11k/QCN9074/hw1.0/board-2.bin`