openwrt/openwrt #14295: [ath79] Build for upgraded 16M flash doesn't work

EvilGremlin, 2023-12-28:
### Describe the bug

Can't figure it out with google. I'm doing all the same as i did for MR3420 some years ago but it doesn't boot. 

### OpenWrt version

23.05.2

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

openwrt-imagebuilder-23.05.2-ath79-tiny.Linux-x86_64

### Device

tplink_tl-wr841-v8

### Image kind

Self-built image

### Steps to reproduce

tiny-tplink.mk
```
define Device/tplink_tl-wr841-v8
  $(Device/tplink-16mlzma)
  SOC := ar9341
```
ar9341_tplink_tl-wr841-v8.dts
```
			uboot: partition@0 {
				label = "u-boot";
				reg = <0x000000 0x020000>;
				read-only;
			};

			partition@20000 {
				compatible = "tplink,firmware";
				label = "firmware";
				reg = <0x020000 0xff0000>;
			};

			art: partition@ff0000 {
				label = "art";
				reg = <0xff0000 0x1000000>;
				read-only;

```
`make image PROFILE="tplink_tl-wr841-v8" PACKAGES="luci-base luci-mod-system luci-mod-status luci-mod-network luci-mod-admin-full luci-proto-ipv6 luci-theme-material opkg -firewall4 -nftables -kmod-nft-offload -kmod-nft-core -luci-proto-ppp -odhcp6c -odhcpd-ipv6only -ppp -ppp-mod-pppoe uhttpd -dnsmasq"`

### Actual behaviour

```
␍␊
[00:55:52:797] ␍␊
[00:55:52:797] *********************************************␍␊
[00:55:52:797] *   U-Boot 1.1.4  (Nov 19 2014, 09:48:57)   *␍␊
[00:55:52:797] *********************************************␍␊
[00:55:52:797] ␍␊
[00:55:52:797] AP123 (AR9341) U-Boot for TL-WR841N/D v8␍␊
[00:55:52:797] ␍␊
[00:55:52:797] DRAM:   64 MB DDR 32-bit␍␊
[00:55:52:797] FLASH:  Macronix MX25L128 (16 MB)␍␊
[00:55:52:797] CLOCKS: 535/400/200/25 MHz (CPU/RAM/AHB/SPI)␍␊
[00:55:52:797] ␍␊
[00:55:52:797] ## Error: MAC is invalid, using fixed!␍␊
[... 188 lines cut ...]
[01:02:41:785] [    2.235351] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001000: 0x4c97 instead␍␊
[01:02:41:785] [    2.245193] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001004: 0xde86 instead␍␊
[01:02:41:830] [    2.255004] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001008: 0x2097 instead␍␊
[01:02:41:830] [    2.264814] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0000100c: 0x3b52 instead␍␊
[01:02:41:830] [    2.274625] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001010: 0x60b9 instead␍␊
[01:02:41:830] [    2.284437] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001014: 0x6b9b instead␍␊
[01:02:41:830] [    2.294249] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001018: 0x86e7 instead␍␊
[01:02:41:874] [    2.304068] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x0000101c: 0x5842 instead␍␊
[01:02:41:874] [    2.313880] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001020: 0xbf0b instead␍␊
[01:02:41:874] [    2.323690] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00001024: 0xb830 instead␍␊
[01:02:41:874] [    2.333500] jffs2: Further such events for this erase block will not be printed␍␊
etc
```

### Expected behaviour

To actually boot.

### Additional info

I use this uboot now, though it's of no relevance: https://github.com/pepe2k/u-boot_mod

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

e9hack, 2023-12-29:
The size values are looking wrong. I did do the same for a TP-Link wdr3600:
```                        
                partitions {
                        compatible = "fixed-partitions";
                        #address-cells = <1>;
                        #size-cells = <1>;

                        uboot: partition@0 {
                                label = "u-boot";
                                reg = <0x000000 0x020000>;
                                read-only;

                                nvmem-layout {
                                        compatible = "fixed-layout";
                                        #address-cells = <1>;
                                        #size-cells = <1>;

                                        macaddr_uboot_1fc00: macaddr@1fc00 {
                                                compatible = "mac-base";
                                                reg = <0x1fc00 0x6>;
                                                #nvmem-cell-cells = <1>;
                                        };
                                };
                        };

                        partition@20000 {
                                compatible = "tplink,firmware";
                                label = "firmware";
                                reg = <0x020000 0xfd0000>;
                        };

                        art: partition@ff0000 {
                                label = "art";
                                reg = <0xff0000 0x010000>;
                                read-only;

                                nvmem-layout {
                                        compatible = "fixed-layout";
                                        #address-cells = <1>;
                                        #size-cells = <1>;

                                        cal_art_1000: cal@1000 {
                                                reg = <0x1000 0x440>;
                                        };

                                        cal_art_5000: cal@5000 {
                                                reg = <0x5000 0x440>;
                                        };
                                };

                        };

```

EvilGremlin, 2023-12-29:
oooh indeed it's length, not end!

brada4, 2023-12-31:
> invalid MAC address, using random

did you copy art to that location in dts?

EvilGremlin, 2023-12-31:
Yep, art backup is flashed there. But you can see in the log it don't use new layout for some reason. `0x0000003f0000-0x000000400000 : "art"`

brada4, 2023-12-31:
clear buikd dir and retry....

EvilGremlin, 2024-01-01:
@brada4 Note that i use imagebuilder! Cleaning build_dir is imposssible, it won't build at all then. Anyway, looks like problem is dts not being compiled into dtb. If it should be compiled - it's a bug. It it shouldn't - dts must be deleted from imagebuilder packages.

brada4, 2024-01-01:
imagebuilder does not recompose the kernel with different dts. You must compile the kernel.

EvilGremlin, 2024-01-01:
Yep, i fugured. Which means presence of dts is misleading and they must be removed.

xabolcs, 2024-01-01:
Nah, they should not leave!
See my HooToo HT-TM2 repository!

tmn505, 2024-01-03:
With a little bit of hackery it's possible to recompile dts in Image Builder. Check example in https://github.com/openwrt/openwrt/issues/14215#issuecomment-1869705547, especially changes in `include/image.mk` from the patch and read the note below the patch. If Your board uses lzma-loader, then You might need patch from jwmullay's linked repository.

P.S. You shouldn't remove original text in first post, because some of the comments look now as out of context. You could use Strikethrough from https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#styling-text.

EvilGremlin, 2024-01-03:
Thanks! 
I wouldn't remove context if there were anything useful before jwmullay's repo link.

brada4, 2024-01-03:
Could you please revert original post to initial order? Seriously it was determined that imagebuilder is not fit for hw mods. No more issue there.