openwrt/openwrt #14297: tp-link archer ax-23 version 1 did not have a DFS Feature. Channel selection cannot seems to work and buggy

mrusli70, 2023-12-29:
### Describe the bug

i install the openwrt 23.05.2 to the wifi router tp-link archer ax-23 version 1. And come to notice that it cannot seems to do a DFS channel selection. I have to change to the US country code in order to go the higher channels. some channels that i selected are not working. it goes black and my phones couldn't detect the wifi signals. 

I have another to-link archer ax-23 version 1.2 and it seems that i cannot upgrade to Openwrt firmware 23.05.2 at all only tp-link archer ax-23 version 1 is working. 

### OpenWrt version

OpenWrt 23.05.2

### OpenWrt release

r23630-842932a63d

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrusli70, 2023-12-29:
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''

brada4, 2023-12-29:
If you set channel to auto in US regulatory domain you must be able to select 80MHz channel 36 even if DFS is not supported as per:
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1777

Disable 5GHz wifi, power off device to clear any remnants of past country code mishaps, save US country code for 5GHz antenna, change channel to auto and bandwidth to 80MHz, then run `logread -f -e hostap | tee /tmp/wifi_log.txt` while enabling wifi. Once wifi appears in whatever client shows SSID-s  (under 5 minutes, may take longer going down and up again if real radar is felt in current channel) 

The radar detected in active channel is handled differently by different hardware, if there is no background survey ability in AP it will switch off radio and do new 5min auto-selection, some clients will disconnect even from correct quick channel update messages from AP. Just patiently wait for radio to stabilise.

00 country code is usable in big cities if at all, it requires reception of a beacon other AP before transmitting in particular channel, that never happens in the forest.