openwrt/openwrt #14299: procd: memory leak

zhaojh329, 2023-12-29:
### Describe the bug

procd takes up a lot of memory, and it is not clear why.

[image]

### OpenWrt version

r23674-60e49cf170

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT6000

### Image kind

Self-built image

### Steps to reproduce

No special operation.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-29:
It is running GL.inet firmware, your router is supported only by OpenWRT snapshot, not 23.05.x series.
Either install OpenWRT snapshot (not very easy if it is first time ever installing OpenWRT) or better contact GL.inet support for fixed 23.x fork version.

xabolcs, 2023-12-29:
#14152 backported the device support to `openwrt-23.05`.

One could use that branch by self compiling or just download the binaries: [mediatek/filogic](https://downloads.openwrt.org/releases/23.05-SNAPSHOT/targets/mediatek/filogic/)

For example, I use an Image Builder provided image from a few days ago for my GL-MT6000:
```
OpenWrt 23.05-SNAPSHOT r23673-59fd8f08cf / LuCI openwrt-23.05 branch git-23.357.58018-024e7ab
```

brada4, 2023-12-29:
Anything badly restarting in `logread` ? It is same procd as 23.05.2, which kind of works fine on raspberry etc.