openwrt/openwrt #14301: iwinfo: ifname reported is wrong for "mesh"

McGiverGim, 2023-12-29:
### Describe the bug

I like to put a "custom" name to the ifname in the wireless section. For example to have a `wlan24g`, `wlan5g` and `mesh`.
The `wlan24g`, `wlan5g` work perfectly, but since some months ago, the `mesh` ifname appears like `m-11s-0` when I use the command `iwinfo` and thus it appears too in the luci interface with this name.
Maybe `mesh` is banned as ifname for some reason?

### OpenWrt version

r24600-9291f6025f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

1. Create some wireless network. In my case a 802.11s one.
2. Put an `ifname` of `mesh`.
3. Save and reload.
4. Use the `iwinfo` command or look at luci. The ifname will be m-11s-0.

### Actual behaviour

The `mesh` interface name appears as `m-11s-0`.

### Expected behaviour

The `mesh` interface name must appear as `mesh`.

### Additional info

This worked right before, but I can't remember when.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

McGiverGim, 2024-02-01:
Latest master seems to have fixed this, I don't know which commit, but I can close it.

closed 2024-02-01 by McGiverGim

McGiverGim, 2024-04-02:
If someone more has this issue, only to confirm, the problem with the name was caused by the `mesh11sd` package, that changes the name on the fly. 
I don't know if this can be modfiied or, in my case, that I have only two nodes, I ended removing this package.

xabolcs, 2024-04-02:
FYI.

As you already see `mesh11sd` renames the interface to `m-xxxx-n`, e.g. `m-11s-0`.

You can influence the `xxxx` part of the above with the option `mesh_basename`:
```

	###########################################################################################
	# mesh_basename (optional)
	# The first 4 characters after non alphanumerics are removed are used as the mesh_basename
	# The mesh_basename is used to construct a unique mesh interface name of the form m-xxxx-n
	# Default: 11s
	# Results in ifname=m-11s-0 for the first mesh interface
	# Example: link
	# Results in ifname=m-link-0
	#
	#option mesh_basename 'link'
```

See [documentation of `mesh11sd`](https://openwrt.org/docs/guide-user/network/wifi/mesh/mesh11sd#configuration) for the details!