openwrt/openwrt #14303: mt7622/mt7915e e8450 kernel panic dmesg log

xartin, 2023-12-29:
### Describe the bug

My router encountered a random kernel  panic and wifi capability ceased functioning requiring a reboot. Only the wifi driver ceased functioning.

Here's the dmesg log [dmesg-openwrt.log](https://github.com/openwrt/openwrt/files/13795745/dmesg-openwrt.log)

The router is connected to a 1500 watt apc ups with a serviced battery module so electricity instability is not a concern that would have otherwise been a plausible contributing factor.   
  
There have been no package version updates or any modifications besides enabling WED and hardware flow offloading to aid with 160hz wifi performance. The router model specifically is a Belkin rt3200

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Unable to reproduce this as the occurrence was very random and the router had been running without interruption for several weeks. i expect there may be potential for rare or infrequent reoccurrence. the router has been functioning reliably since being rebooted.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2023-12-29:
Since 23.05 and SNAPSHOT, if you enabled WED mt7915 driver is not able to recover from an error caused by the chip's firmware bugging out. If you have WED enabled, try disabling that and try again. You should be able to test if it worked immediately if `echo 1 > /sys/kernel/debug/ieee80211/wl1/mt76/sys_recovery` does not cause the driver to crash.

xartin, 2023-12-29:
So in theory if I attempted testing using that debug recovery command with WED enabled there's a greater chance of reproducing and validating the kernel panic error?

rany2, 2023-12-29:
Yes, if you want you can try with WED enabled and disabled but it's already a known issue with 23.05 and SNAPSHOT. It's been extensively written about over here: https://github.com/openwrt/mt76/issues/754