openwrt/openwrt #14305: ramips: mt76x8:  kernel panic on Tama W06

ikazuhiro, 2023-12-30:
### Describe the bug

When I build image for Tama W06 with current master branch, kernel panic occurs and the device never boot.   I confirmed reverting commit 1f818b09f8ae1cbb4a2ae7ddaccaa9ce0c12ca93 resolved the issue.  Here is bootlog

```
[    0.000000] Linux version 5.15.145 (kazuhiro@fafnir) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23352+9-3baa45fbd8) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Sat Dec 30 05:09:52 2023
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is Tama W06
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=0000000f
[    0.000000] Readback ErrCtl register=0000000f
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56032K/65536K available (5576K kernel code, 600K rwdata, 1212K rodata, 1180K init, 208K bss, 9504K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] intc: using register map from devicetree
[    0.000000] Kernel panic - not syncing: unable to get CPU clock, err=-517
[    0.000000] Rebooting in 1 seconds..
[    0.000000] Reboot failed -- System halted
```

### OpenWrt version

latest master branch

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Tama W06

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

DragonBluep, 2023-12-30:
Please test this patch.
```
--- a/target/linux/ramips/dts/mt7628an.dtsi
+++ b/target/linux/ramips/dts/mt7628an.dtsi
@@ -39,7 +39,7 @@
 		#size-cells = <1>;
 
 		sysc: syscon@0 {
-			compatible = "ralink,mt7628-sysc", "syscon";
+			compatible = "ralink,mt7628-sysc", "ralink,mt7688-sysc", "syscon";
 			reg = <0x0 0x100>;
 			#clock-cells = <1>;
 			#reset-cells = <1>;
```

ikazuhiro, 2023-12-31:
Your patch resolved the issue.
Thank you!

commit c0d7842b (2024-01-06): ramips: add missing syscon compatible strings for MT7688 and RT3052

MT7688 devices use the "mt7628an.dtsi" as the template. And RT3052
devices use the "rt3050.dtsi" as template. Therefore, we need to add
the corresponding system controller compatible strings to make them
work properly.

Fixes: 1f818b09f8ae ("ramips: add proper system clock and reset driver support for legacy SoCs")
Fixes: #14305
closed 2024-01-06 by openwrt-bot