openwrt/openwrt #14306: Download throttle bug in SQM Cake

gameinn1, 2023-12-30:
### Describe the bug

I'm not entirely sure this is directly related to OpenWRT but more of a problem with SQM Cake. Maybe someone here can look into it or explain why it happens.

I am able to get SQM Cake to go from this (which I think is close to as perfect as you could get): https://www.waveform.com/tools/bufferbloat?test-id=c4fd02e6-1730-47d0-8b94-d56151c11e8b

To this: https://www.waveform.com/tools/bufferbloat?test-id=c1304483-6794-46b4-87c1-7ca41217359c

Notice the massive jump in 95% idle ping and jitter. It feels really bad.

### OpenWrt version

23.05

### OpenWrt release

N/A

### OpenWrt target/subtarget

N/A

### Device

Zyxel WSM20

### Image kind

Official downloaded image

### Steps to reproduce

1: Do a bufferbloat test on a quiet line to confirm A+ rating and low jitter.
2: Set up a client with a download throttle around say half of your download limit and let it run. This can be a torrent, steam download etc.
3: Do another bufferbloat test.

### Actual behaviour

After step 3 the line is much worse with really bad jitter and 95% pings: https://www.waveform.com/tools/bufferbloat?test-id=c1304483-6794-46b4-87c1-7ca41217359c

### Expected behaviour

After step 3 the line would function close to an idle line (or atleast get an A+ rating). Obviously some higher ping and jitter expected in the 2-3ms range but not this extreme. This is the whole purpose of SQM. If someone throttling a download does this what's the point.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-12-30:
2.5) close and re-open browser tab between tests

rany2, 2023-12-30:
It's not a bug, your mt7621 cannot handle CAKE as it is very CPU intensive especially when you get to speeds beyond your average ADSL speeds of 10Mbps down. That's why it's not the default.

You could try to overclock it to 1.2Ghz with a kernel patch.

rany2, 2023-12-30:
Your ISP seems to have an excellent QoS though, so I don't think you really need to setup Cake. If you're doing this to divide the bandwidth among your clients fairly, you could try less CPU intensive fq_codel based options.

brada4, 2023-12-30:
There are few simpler ways:
1/ limit on per-user VLAN using nftables script (in raw table against customer vlan drop above x kb/sec)
2/ hfsc is kind of very old, but it has ability to allocate guarantees per interface, then redistribute idle BW to other consumers.

gameinn1, 2023-12-30:
> It's not a bug, your mt7621 cannot handle CAKE as it is very CPU intensive especially when you get to speeds beyond your average ADSL speeds of 10Mbps down. That's why it's not the default.
> 
> You could try to overclock it to 1.2Ghz with a kernel patch but it probably won't help much.

I know after asking other people that the mt7621 can do cake at over 130Mbps. I can't see how it operates fine when you don't throttle download traffic but as soon as you do it completely fumbles.

I won't completely rule it out being cpu though. Jitter does improve a considerable amount if I go to fq_codel.

gameinn1, 2023-12-30:
> Your ISP seems to have an excellent QoS though, so I don't think you really need to setup Cake. If you're doing this to divide the bandwidth among your clients fairly, you could try less CPU intensive fq_codel based options.

You mis understand. The good result is cake. Just not when a throttled download happens. Here's my standard ISP connection with no sqm: https://www.waveform.com/tools/bufferbloat?test-id=f43b507a-5111-4e26-bbde-0c24cd82d610

brada4, 2023-12-30:
Add file in /etc/nftables.d (not compatible to offload) to limit output rate below what line can sustain.
2400kB/s = 19.2Mbps
```
chain up_limit {
        type filter hook output priority raw; policy accept;
        oif "wan" counter limit rate over 2400 kbytes/second counter drop
}
```
Counters serve as drop indicators , not functionally necessary. Odds are high fq_codel (aka "no sqm configured") will do the rest. Not compatible with fw offload just like any qos script.

Neustradamus, 2024-11-03:
@gameinn1: Have you always the problem?

closed 2025-02-19 by gameinn1