openwrt/openwrt #14315: PPPoE instability since 22.03.4

netstrong, 2024-01-01:
### Describe the bug

I'm not a professional user. I just using the OpenWrt on my TP-Link TD-W8970 v1 as a SOHO home modem/router in order to connect to the ADSL network.
I think it is worth to reach contact with you as I feel something is going wrong in the couple of last releases for this particular device, as it shows instability over PPPoE connection; I think it starts happening since version 22.03.4 until now. actually the 22.03.4 was disaster about this issue, it seems better now, but still needs attention.
I almost sure there is something wrong with the OpenWrt, as my othe ADSL device in the same area running with no problem at all with the same ISP.

It'll disconnected after a while and unable to recover the WAN connection until I manually hard reboot the device.
My current release is: 23.05.2

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

lantiq/xrx200

### Device

TP-LINK TD-W8970

### Image kind

Official downloaded image

### Steps to reproduce

you need to run this over the ADSL/PPPoE network in order to see the problem.

### Actual behaviour

PPPoE connection (My Internet connection) drops after a while and the device attemps for recovering it always fails. I need to reboot the whole device to back it into operation.

### Expected behaviour

My other device (ADSL Modem) seems to be fine (stock firmware of it's own company) even with very long uptime.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-02:
Is the device absolutely unreachable from LAN side or you can connect it?

netstrong, 2024-01-02:
> Is the device absolutely unreachable from LAN side or you can connect it?

Nothing wrong with LAN connectivity. Only dsl/PPPoE (Internet) fails.
I still be able to reach my TP-Link device through the LAN.

brada4, 2024-01-02:
what is found in `logread` when connection goes down
And cN itbe recovered res%arting network service `service network restart`

netstrong, 2024-01-03:
> what is found in `logread` when connection goes down And cN itbe recovered res%arting network service `service network restart`

From the moment I post this issue here, it doesn't happened yet (a WAN connection loss that device can't restore by itself).
Right now I log on into my OpenWrt using web (luci), and find out my PPPoE connection lost once, an hour ago, **but the device was able to successfully re-connect it** (it doesn't happen all the time like this).

According to the logs, I lost my connection after 13 hours of stable connection, it all started with this:

```
Wed Jan  3 20:14:34 2024 daemon.notice netifd: Network device 'dsl0' link is down
Wed Jan  3 20:14:34 2024 daemon.notice netifd: Interface 'wan' has link connectivity loss
Wed Jan  3 20:14:34 2024 kern.warn kernel: [49016.652593] leave showtime
Wed Jan  3 20:14:35 2024 daemon.info pppd[3278]: Terminating on signal 15
Wed Jan  3 20:14:35 2024 daemon.info pppd[3278]: Connect time 815.3 minutes.
Wed Jan  3 20:14:35 2024 daemon.info pppd[3278]: Sent 588856171 bytes, received 1473442000 bytes.
Wed Jan  3 20:14:35 2024 daemon.notice netifd: Network device 'pppoe-wan' link is down
Wed Jan  3 20:14:39 2024 daemon.notice netifd: Interface 'wan' is now down
```

And in less than a minute:

```
Wed Jan  3 20:15:03 2024 daemon.notice netifd: Network device 'dsl0' link is up
Wed Jan  3 20:15:03 2024 daemon.notice netifd: Interface 'wan' has link connectivity
Wed Jan  3 20:15:03 2024 daemon.notice netifd: Interface 'wan' is setting up now
Wed Jan  3 20:15:03 2024 kern.warn kernel: [49045.452443] enter showtime, cell rate: 0 - 2591, 1 - 2591, xdata addr: 0xa2bb0000
Wed Jan  3 20:15:03 2024 kern.warn kernel: [49045.464946] enter showtime, cell rate: 0 - 2591, 1 - 2591, xdata addr: 0xa2bb0000
Wed Jan  3 20:15:06 2024 daemon.info pppd[4829]: Plugin pppoe.so loaded.
Wed Jan  3 20:15:06 2024 daemon.info pppd[4829]: PPPoE plugin from pppd 2.4.9
Wed Jan  3 20:15:06 2024 daemon.notice pppd[4829]: pppd 2.4.9 started by root, uid 0
Wed Jan  3 20:15:06 2024 daemon.info pppd[4829]: PPP session is 27063
Wed Jan  3 20:15:06 2024 daemon.warn pppd[4829]: Connected to [A-MAC-ADDRESS] via interface dsl0
Wed Jan  3 20:15:06 2024 kern.info kernel: [49048.624669] pppoe-wan: renamed from ppp0
Wed Jan  3 20:15:06 2024 daemon.info pppd[4829]: Renamed interface ppp0 to pppoe-wan
Wed Jan  3 20:15:06 2024 daemon.info pppd[4829]: Using interface pppoe-wan
Wed Jan  3 20:15:06 2024 daemon.notice pppd[4829]: Connect: pppoe-wan <--> dsl0
Wed Jan  3 20:15:09 2024 daemon.info pppd[4829]: Remote message: Authentication success,Welcome!
```

I'll looking forward for catching one example of failed attempt and I'll post the new logs here.

brada4, 2024-01-03:
you might want to add pppd debug option [(somewhere in the wiki)](https://openwrt.org/docs/guide-user/network/wan/wan_interface_protocols#protocol_pppoe_ppp_over_ethernet) to better sift blame between DSL/PPP server/OpenWRT.

nicefile, 2024-01-04:
`Wed Jan  3 20:14:34 2024 daemon.notice netifd: Network device 'dsl0' link is down`
seems DSL line disconnect you . It might be DSL firmware blob issue or even hardware issue.
You might check DSL uptime and other DSL stats via
`ubus call dsl metrics`

khanjui, 2024-01-05:
https://forum.openwrt.org/t/lantiq-vrx200-xdsl-firmware-recommendation-thread/52937/90

https://xdarklight.github.io/lantiq-xdsl-firmware-info/

brada4, 2024-01-05:
the problem is pppoe does not reconnect sometimes, it is supposed to reconnect even if dsl reconnects every minute.

netstrong, 2024-01-16:
Hi again,
A few minutes ago, I faced with another `pppoe` failover. The Openwrt seems unable to restart the connection in 15 minutes, so I manually rebooted the device.
While the device was still attempting for re-connection, I login using `luci` and copy these logs:

```
Tue Jan 16 22:25:24 2024 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jan 16 22:25:24 2024 daemon.info pppd[5662]: Plugin pppoe.so loaded.
Tue Jan 16 22:25:24 2024 daemon.info pppd[5662]: PPPoE plugin from pppd 2.4.9
Tue Jan 16 22:25:24 2024 daemon.notice pppd[5662]: pppd 2.4.9 started by root, uid 0
Tue Jan 16 22:25:24 2024 daemon.debug pppd[5662]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Jan 16 22:25:24 2024 daemon.debug pppd[5662]:  dst ff:ff:ff:ff:ff:ff  src 00:00:01:00:00:00
Tue Jan 16 22:25:24 2024 daemon.debug pppd[5662]:  [service-name] [host-uniq  00 00 16 1e]
Tue Jan 16 22:25:29 2024 daemon.debug pppd[5662]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Jan 16 22:25:29 2024 daemon.debug pppd[5662]:  dst ff:ff:ff:ff:ff:ff  src 00:00:01:00:00:00
Tue Jan 16 22:25:29 2024 daemon.debug pppd[5662]:  [service-name] [host-uniq  00 00 16 1e]
Tue Jan 16 22:25:34 2024 daemon.debug pppd[5662]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Jan 16 22:25:34 2024 daemon.debug pppd[5662]:  dst ff:ff:ff:ff:ff:ff  src 00:00:01:00:00:00
Tue Jan 16 22:25:34 2024 daemon.debug pppd[5662]:  [service-name] [host-uniq  00 00 16 1e]
Tue Jan 16 22:25:39 2024 daemon.warn pppd[5662]: Timeout waiting for PADO packets
Tue Jan 16 22:25:39 2024 daemon.err pppd[5662]: Unable to complete PPPoE Discovery
Tue Jan 16 22:25:39 2024 daemon.info pppd[5662]: Exit.
Tue Jan 16 22:25:39 2024 daemon.notice netifd: Interface 'wan' is now down
```

Note: I was omitted the `option pppd_options 'debug'` into `/etc/config/network` since last week.

brada4, 2024-01-16:
Ok. Like dsl device hung? At least no response came back from pppoe server. Not your fault, you can rise log buffer to megabyte or so to absorb more noise.
- Try firmware advice
- If it ever repeats try pinging pppoe gateway (just make emergency data gather script)
Maybe helps to recover , not guaranteed
- unplug replug phone line
- ifconfig dsl down ; up
- service network restart

netstrong, 2024-01-16:
> At least no response came back from pppoe server.

I'm not sure. because as I reboot the device, everything back to normal; so, if it isn't my side, why rebooting will fix it?

>* Try firmware advice

you mean changing the dsl firmware? I think it is too advanced for me! the ideal solution is to fix this problem from Openwrt and then release a fixed version, so all users be able to install it just by upgrading Openwrt firmware.

> Maybe helps to recover , not guaranteed
> 
> * unplug replug phone line
> * ifconfig dsl down ; up
> * service network restart

I'll try that.

brada4, 2024-01-16:
Rebooting flips power on "DSL modem" module, I am trying to research with your hands if down+up or clearing main IP and adding it back via netifd restart is able to recover modem.
Something like this:
```
#!/bin/sh
cd `mktemp -d`
logread > syslog.out
ping -c5 ppoe gw ip > ping.out
service > service.out
tar cf report.tar *.out
ln -s report.tar ~/ 
```
Then download file from your homedir and try my suggested quackery.
What is interesting - what happened to DSL and PPPoE interfaces right before first pppd restart. Essentially log will be quite silent until "internet disconnects"

netstrong, 2024-02-09:
Hello again 👋;
I've downgrade my device into `OpenWrt 22.03.2` for a week now and it seems I don't have this problem in this version.

Although I think I had some instability since then, for example sometimes the WAN communication pauses for some seconds (the ping command returned timed out in that period), but I didn't lose the connection like before and the connectivity back to normal without interruption.

matus-sabo, 2024-02-12:
I did have similar problem, setting `option force_link ‘1’` on wan interface in `/etc/config/network` solved my issue.

arapov, 2024-04-29:
I am seeing the same issue after upgrading to 23.05

x29a, 2025-01-04:
Im facing a similar issue, but just quite recently (starting in 2025) with an unchanged self-built 23.05.5, so i assume its a change from my ISP (telekom) and since im using VDSL it might also be regarding the lantiq xrx200 modem.

Im on custom firmware

```
        "api_version": "4.17.18.6",
        "firmware_version": "5.9.1.4.0.7",
        "chipset": "Lantiq-VRX200",
        "driver_version": "1.5.17.6",
```

and the link just goes down with the following `logread -f`

```
Sat Jan  4 14:05:00 2025 daemon.notice netifd: Network device 'dsl0' link is down
Sat Jan  4 14:05:00 2025 kern.warn kernel: [36215.492715] leave showtime
Sat Jan  4 14:05:00 2025 daemon.notice netifd: VLAN 'dsl0.7' link is down
Sat Jan  4 14:05:00 2025 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Jan  4 14:05:02 2025 daemon.info pppd[20737]: Terminating on signal 15
Sat Jan  4 14:05:02 2025 daemon.info pppd[20737]: Connect time 51.4 minutes.
Sat Jan  4 14:05:02 2025 daemon.info pppd[20737]: Sent 21425894 bytes, received 551422475 bytes.
Sat Jan  4 14:05:02 2025 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sat Jan  4 14:05:02 2025 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Sat Jan  4 14:05:02 2025 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Jan  4 14:05:05 2025 daemon.notice netifd: Interface 'wan' is now down
Sat Jan  4 14:05:05 2025 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Sat Jan  4 14:06:29 2025 daemon.notice netifd: Network device 'dsl0' link is up
Sat Jan  4 14:06:29 2025 daemon.notice netifd: VLAN 'dsl0.7' link is up
Sat Jan  4 14:06:29 2025 daemon.notice netifd: Interface 'wan' has link connectivity
Sat Jan  4 14:06:29 2025 daemon.notice netifd: Interface 'wan' is setting up now
Sat Jan  4 14:06:29 2025 kern.warn kernel: [36304.377534] enter showtime
Sat Jan  4 14:06:29 2025 kern.warn kernel: [36304.395810] enter showtime
Sat Jan  4 14:06:33 2025 daemon.info pppd[21663]: Plugin pppoe.so loaded.
Sat Jan  4 14:06:33 2025 daemon.info pppd[21663]: PPPoE plugin from pppd 2.4.9
Sat Jan  4 14:06:33 2025 daemon.notice pppd[21663]: pppd 2.4.9 started by root, uid 0
Sat Jan  4 14:06:48 2025 daemon.warn pppd[21663]: Timeout waiting for PADO packets
Sat Jan  4 14:06:48 2025 daemon.err pppd[21663]: Unable to complete PPPoE Discovery
Sat Jan  4 14:06:48 2025 daemon.info pppd[21663]: Exit.
Sat Jan  4 14:06:49 2025 daemon.notice netifd: Interface 'wan' is now down
Sat Jan  4 14:06:49 2025 daemon.notice netifd: Interface 'wan' is setting up now
Sat Jan  4 14:06:49 2025 daemon.info pppd[21689]: Plugin pppoe.so loaded.
Sat Jan  4 14:06:49 2025 daemon.info pppd[21689]: PPPoE plugin from pppd 2.4.9
Sat Jan  4 14:06:49 2025 daemon.notice pppd[21689]: pppd 2.4.9 started by root, uid 0
Sat Jan  4 14:06:49 2025 daemon.info pppd[21689]: PPP session is 52
Sat Jan  4 14:06:49 2025 daemon.warn pppd[21689]: Connected to d4:04:ff:xx:xx:xx via interface dsl0.7
Sat Jan  4 14:06:49 2025 kern.info kernel: [36324.586899] pppoe-wan: renamed from ppp0
Sat Jan  4 14:06:49 2025 daemon.info pppd[21689]: Renamed interface ppp0 to pppoe-wan
Sat Jan  4 14:06:49 2025 daemon.info pppd[21689]: Using interface pppoe-wan
Sat Jan  4 14:06:49 2025 daemon.notice pppd[21689]: Connect: pppoe-wan <--> dsl0.7
Sat Jan  4 14:06:53 2025 daemon.info pppd[21689]: Remote message: SRU=21988#SRD=59940#
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: PAP authentication succeeded
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: peer from calling number D4:04:FF:xx:xx:xx authorized
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: local  IP address 79.240.xxx.xxx
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: remote IP address 62.155.xxx.xxx
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: primary   DNS address 217.237.151.115
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: secondary DNS address 217.237.148.102
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: local  LL address fe80::c8bd:8047:xxxx:xxxx
Sat Jan  4 14:06:53 2025 daemon.notice pppd[21689]: remote LL address fe80::d604:ffff:xxxx:xxxx
Sat Jan  4 14:06:53 2025 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sat Jan  4 14:06:53 2025 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Sat Jan  4 14:06:53 2025 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Jan  4 14:06:53 2025 daemon.notice netifd: Interface 'wan' is now up
Sat Jan  4 14:06:53 2025 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sat Jan  4 14:06:53 2025 daemon.info dnsmasq[1]: using nameserver 217.237.151.115#53
Sat Jan  4 14:06:53 2025 daemon.info dnsmasq[1]: using nameserver 217.237.148.102#53
```

so i rather think its an ISP/modem/VDSL issue :/

netstrong, 2025-01-04:
> Im facing a similar issue, but just quite recently (starting in 2025) with an unchanged self-built 23.05.5, so i assume its a change from my ISP (telekom) and since im using VDSL it might also be regarding the lantiq xrx200 modem.
> 

I had my problem fixed. Now I'm of the latest version (OpenWrt 23.05.5) and the link down problem didn't happened to me since I change my telephone socket and some re-wiring.
Even though the SNR ratio seems OK back then, my problem only gone when I made some physical changes over cables and sockets.

brada4, 2025-01-04:
If you have some DSL stats recorded you might be able to derive one that signifies poor physical link. In basic setup speed is dictated by distance to providers cabinet only, then come mice and rain and other defects, but on each sunny day it should be stable and at same speed.

x29a, 2025-01-05:
thanks, im investigating using a backup `zyxel MG1312-B30A` to see how that behaves in the same setup and so far, no reconnect for over 6h where my 9980 would reconnect multiple times per hour. could be hardware fault as well. will need some more investigation.

namchuai, 2025-05-03:
I just have this problem and have to reboot. I'm using OpenWrt 24.10.1 on pi4.

brada4, 2025-05-04:
PADO response missing means 
1/ physicsl (dsl/pon) connection is not yet ready
2/ vlan is wrong in connection setup

x29a, 2025-05-04:
@namchuai which modem are you using on the pi4?

namchuai, 2025-05-05:
> [@namchuai](https://github.com/namchuai) which modem are you using on the pi4?

Sorry, I'm a noob and I don't know how to get you that info. I will try to get the log next time the router failed pppoe.

x29a, 2025-05-05:
no worries, which router do you have?

namchuai, 2025-05-06:
> no worries, which router do you have?

I'm using raspberry pi4 to running OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.116.36015~cca4654

Here's my log when the wan is down
```
Tue May  6 08:11:52 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:11:52 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:11:52 2025 daemon.info pppd[20575]: Plugin pppoe.so loaded.
Tue May  6 08:11:52 2025 daemon.info pppd[20575]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:11:52 2025 daemon.notice pppd[20575]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:11:53 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:12:27 2025 daemon.warn pppd[20575]: Timeout waiting for PADO packets
Tue May  6 08:12:27 2025 daemon.err pppd[20575]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:12:27 2025 daemon.info pppd[20575]: Exit.
Tue May  6 08:12:27 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:12:27 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:12:27 2025 daemon.info pppd[20715]: Plugin pppoe.so loaded.
Tue May  6 08:12:27 2025 daemon.info pppd[20715]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:12:27 2025 daemon.notice pppd[20715]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:12:28 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:13:02 2025 daemon.warn pppd[20715]: Timeout waiting for PADO packets
Tue May  6 08:13:03 2025 daemon.err pppd[20715]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:13:03 2025 daemon.info pppd[20715]: Exit.
Tue May  6 08:13:03 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:13:03 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:13:03 2025 daemon.info pppd[20858]: Plugin pppoe.so loaded.
Tue May  6 08:13:03 2025 daemon.info pppd[20858]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:13:03 2025 daemon.notice pppd[20858]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:13:04 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:13:32 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:13:38 2025 daemon.warn pppd[20858]: Timeout waiting for PADO packets
Tue May  6 08:13:38 2025 daemon.err pppd[20858]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:13:38 2025 daemon.info pppd[20858]: Exit.
Tue May  6 08:13:38 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:13:38 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:13:38 2025 daemon.info pppd[20996]: Plugin pppoe.so loaded.
Tue May  6 08:13:38 2025 daemon.info pppd[20996]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:13:38 2025 daemon.notice pppd[20996]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:13:39 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:14:13 2025 daemon.warn pppd[20996]: Timeout waiting for PADO packets
Tue May  6 08:14:13 2025 daemon.err pppd[20996]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:14:13 2025 daemon.info pppd[20996]: Exit.
Tue May  6 08:14:13 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:14:13 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:14:13 2025 daemon.info pppd[21110]: Plugin pppoe.so loaded.
Tue May  6 08:14:13 2025 daemon.info pppd[21110]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:14:13 2025 daemon.notice pppd[21110]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:14:14 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:14:49 2025 daemon.warn pppd[21110]: Timeout waiting for PADO packets
Tue May  6 08:14:49 2025 daemon.err pppd[21110]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:14:49 2025 daemon.info pppd[21110]: Exit.
Tue May  6 08:14:49 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:14:49 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:14:49 2025 daemon.info pppd[21213]: Plugin pppoe.so loaded.
Tue May  6 08:14:49 2025 daemon.info pppd[21213]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:14:49 2025 daemon.notice pppd[21213]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:14:50 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:15:24 2025 daemon.warn pppd[21213]: Timeout waiting for PADO packets
Tue May  6 08:15:24 2025 daemon.err pppd[21213]: Unable to complete PPPoE Discovery phase 1
Tue May  6 08:15:24 2025 daemon.info pppd[21213]: Exit.
Tue May  6 08:15:24 2025 daemon.notice netifd: Interface 'wan' is now down
Tue May  6 08:15:24 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue May  6 08:15:24 2025 daemon.info pppd[21316]: Plugin pppoe.so loaded.
Tue May  6 08:15:24 2025 daemon.info pppd[21316]: PPPoE plugin from pppd 2.5.1
Tue May  6 08:15:24 2025 daemon.notice pppd[21316]: pppd 2.5.1 started by root, uid 0
Tue May  6 08:15:25 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:15:45 2025 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx:xx
Tue May  6 08:15:45 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.169 xx:xx:xx:xx:xx:xx
[... cut ...]
Tue May  6 08:16:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.169 xx:xx:xx:xx:xx:xx
Tue May  6 08:16:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.227 xx:xx:xx:xx:xx:xx
Tue May  6 08:16:48 2025 daemon.info dnsmasq-dhcp[1]: DHCPNAK(br-lan) 192.168.1.227 xx:xx:xx:xx:xx:xx wrong server-ID
Tue May  6 08:16:50 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
Tue May  6 08:16:58 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
```

omaqdam, 2025-05-06:
> > no worries, which router do you have?
> 
> I'm using raspberry pi4 to running OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.116.36015~cca4654
> 
> <img alt="Image" width="1117" src="https://private-user-images.githubusercontent.com/10397206/440622823-68095992-1bf4-462c-aea6-bccaddbb9f46.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDY1MDg3MzEsIm5iZiI6MTc0NjUwODQzMSwicGF0aCI6Ii8xMDM5NzIwNi80NDA2MjI4MjMtNjgwOTU5OTItMWJmNC00NjJjLWFlYTYtYmNjYWRkYmI5ZjQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTA2VDA1MTM1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThmMzFjNjVlMjI0YmY2YmM2YmY4MGIyNDU5NWY5MDJkMGZjMWU2YzQyNDk0NTNmMTZmM2QyMWIzZjRhNWIyYmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cJuh9lUkLOYn-eWDSxXmwIncAnCDYTXgj2c7lfmYPt4">
> Here's my log when the wan is down
> 
> ```
> Tue May  6 08:11:52 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:11:52 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:11:52 2025 daemon.info pppd[20575]: Plugin pppoe.so loaded.
> Tue May  6 08:11:52 2025 daemon.info pppd[20575]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:11:52 2025 daemon.notice pppd[20575]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:11:53 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:12:27 2025 daemon.warn pppd[20575]: Timeout waiting for PADO packets
> Tue May  6 08:12:27 2025 daemon.err pppd[20575]: Unable to complete PPPoE Discovery phase 1
> Tue May  6 08:12:27 2025 daemon.info pppd[20575]: Exit.
> Tue May  6 08:12:27 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:12:27 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:12:27 2025 daemon.info pppd[20715]: Plugin pppoe.so loaded.
> Tue May  6 08:12:27 2025 daemon.info pppd[20715]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:12:27 2025 daemon.notice pppd[20715]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:12:28 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:13:02 2025 daemon.warn pppd[20715]: Timeout waiting for PADO packets
> Tue May  6 08:13:03 2025 daemon.err pppd[20715]: Unable to complete PPPoE Discovery phase 1
> Tue May  6 08:13:03 2025 daemon.info pppd[20715]: Exit.
> Tue May  6 08:13:03 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:13:03 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:13:03 2025 daemon.info pppd[20858]: Plugin pppoe.so loaded.
> Tue May  6 08:13:03 2025 daemon.info pppd[20858]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:13:03 2025 daemon.notice pppd[20858]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:13:04 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:13:32 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:13:38 2025 daemon.warn pppd[20858]: Timeout waiting for PADO packets
> Tue May  6 08:13:38 2025 daemon.err pppd[20858]: Unable to complete PPPoE Discovery phase 1
> Tue May  6 08:13:38 2025 daemon.info pppd[20858]: Exit.
> Tue May  6 08:13:38 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:13:38 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:13:38 2025 daemon.info pppd[20996]: Plugin pppoe.so loaded.
> Tue May  6 08:13:38 2025 daemon.info pppd[20996]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:13:38 2025 daemon.notice pppd[20996]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:13:39 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:14:13 2025 daemon.warn pppd[20996]: Timeout waiting for PADO packets
> Tue May  6 08:14:13 2025 daemon.err pppd[20996]: Unable to complete PPPoE Discovery phase 1
> Tue May  6 08:14:13 2025 daemon.info pppd[20996]: Exit.
> Tue May  6 08:14:13 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:14:13 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:14:13 2025 daemon.info pppd[21110]: Plugin pppoe.so loaded.
> Tue May  6 08:14:13 2025 daemon.info pppd[21110]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:14:13 2025 daemon.notice pppd[21110]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:14:14 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:14:49 2025 daemon.warn pppd[21110]: Timeout waiting for PADO packets
> Tue May  6 08:14:49 2025 daemon.err pppd[21110]: Unable to complete PPPoE Discovery phase 1
> Tue May  6 08:14:49 2025 daemon.info pppd[21110]: Exit.
> Tue May  6 08:14:49 2025 daemon.notice netifd: Interface 'wan' is now down
> Tue May  6 08:14:49 2025 daemon.notice netifd: Interface 'wan' is setting up now
> Tue May  6 08:14:49 2025 daemon.info pppd[21213]: Plugin pppoe.so loaded.
> Tue May  6 08:14:49 2025 daemon.info pppd[21213]: PPPoE plugin from pppd 2.5.1
> Tue May  6 08:14:49 2025 daemon.notice pppd[21213]: pppd 2.5.1 started by root, uid 0
> Tue May  6 08:14:50 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:15:24 2025 daemon.warn pppd[21213]: Timeout waiting for PADO packets
[... cut ...]
> Tue May  6 08:16:50 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> Tue May  6 08:16:58 2025 daemon.warn odhcpd[1032]: No default route present, overriding ra_lifetime to 0!
> ```

from given info, i assume that ur wan connection was working fine and then suddenly this issue came up.
i went through the attached logs and i believe if u haven't changed any configuration probably the issue either it's physically (cables ..etc) from your side or from the ISP side.

namchuai, 2025-05-06:
> from given info, i assume that ur wan connection was working fine and then suddenly this issue came up. i went through the attached logs and i believe if u haven't changed any configuration probably [...]

This problem does not occur when I'm using OpenWrt 23.x.x. I upgrade to 24 because it fix the problem with docker. Same hardware, same cables.
Now, I lost internet connection around every 2 days (I don't recall exactly) but works normally again after a reboot.

kcduke42, 2025-09-16:
Same here. The session drops every few days. Reloading the interface doesn't help. Rebooting the router usually fixes it. 
`System
Hostname	OpenWrt
Model	MERCUSYS MR70X v1
Architecture	MediaTek MT7621 ver:1 eco:3
Target Platform	ramips/mt7621
Firmware Version	OpenWrt 23.05.4 r24012-d8dd03c46f / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38
Kernel Version	5.15.162`
Here are the logs:
`Tue Sep 16 11:06:33 2025 daemon.notice netifd: Interface 'wan' is now down
Tue Sep 16 11:06:33 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue Sep 16 11:06:33 2025 daemon.info pppd[21653]: Plugin pppoe.so loaded.
Tue Sep 16 11:06:33 2025 daemon.info pppd[21653]: PPPoE plugin from pppd 2.4.9
Tue Sep 16 11:06:34 2025 daemon.notice pppd[21653]: pppd 2.4.9 started by root, uid 0
Tue Sep 16 11:06:34 2025 daemon.debug pppd[21653]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:34 2025 daemon.debug pppd[21653]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:34 2025 daemon.debug pppd[21653]:  [service-name] [host-uniq  95 54 00 00]
Tue Sep 16 11:06:37 2025 daemon.info hostapd: phy1-ap0: STA 08:ed:b9:10:69:99 IEEE 802.11: authenticated
Tue Sep 16 11:06:37 2025 daemon.info hostapd: phy1-ap0: STA 08:ed:b9:10:69:99 IEEE 802.11: associated (aid 1)
Tue Sep 16 11:06:37 2025 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 08:ed:b9:10:69:99 auth_alg=open
Tue Sep 16 11:06:37 2025 daemon.info hostapd: phy1-ap0: STA 08:ed:b9:10:69:99 WPA: pairwise key handshake completed (RSN)
Tue Sep 16 11:06:37 2025 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 08:ed:b9:10:69:99
Tue Sep 16 11:06:37 2025 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.171 08:ed:b9:10:69:99
Tue Sep 16 11:06:37 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.171 08:ed:b9:10:69:99 STEND_ARMA
Tue Sep 16 11:06:39 2025 daemon.debug pppd[21653]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:39 2025 daemon.debug pppd[21653]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:39 2025 daemon.debug pppd[21653]:  [service-name] [host-uniq  95 54 00 00]
Tue Sep 16 11:06:44 2025 daemon.debug pppd[21653]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:44 2025 daemon.debug pppd[21653]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:44 2025 daemon.debug pppd[21653]:  [service-name] [host-uniq  95 54 00 00]
Tue Sep 16 11:06:49 2025 daemon.warn pppd[21653]: Timeout waiting for PADO packets
Tue Sep 16 11:06:49 2025 daemon.err pppd[21653]: Unable to complete PPPoE Discovery
Tue Sep 16 11:06:49 2025 daemon.info pppd[21653]: Exit.
Tue Sep 16 11:06:49 2025 daemon.notice netifd: Interface 'wan' is now down
Tue Sep 16 11:06:49 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue Sep 16 11:06:49 2025 daemon.info pppd[21893]: Plugin pppoe.so loaded.
Tue Sep 16 11:06:49 2025 daemon.info pppd[21893]: PPPoE plugin from pppd 2.4.9
Tue Sep 16 11:06:49 2025 daemon.notice pppd[21893]: pppd 2.4.9 started by root, uid 0
Tue Sep 16 11:06:49 2025 daemon.debug pppd[21893]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:49 2025 daemon.debug pppd[21893]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:49 2025 daemon.debug pppd[21893]:  [service-name] [host-uniq  85 55 00 00]
Tue Sep 16 11:06:54 2025 daemon.debug pppd[21893]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:54 2025 daemon.debug pppd[21893]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:54 2025 daemon.debug pppd[21893]:  [service-name] [host-uniq  85 55 00 00]
Tue Sep 16 11:06:59 2025 daemon.debug pppd[21893]: Send PPPOE Discovery V1T1 PADI session 0x0 length 12
Tue Sep 16 11:06:59 2025 daemon.debug pppd[21893]:  dst ff:ff:ff:ff:ff:ff  src 66:57:48:82:3b:09
Tue Sep 16 11:06:59 2025 daemon.debug pppd[21893]:  [service-name] [host-uniq  85 55 00 00]
Tue Sep 16 11:07:04 2025 daemon.warn pppd[21893]: Timeout waiting for PADO packets
Tue Sep 16 11:07:04 2025 daemon.err pppd[21893]: Unable to complete PPPoE Discovery
Tue Sep 16 11:07:04 2025 daemon.info pppd[21893]: Exit.
Tue Sep 16 11:07:04 2025 daemon.notice netifd: Interface 'wan' is now down
Tue Sep 16 11:07:04 2025 daemon.notice netifd: Interface 'wan' is setting up now
Tue Sep 16 11:07:04 2025 daemon.info pppd[22134]: Plugin pppoe.so loaded.`

brada4, 2025-09-16:
Do you get pado reply?

kcduke42, 2025-09-16:
Yes, after a reboot, PADO packets miraculously appear:

```
Tue Sep  2 11:15:02 2025 kern.notice kernel: [    0.000000] Linux version 5.15.162 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24012-d8dd03c46f) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Jul 15 22:14:18 2024
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] MIPS: machine is MERCUSYS MR70X v1
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Sep  2 11:15:02 2025 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Zone ranges:
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Early memory node ranges
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] percpu: Embedded 12 pages/cpu s17808 r8192 d23152 u49152
Tue Sep  2 11:15:02 2025 kern.debug kernel: [    0.000000] pcpu-alloc: s17808 r8192 d23152 u49152 alloc=12*4096
Tue Sep  2 11:15:02 2025 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Tue Sep  2 11:15:02 2025 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Writing ErrCtl register=0004a004
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Readback ErrCtl register=0004a004
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] Memory: 119220K/131072K available (7317K kernel code, 628K rwdata, 884K rodata, 1272K init, 225K bss, 11852K reserved, 0K cma-reserved)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.008045] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.066215] pid_max: default: 32768 minimum: 301
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.071623] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.078838] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.090392] rcu: Hierarchical SRCU implementation.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.095861] smp: Bringing up secondary CPUs ...
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.101244] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.101269] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Sep  2 11:15:02 2025 kern.warn kernel: [    0.101284] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.101329] CPU1 revision is: 0001992f (MIPS 1004Kc)
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.160637] Synchronize counters for CPU 1: done.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.193100] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Sep  2 11:15:02 2025 kern.info kernel: [    0.193123] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... cut ...]
Tue Sep 16 11:45:23 2025 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.178 bc:32:5f:67:e7:7a NVR-0ED7
Tue Sep 16 11:45:27 2025 daemon.err uhttpd[2200]: [info] luci: accepted login on /admin/network/network for root from 192.168.1.171
Tue Sep 16 11:45:41 2025 authpriv.info dropbear[4323]: Child connection from 192.168.1.171:2134
Tue Sep 16 11:45:48 2025 authpriv.notice dropbear[4323]: Password auth succeeded for 'root' from 192.168.1.171:2134
Tue Sep 16 11:45:59 2025 cron.err crond[2103]: time disparity of 20190 minutes detected
```

brada4, 2025-09-17:
Also tcpdump on wan.vlan when restarting pppoe to see if / how PADO packets get sent.

kcduke42, 2025-09-17:
Did a tcpdump of the WAN interface (attached). It only shows PADI. It looks like we need to put a network hub in line and capture the dump from there.
[pppoe1.zip](https://github.com/user-attachments/files/22378207/pppoe1.zip)

brada4, 2025-09-17:
Upgrade to 24.10.2 before exploring further. 23.05 is eol.

You need to mirror port outside system to prove packets on the wire.