openwrt/openwrt #14318: IGMP handling issues on MT762 - broken snooping and loops

albertofustinoni, 2024-01-02:
### Describe the bug

I am Running OpenWrt 23.05.2 on two [Netgear WAX202](https://openwrt.org/toh/netgear/wax202) (and before that on two [Linksys E8450](https://openwrt.org/toh/linksys/e8450)) and encountered issues with IGMP handling. The one thing in common is that both models are `ramips/mt7621` platform.

1. Turning on IGMP snooping crashes the device or otherwise makes it inaccessible and unable to pass traffic. Only way to recover is from this is to reset.
2. Multicast traffic causes the a lot of  `received packet on wan with own address as source address` messages to appear in the system log and some weird connectivity issues, like SSH sessions hanging and being dropped.

I first noticed the kernel warning several months ago on the E8450s and [posted in the forum](https://forum.openwrt.org/t/fixing-received-packet-with-own-address-as-source-address/164840) thinking it was a configuration mistake. I never could figure out what was wrong and ended up thinking it was a hardware issue. A few days ago, I bought the WAX202s, configured them in the same way and was relieved to see no such messages in the system log.

This until I noticed that I had forgotten to set up `umdns` to allow [DAWN](https://openwrt.org/docs/guide-user/network/wifi/dawn#fixing_umdns_openwrt_2102_only) on the new devices to communicate with each other: by default umdns is configured to bind to the `lan` interface, which in my configuration does not exist, so the two DAWN instances could not see each other via mDNS.

As soon as I got umdns to bind to the existing `mgmt` interface, the SSH session crashed. I could still log in to the web UI and at around the same time a batch of `received packet on wan with own address as source address` were generated in the system log.

I am using the two devices as dumb bridged APs with multiple VLANs defined and dynamic VLAN assignment on a single WPA2 Enterprise SSID via RADIUS, which is why I have redundant bridges in my config (DSA defined 802.1q VLANS and an additional software bridge for each of them). I have also configured 802.11k/k/v roaming and use DAWN for band steering and to improve roaming performance. All of this seems to work fine for unicast traffic (VLANs are assigned to clients based on credentials and I VOIP calls don't drop or glitch when moving around).

My network and wireless config is as below (secrets changed):

```
cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd2a:73d9:4d83::/48'
        option packet_steering '1'

config device
        option name 'lan1'
        option macaddr '34:98:b5:18:90:ad'

config device
        option name 'lan2'
        option macaddr '34:98:b5:18:90:ae'

config device
        option name 'lan3'
        option macaddr '34:98:b5:18:90:af'

config device
        option name 'wan'
        option macaddr '34:98:b5:18:90:b0'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'wan'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'wan:u*'

config bridge-vlan
        option device 'br-lan'
        option vlan '10'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'wan:t'

config bridge-vlan
        option device 'br-lan'
        option vlan '20'
[... 200 lines cut ...]
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAX202

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-02:
Is firewall that short? It must have some zone2zone  rules. ???
for more diagnostics
```
sysctl net.ipv4.conf.default.log_martians=1 net.ipv4.conf.all.log_martians=1 # tee -a /etc/sysctl.conf
```
should start barfing in syslog `logread` right away if network is looped outside router.
uncomment 2nd half to save permanently if not.

nothing immeadiately wrong in your config.
you may want to renumber away from vlan1, not everywhere but sometimes it is used intrnally by switch asics

albertofustinoni, 2024-01-03:
Firewall *is* that short: I removed all the zone2zone rules myself. These should be dumb L2 APs, they are not meant to do routing or firewalling.

I've tried adding `sysctl net.ipv4.conf.default.log_martians=1 net.ipv4.conf.all.log_martians=1` as you suggested and then `logread`'s last entries are like

```
Wed Jan  3 10:34:34 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:38 2024 daemon.notice hostapd: Beacon request: c2:f9:15:b4:65:dc is not connected
Wed Jan  3 10:34:38 2024 daemon.warn dawn: Client / BSSID = C2:F9:15:B4:65:DC / 34:98:B5:18:90:B1: BEACON REQUEST failed
Wed Jan  3 10:34:38 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:43 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:47 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:49 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 164 ack=1
Wed Jan  3 10:34:49 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 32:3c:b4:e5:28:97 165 ack=1
Wed Jan  3 10:34:49 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 166 ack=1
Wed Jan  3 10:34:52 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:56 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:34:57 2024 authpriv.info dropbear[12820]: Child connection from 10.0.1.115:55436
Wed Jan  3 10:34:57 2024 authpriv.notice dropbear[12820]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:pu5XkATr3wD+5IpJuJiD+6bQOONVWE0neCOcMYuPt+0 from 10.0.1.115:55436
Wed Jan  3 10:34:58 2024 daemon.notice hostapd: Beacon request: c2:f9:15:b4:65:dc is not connected
Wed Jan  3 10:34:58 2024 daemon.warn dawn: Client / BSSID = C2:F9:15:B4:65:DC / 34:98:B5:18:90:B1: BEACON REQUEST failed
Wed Jan  3 10:35:00 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:05 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:09 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:10 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 32:3c:b4:e5:28:97 167 ack=1
Wed Jan  3 10:35:14 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:18 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:18 2024 daemon.notice hostapd: Beacon request: c2:f9:15:b4:65:dc is not connected
Wed Jan  3 10:35:18 2024 daemon.warn dawn: Client / BSSID = C2:F9:15:B4:65:DC / 34:98:B5:18:90:B1: BEACON REQUEST failed
Wed Jan  3 10:35:22 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:27 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:31 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 168 ack=1
Wed Jan  3 10:35:31 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 32:3c:b4:e5:28:97 169 ack=1
Wed Jan  3 10:35:31 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 170 ack=1
Wed Jan  3 10:35:31 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:36 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:38 2024 daemon.notice hostapd: Beacon request: c2:f9:15:b4:65:dc is not connected
Wed Jan  3 10:35:38 2024 daemon.warn dawn: Client / BSSID = C2:F9:15:B4:65:DC / 34:98:B5:18:90:B1: BEACON REQUEST failed
Wed Jan  3 10:35:40 2024 daemon.notice hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: did not acknowledge authentication response
Wed Jan  3 10:35:45 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:50 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:52 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 32:3c:b4:e5:28:97 171 ack=1
Wed Jan  3 10:35:54 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:59 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:35:59 2024 daemon.notice hostapd: Beacon request: c2:f9:15:b4:65:dc is not connected
Wed Jan  3 10:35:59 2024 daemon.warn dawn: Client / BSSID = C2:F9:15:B4:65:DC / 34:98:B5:18:90:B1: BEACON REQUEST failed
Wed Jan  3 10:36:03 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:36:08 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:36:12 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
Wed Jan  3 10:36:13 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 172 ack=1
Wed Jan  3 10:36:13 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 32:3c:b4:e5:28:97 173 ack=0
Wed Jan  3 10:36:13 2024 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 66:37:c5:28:ae:20 174 ack=1
Wed Jan  3 10:36:16 2024 daemon.info hostapd: phy0-ap1: STA 0c:8b:95:00:e8:64 IEEE 802.11: authenticated
```

so there should be no loop somewhere else right?

brada4, 2024-01-03:
waitt for one  'packet on wan with own address as source address'

brada4, 2024-01-03:
Martian detection is expected to catch mixups outside system like loops, leakss and spoofs. With that indication absent for few own address logs we go look for local bugs of same kind.

albertofustinoni, 2024-01-10:
@brada4 I've been doing as you instructed but no `packet on wan with own address as source address` entries showed up.

Should I enable umdns on the interface again to generate some? What are you interested in me recording?

brada4, 2024-01-10:
And no martian sources either? `logread -e artian`
It was intended to diagnose better, absolutely impossible it fixes anything.

sweroam-mibr, 2024-01-16:
Have issues since https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 have done sone testing...  

Setup
i have a router where lan1 is connected to sw1:47 and lan2 is connected to sw2:47 the sw1 is then connected to sw2. (loop) but stp should fix this, works well if i disable the 2g patch in the DTS.

brada4, 2024-01-16:
Seems ?dsa? makes a side pipe in place of CPU-controllable multicast forwarder.

sweroam-mibr, 2024-01-16:
> Seems ?dsa? makes a side pipe in place of CPU-controllable multicast forwarder.

It seems like that when it mux the gmac0 and gmac1 it bypasses the CPU. @arinc9 i guess you are the one that knows best.

brada4, 2024-01-16:
It should not do it silently, you kind of have firewall with eventual back-offload on CPU.

arinc9, 2024-01-16:
> > Seems ?dsa? makes a side pipe in place of CPU-controllable multicast forwarder.
> 
> It seems like that when it mux the gmac0 and gmac1 it bypass the CPU.

I've got no idea what to make of this quote and the one above.

> ?dsa? makes a side pipe

What's a "side pipe"?

> ?dsa? makes a side pipe in place of CPU-controllable multicast forwarder.

What's "CPU-controllable multicast forwarder"? And what do you mean DSA makes a side pipe in place of it?

> when it mux the gmac0 and gmac1 it bypass the CPU.

Sorry, what is this "it" that bypasses the CPU...

> It should not do it silently, you kind of have firewall with eventual back-offload on CPU.

What is "back-offload"? How is it eventual and how is it in relation to CPU?

Since I've been pinged left, right, and center by sweroam-mibr, I'm going to leave my test result showing that STP works perfectly well with the wan interface that was a DSA user port interface before, and, with the PHY muxing patch, is now a regular network interface.

Device with MT7530 switch:
lan1 interface is connected to p1 interface on the other side.
eth1 (wan) interface is connected to p2 interface on the other side.

lan1 & eth1 interfaces are attached to br0.
stp_state is enabled on br0.
br0 is given 192.168.2.1/24

Device on the other side:
p1 & p2 interfaces are attached to br0.
stp_state is enabled on br0.
br0 is given 192.168.2.2/24

It works as expected:
```
[  807.799124] mtk_soc_eth 1e100000.ethernet eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[  807.807831] br0: port 5(eth1) entered blocking state
[  807.812835] br0: port 5(eth1) entered listening state
[  816.110859] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[  816.118954] br0: port 2(lan1) entered blocking state
[  816.123967] br0: port 2(lan1) entered listening state
[  816.693737] br0: port 5(eth1) entered blocking state
[  831.254900] br0: port 2(lan1) entered learning state
[  846.358898] br0: port 2(lan1) entered forwarding state
# ping 192.168.2.2
PING 192.168.2.2 (192.168.2.2): 56 data bytes
64 bytes from 192.168.2.2: seq=0 ttl=64 time=0.941 ms
64 bytes from 192.168.2.2: seq=1 ttl=64 time=0.487 ms
64 bytes from 192.168.2.2: seq=2 ttl=64 time=0.454 ms
64 bytes from 192.168.2.2: seq=3 ttl=64 time=0.456 ms
64 bytes from 192.168.2.2: seq=4 ttl=64 time=0.382 ms
[  886.389826] mt7530-mdio mdio-bus:1f lan1: Link is Down
[  886.395184] br0: port 2(lan1) entered disabled state
[  886.400320] br0: port 5(eth1) entered listening state
[  901.654894] br0: port 5(eth1) entered learning state
[  916.758909] br0: port 5(eth1) entered forwarding state
64 bytes from 192.168.2.2: seq=35 ttl=64 time=2029.003 ms
64 bytes from 192.168.2.2: seq=36 ttl=64 time=1019.783 ms
64 bytes from 192.168.2.2: seq=37 ttl=64 time=17.150 ms
64 bytes from 192.168.2.2: seq=38 ttl=64 time=0.404 ms
64 bytes from 192.168.2.2: seq=39 ttl=64 time=0.365 ms
64 bytes from 192.168.2.2: seq=40 ttl=64 time=0.273 ms
64 bytes from 192.168.2.2: seq=41 ttl=64 time=0.382 ms
64 bytes from 192.168.2.2: seq=42 ttl=64 time=0.415 ms
64 bytes from 192.168.2.2: seq=43 ttl=64 time=0.273 ms
[  924.792365] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[  924.800657] br0: port 2(lan1) entered blocking state
[  924.805699] br0: port 2(lan1) entered listening state
64 bytes from 192.168.2.2: seq=44 ttl=64 time=10.976 ms
64 bytes from 192.168.2.2: seq=45 ttl=64 time=0.517 ms
[  926.672935] br0: topology change detected, sending tcn bpdu
[  926.678606] br0: port 5(eth1) entered blocking state
[  940.054890] br0: port 2(lan1) entered learning state
[  955.158888] br0: port 2(lan1) entered forwarding state
64 bytes from 192.168.2.2: seq=74 ttl=64 time=1008.800 ms
64 bytes from 192.168.2.2: seq=75 ttl=64 time=4.881 ms
64 bytes from 192.168.2.2: seq=76 ttl=64 time=0.533 ms
64 bytes from 192.168.2.2: seq=77 ttl=64 time=0.443 ms
64 bytes from 192.168.2.2: seq=78 ttl=64 time=0.478 ms
64 bytes from 192.168.2.2: seq=79 ttl=64 time=0.362 ms
[  960.375049] mtk_soc_eth 1e100000.ethernet eth1: Link is Down
[  960.380959] br0: port 5(eth1) entered disabled state
64 bytes from 192.168.2.2: seq=80 ttl=64 time=0.579 ms
64 bytes from 192.168.2.2: seq=81 ttl=64 time=0.555 ms
64 bytes from 192.168.2.2: seq=82 ttl=64 time=0.484 ms
64 bytes from 192.168.2.2: seq=83 ttl=64 time=0.448 ms
64 bytes from 192.168.2.2: seq=84 ttl=64 time=0.439 ms
64 bytes from 192.168.2.2: seq=85 ttl=64 time=0.442 ms
64 bytes from 192.168.2.2: seq=86 ttl=64 time=0.349 ms
64 bytes from 192.168.2.2: seq=87 ttl=64 time=0.445 ms
64 bytes from 192.168.2.2: seq=88 ttl=64 time=0.439 ms
64 bytes from 192.168.2.2: seq=89 ttl=64 time=0.345 ms
64 bytes from 192.168.2.2: seq=90 ttl=64 time=0.438 ms
64 bytes from 192.168.2.2: seq=91 ttl=64 time=0.443 ms
64 bytes from 192.168.2.2: seq=92 ttl=64 time=0.440 ms
[  973.687118] mtk_soc_eth 1e100000.ethernet eth1: Link is Up - 100Mbps/Full - flow control rx/tx
[  973.695841] br0: port 5(eth1) entered blocking state
[  973.700844] br0: port 5(eth1) entered listening state
64 bytes from 192.168.2.2: seq=93 ttl=64 time=0.584 ms
[  974.672445] br0: port 5(eth1) entered blocking state
64 bytes from 192.168.2.2: seq=94 ttl=64 time=0.555 ms
64 bytes from 192.168.2.2: seq=95 ttl=64 time=0.533 ms
64 bytes from 192.168.2.2: seq=96 ttl=64 time=0.459 ms
64 bytes from 192.168.2.2: seq=97 ttl=64 time=0.483 ms
64 bytes from 192.168.2.2: seq=98 ttl=64 time=0.443 ms
64 bytes from 192.168.2.2: seq=99 ttl=64 time=0.452 ms
```

arinc9, 2024-01-16:
@albertofustinoni Why:
```
config device
        option type 'bridge'
        option name 'br-vlan50'
        list ports 'br-lan.50'

config interface 'vlan50'
        option proto 'none'
        option device 'br-vlan50'
```

Why not just:
```
config interface 'vlan50'
        option proto 'none'
        option device 'br-lan.50'
```

brada4, 2024-01-16:
I mean that switch snoops multicast subscriptions and subscribes lan hosts to wan multicast on its own.

sweroam-mibr, 2024-06-10:
Applied this patch, after that spanning tree works again 😀 my loops are gone on mt7621. 

https://lore.kernel.org/netdev/=%3Futf-8%3Fq%3F=3C20240201-b4-for-net-mt7530-fix-link-local-that-ingr%3F=%20=%3Futf-8%3Fq%3Fess-vlan-filtering-ports-v1-1-881c1c96b27f=40arinc9=2Ecom=3E%3F=/

azazaazur, 2026-01-14:
This issue seems to be fixed on 24.10.5 as I have the exact same board (ER605v2) and enabling IGMP snooping actually significantly decreased my CPU usage (possibly because as OP mentioned the loop issue seems to be fixed on kernel 6.6).