openwrt/openwrt #14320: fw4: tcp_ecn option is ignored

timur-davletshin, 2024-01-02:
### Describe the bug

Setting following in /etc/config/firewall (described in https://openwrt.org/docs/guide-user/firewall/firewall_configuration) actually does nothing at least in OpenWrt 23.05.*
```
config defaults
...
        option tcp_ecn '1'
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/geode

### Device

cat: can't open '/tmp/sysinfo/model': No such file or directory

### Image kind

Official downloaded image

### Steps to reproduce

Run: sysctl net.ipv4.tcp_ecn

### Actual behaviour

net.ipv4.tcp_ecn = 2 (kernel default)

### Expected behaviour

net.ipv4.tcp_ecn = 1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-02:
you can use sysctl.conf . fw4 discards that option silently.

timur-davletshin, 2024-01-02:
@brada4 yep, this is what I do. But it would be nice to have it fixed. There is no reason to discard it. It is functional and documented in OpenWrt's wiki.

brada4, 2024-01-02:
I just found in source at ../firewall4 , this is like own IP stack tuning, not changing forwarded flows at all.
Actually there are some net.netfilter options that one might want to tune like number of ct tates.

timur-davletshin, 2024-01-02:
> this is like own IP stack tuning, not changing forwarded flows at all.

I'm are well-aware of that, it has no impact on forwarded packages or SQM. Nevertheless it should be fixed.

> Actually there are some net.netfilter options that one might want to tune like number of ct states.

Not related to the issue.

brada4, 2024-01-02:
Actually none of sysctl-s were implemented by @jow- 3 years ago. Over time most of them were duplicates of per-interface  parameters back then and nobody noticed change for years.

timur-davletshin, 2024-01-02:
Documented but never implemented... well, I should say, that is a bit odd. Maybe it is time to implement it since it's not available for per-interface configuration?

brada4, 2024-01-02:
Patience, lets see if it was deliberate or accidental omission.

jow-, 2024-01-04:
Deliberate omission, does not really belong into the firewall imho. Netifd should gain support, fw4 should habndle those options as unsupported, to emit a warning when present.

timur-davletshin, 2024-01-05:
Well, someone has to remove it from wiki then.