openwrt/openwrt #14321: Luci over HTTPS is incredible slow -> TLS handshake seems the problem

jasmir, 2024-01-02:
### Describe the bug

Luci over HTTPS is incredible slow. It seems to be a Problem with the SSL-Handshake. I have this problem on all of my OpenWRT Routers/APs from different brands. The more powerfull devices [on the CPU side] are less affected:
[image]

Doing some TCP-Dumping, it seems that the gap is between "Server Key Exchange" and "Server Hello Done":
[image]
The uHTTPd is using all of the CPU during this time:
`2450     1 root     R     3260   3%  95% /usr/sbin/uhttpd -f -h /www -r gateway -x /cgi-bin -u /ubus -k 20 -n 5 -N 100 -R -p 0.0.0.0:80 -p [::]:80 -C /etc/uhttpd.crt -K /etc/uhttpd.key -s 0.0.0.0:443 -s [::]:443 -q`

I can reproduce the Problem with Firefox and cURL.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq806x/generic & ramips/mt7621 & ath79/generic

### Device

Linksys EA8500 & ASUS RT-AX53U & GL.iNet GL-X750 & D-Link DAP-X1860 A1

### Image kind

Official downloaded image

### Steps to reproduce

Setting up uHTTPd:
```
config uhttpd 'main'
	list listen_http '0.0.0.0:80'
	list listen_http '[::]:80'
	list listen_https '0.0.0.0:443'
	list listen_https '[::]:443'
	option home '/www'
	option rfc1918_filter '1'
	option max_requests '5'
	option max_connections '100'
	option cert '/etc/uhttpd.crt'
	option key '/etc/uhttpd.key'
	option cgi_prefix '/cgi-bin'
	list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
	option http_keepalive '20'
	option ubus_prefix '/ubus'
	option redirect_https 'on'
```
Copy the needed cert & key to the location (I use 4096 RSA, SHA-256 with RSA Encryption Certs from my own CA)
Start using Luci with Firefox -or- testing the Handshake with cURL:
`time curl -vvvk https://gateway`

### Actual behaviour

```
* Connected to <deleted> (<deleted>) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* TLSv1.0 (OUT), TLS header, Certificate Status (22):
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Certificate (11):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Server finished (14):
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):
* TLSv1.2 (OUT), TLS header, Finished (20):
* TLSv1.2 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Finished (20):
* TLSv1.2 (IN), TLS header, Finished (20):
* TLSv1.2 (IN), TLS header, Certificate Status (22):
* TLSv1.2 (IN), TLS handshake, Finished (20):
* SSL connection using TLSv1.2 / ECDHE-RSA-CHACHA20-POLY1305
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: <deleted>
*  start date: Dec 31 13:31:18 2023 GMT
*  expire date: Dec 30 13:31:18 2026 GMT
*  issuer: <deleted>
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* TLSv1.2 (OUT), TLS header, Supplemental data (23):
> GET / HTTP/1.1
> Host: <deleted>
> User-Agent: curl/7.81.0
> Accept: */*
> 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Connection: Keep-Alive
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< Keep-Alive: timeout=20
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< ETag: "90c-30c-65537843"
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< Last-Modified: Tue, 14 Nov 2023 13:38:11 GMT
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< Date: Tue, 02 Jan 2024 16:25:11 GMT
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< Content-Type: text/html
* TLSv1.2 (IN), TLS header, Supplemental data (23):
< Content-Length: 780
< 
* TLSv1.2 (IN), TLS header, Supplemental data (23):
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.1//EN" "http://www.w3.org/TR/xhtml11/DTD/xhtml11.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
	<head>
		<meta http-equiv="Cache-Control" content="no-cache, no-store, must-revalidate" />
		<meta http-equiv="Pragma" content="no-cache" />
                <meta http-equiv="Expires" content="0" />
		<meta http-equiv="refresh" content="0; URL=cgi-bin/luci/" />
		<style type="text/css">
			body { background: white; font-family: arial, helvetica, sans-serif; }
			a { color: black; }

			@media (prefers-color-scheme: dark) {
				body { background: black; }
				a { color: white; }
			}
		</style>
	</head>
	<body>
		<a href="cgi-bin/luci/">LuCI - Lua Configuration Interface</a>
	</body>
</html>
* Connection #0 to host <deleted> left intact

real	0m8,832s
user	0m0,003s
sys	0m0,005s
```

### Expected behaviour

Something more like 0,1sec. for this request.

### Additional info

I use a image from the Firmware-Selector with customized packets:

[Example from the GL.iNet GL-X750]
- removed wpad-basic-mbedtls from the packages
- added packages: wpad-wolfssl luci muninlite kmod-usb-serial kmod-usb-net kmod-usb-serial-wwan kmod-usb-serial-option kmod-usb-net-qmi-wwan kmod-usb-net-cdc-mbim luci-proto-modemmanager luci-app-wireguard luci-app-adblock-fast iperf3 coreutils-sleep luci-app-watchcat tcpdump python3-light python3-multiprocessing python3-distutils python3-logging openssh-sftp-server luci-app-statistics

It differs slightly for the other APs, but on all devices I replace wpad-basic-mbedtls with wpad-wolfssl

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jasmir, 2024-01-02:
I've reported _all_ targets which I have tested.

kirdesde, 2024-01-02:
Reason is your 4096 RSA key. That needs a lot of cpu power.
OpenWrt default key is ecdsa and is less cpu hungry.

neheb, 2024-01-02:
Yeah that's going to be slow as molasses no matter the TLS library.

closed 2024-01-03 by ynezz