openwrt/openwrt #14339: [libiwinfo] per_sta_vif causes assoclist to not function as expected

rany2, 2024-01-05:
### Describe the bug

If `per_sta_vif` is enabled, I need to go to every individual STA's vif and run `assoclist` against it. So for example, consider `psk5ghz` interface which has `per_sta_vif` enabled in hostapd config. Running `iwinfo psk5ghz assoclist` returns "No station connected" but what I expect is for I go through all the STA vifs hostapd created and run assoclist against it.

### OpenWrt version

r24742-6be6cd64c4

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

N/A

### Device

N/A

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-05:
do you get proper result via
`iw dev phy1-ap0 station dump`
?

rany2, 2024-01-05:
@brada4 nope, but `iw dev phy1-ap0.4097 station dump` (for example) works properly. What I expected `iwinfo` would do is to match these as well when providing me with assoclist.

rany2, 2024-01-05:
I understand if this gets marked as WONTFIX, but this also breaks luci's wifi status page as it ends up showing no clients are connected to AP.

rany2, 2024-01-05:
I submitted a PR here: https://github.com/openwrt/iwinfo/pull/5