openwrt/openwrt #14358: killing wireless extensions break reaver and hashcat

bol-van, 2024-01-07:
### Describe the bug

In openwrt 23.05 wireless extensions were disabled.
I don't know how much space does it save. Don't think too much.
But this change break  reaver,wash and hcxdumptool.
They rely on ioctls from wireless extensions

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

Netgear WNDR3800

### Image kind

Self-built image

### Steps to reproduce

install hcxdumptool and try to use it

### Actual behaviour

not working. displays error

### Expected behaviour

should work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2024-01-07:
I took a quick look at the code of these packages:
- hcxdumptool doesn't use wireless extensions, so it seems unrelated.
- reaver/wash only uses a small part (for setting frequency).

Wireless Extensions have been deprecated upstream for a long time, and for some newer drivers, the kernel will refuse to use them due to many limitations. I don't think it makes any sense to bring them back for these packages. Somebody should just fix reaver/wash instead.

closed 2024-01-07 by nbd168

bol-van, 2024-01-07:
Stiill, hcxdump tool does not work on openwrt 23.05 unless manually recompile with wext enabled.
I checked this

bol-van, 2024-01-07:
re-checked. hcxdumptool calls : ioctl(3, SIOCSIWFREQ

rany2, 2024-01-07:
@bol-van Try updating to hcxdumptool 6.3.0 from snapshot, the version in 23.05 is old and still has some WEXT code; https://github.com/openwrt/packages/blob/openwrt-23.05/net/hcxdumptool/Makefile