openwrt/openwrt #14359: unable to get ipv6 wireguarded on lan with proton

lm2lm2, 2024-01-07:
### Describe the bug

doing this :

getting this :
https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6

Warning: Unable to locate ipset utility, disabling ipset support
Warning: Option @zone[1].masq6 is unknown
Warning: Section @zone[1] (wan) cannot resolve device of network 'wan'
Warning: Section @zone[1] (wan) cannot resolve device of network 'wan6'
Warning: Option @zone[2].masq6 is unknown

### OpenWrt version

26

### OpenWrt release

26

### OpenWrt target/subtarget

broadcom

### Device

router

### Image kind

Self-built image

### Steps to reproduce

lsmod
b43                   363699  0 
cfg80211              235099  2 b43,mac80211
compat                  6615  3 b43,mac80211,cfg80211
gpio_button_hotplug     7088  0 
ip_tables              10157  4 iptable_nat,iptable_mangle,iptable_filter
ip6_tables              9953 24 ip6table_mangle,ip6table_filter
ip6_udp_tunnel          1687  1 wireguard
ip6t_REJECT             1072  2 
ip6table_filter          752  1 
ip6table_mangle         1264  1 
ipt_ECN                 1520  0 
ipt_MASQUERADE           752  1 
ipt_REJECT              1072  2 
iptable_filter           752  1 
iptable_mangle          1008  1 
iptable_nat              816  1 
mac80211              473098  1 b43
nf_conntrack           58117 13 nf_conntrack_ipv6,ipt_MASQUERADE,xt_state,xt_nat,xt_conntrack,xt_REDIRECT,xt_NETMAP,nf_nat_masquerade_ipv4,nf_conntrack_ipv4,nf_nat_ipv4,nf_nat,nf_flow_table,nf_conntrack_rtcache
nf_conntrack_ipv4       4400 13 
nf_conntrack_ipv6       4672  5 
nf_conntrack_rtcache    2672  0 
nf_defrag_ipv4          1110  1 nf_conntrack_ipv4
nf_defrag_ipv6          5054  1 nf_conntrack_ipv6
nf_flow_table          13983  2 xt_FLOWOFFLOAD,nf_flow_table_hw
nf_flow_table_hw        2160  1 
nf_log_common           2671  2 nf_log_ipv4,nf_log_ipv6
nf_log_ipv4             3440  0 
nf_log_ipv6             4144  0 
nf_nat                  9515  5 xt_nat,xt_NETMAP,nf_nat_redirect,nf_nat_masquerade_ipv4,nf_nat_ipv4
nf_nat_ipv4             3377  1 iptable_nat
nf_nat_masquerade_ipv4    1788  1 ipt_MASQUERADE
nf_nat_redirect         1563  1 xt_REDIRECT
nf_reject_ipv4          2179  1 ipt_REJECT
nf_reject_ipv6          2632  1 ip6t_REJECT
udp_tunnel              2863  1 wireguard
wireguard             113643  0 
x_tables               14127 34 ipt_REJECT,ipt_MASQUERADE,xt_time,xt_tcpudp,xt_tcpmss,xt_statistic,xt_state,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_length,xt_hl,xt_ecn,xt_dscp,xt_conntrack,xt_comment,xt_TCPMSS,xt_REDIRECT,xt_NETMAP,xt_LOG,xt_HL,xt_FLOWOFFLOAD,xt_DSCP,xt_CLASSIFY,iptable_mangle,iptable_filter,ipt_ECN,ip_tables,ip6table_mangle,ip6table_filter,ip6_tables,ip6t_REJECT
xt_CLASSIFY              688  0 
xt_DSCP                 1584  0 
xt_FLOWOFFLOAD          3136  0 
xt_HL                   1392  0 
xt_LOG                   880  0 
xt_NETMAP               1648  0 
xt_REDIRECT              816  0 
xt_TCPMSS               2864  8 
xt_comment               560163 
xt_conntrack            2416 16 
xt_dscp                 1072  0 
xt_ecn                  1456  0 
xt_hl                    880  0 
xt_length                816  0 
xt_limit                1392 20 
xt_mac                   688  0 
xt_mark                  752  0 
xt_multiport            1328  0 
xt_nat                  1648  0 
xt_state                 816  0 
xt_statistic             944  0 
xt_tcpmss               1136  0 
xt_tcpudp               1840 16 
xt_time                 1840  0 

### Additional info

need help to put openwrt's wireguard's protonvpn's ipv6 deployed on LAN

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-01-07:
What's OpenWRT 26? Anyway, this was fixed in this commit: https://github.com/openwrt/firewall4/commit/16a10706f24393b11500408e586b495073622e85

brada4, 2024-01-07:
>  nf_conntrack_rtcache  nf_flow_table 13983 2 xt_FLOWOFFLOAD,nf_flow_table_hw

You are certainly on an unsupported fork.