openwrt/openwrt #14365: Routing/NAT Offloading with PPPOE-WAN incorrect setup with nft

KrazyIvan-, 2024-01-08:
### Describe the bug

Routing/NAT Offloading with PPPOE-WAN incorrect setup with nft which lead to decrease speed to 200Mb/s (even you have 1G/s)
any setting as Packet Steering/irqbalace does not help. (4 CPU core does not solve the problem, CPU usage are still below 1%)

By default rules is negerate by openwrt is as: 
flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, phy0-ap0, phy1-ap0, wan }
		flags offload
		counter
	}

which is incorrect when using pppoe-wan (and 'wan' must not be include in devices else still get 200Mb/s max)

correct way is: 
flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan }  #wan should not be add!!!,  pppoe-wan must
		flags offload
		counter
	}
now you can achieve full 1Gb/s

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ipq40xx/generic

### Device

MobiPromo CM520-79F

### Image kind

Official downloaded image

### Steps to reproduce

1) create wan as pppoe-wan
config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.2.254'
	option netmask '255.255.255.0'
	option ip6assign '64'
	list ip6class 'local'

config interface 'wan'
	option device 'wan'
	option proto 'pppoe'
	option username 'user'
	option password 'password'
	option ipv6 '1'

config interface 'wan6'
	option device '@wan'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix 'auto'
	option sourcefilter '0'
	option norelease '1'

2) enable Routing/NAT Offloading , check Software flow offloading (Hardware flow offloading does not matter)

cheack for: nft list table inet fw4
you end up with:
flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, phy0-ap0, phy1-ap0, wan }
		flags offload
		counter
	}

test your speed to internet. max you will get 200-250Mb/s ( make sure you have more then 500Mb/s actual line)

### Actual behaviour

flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, phy0-ap0, phy1-ap0, wan }
		flags offload
		counter
	}

### Expected behaviour

flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan }  #wan should not be add!!!
		flags offload
		counter
	}

### Additional info

work aroud:
change /etc/config/firewall
from
config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	list network 'wan' 
	list network 'wan6'

to:
config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
#	list network 'wan'  #remove this line
	list device 'pppoe-wan' # add this
	list network 'wan6'

now devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan }  will be generate correctly

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-08:
Is wan a dsa port like wan@eth0 in `ip link` ?

KrazyIvan-, 2024-01-08:
yes

ip link 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f0 brd ff:ff:ff:ff:ff:ff
3: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f0 brd ff:ff:ff:ff:ff:ff
4: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f0 brd ff:ff:ff:ff:ff:ff
5: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f1 brd ff:ff:ff:ff:ff:ff
8: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f0 brd ff:ff:ff:ff:ff:ff
9: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1408 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/none 
10: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f0 brd ff:ff:ff:ff:ff:ff
11: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN mode DEFAULT group default qlen 3
    link/ppp 
12: phy1-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP mode DEFAULT group default qlen 1000
    link/ether ca:79:f8:00:b0:f1 brd ff:ff:ff:ff:ff:ff

brada4, 2024-01-08:
You identified 2 issues
Issue 1 (duplicate of, solution is in particular post - override flow offload with one between br-lan and WAN where real packet conversion takes place, i'd enhance it with variables from fw4 print)
https://github.com/openwrt/openwrt/issues/10224#issuecomment-1820224768
You can add following measurements with [waveform test](https://www.waveform.com/tools/bufferbloat):
Note your router name and tests:
- If possible directly connected
- no offload
- sw offload
- hw offload
- custom offload between pppoe-wan and br-lan only

Issue 2 You are correct, firewall-wise WAN interface is PPPoE, the physical WAN interface works by default permit output and needs no other services than single PPPoE session. Maybe @jow- can have his say here around netifd.

KrazyIvan-, 2024-01-08:
test I done on over LAN 1G 
PC lan to lan1 (openwrt) (960MB/s) (offload state does not matter)
PC lan to pppoe-wan (no offload 180Mb/s)
PC lan to pppoe-wan (currrent offload with wan devices = { lan1, lan2, phy0-ap0, phy1-ap0, wan }   200-210Mb/s)
PC lan to pppoe-wan (with offload  ( devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan }) 560Mb/s) (my ISP is 500M)

Packet Steering does not work its always via CPU0 (CPU load below 1%  other 3 never used) in all 4 test above
irqbalance work but no effect on speed (CPU load below 1%, now all 4 CPU are used)  in all 4 test above

[https://github.com/openwrt/openwrt/issues/10224#issuecomment-1821243089](url)

its work, not best, get around 490Mb/s, (and is correct for kernel Since Linux kernel 5.13+, but somehow is real test is not optimal way)
```
 flowtable ft {
                hook ingress priority filter
                devices = { br-lan, phy0-ap0, phy1-ap0, pppoe-wan, wg, wg1 }
                counter
        }
```

 but with 
` devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan }) `
you can get 550Mbis more

(even exclude  phy0-ap0, phy1-ap0)  in all test does not effect since test is via LAN. 
But having both  pppoe-wan and  wan togather
` devices = { lan1, lan2, phy0-ap0, phy1-ap0, pppoe-wan, wan }) `
also only can get about 220Mb/s

Since Linux kernel 5.13, you can add bridge ports to the flowtable. The flowtable infrastructure discovers the topology behind the bridge device. This allows the flowtable to define a fastpath bypass between the bridge ports 

Since Linux kernel 5.13, the flowtable infrastructure discovers the real netdevice behind VLAN and PPPoE netdevices. The flowtable software datapath parses the VLAN and PPPoE layer 2 headers to extract the ethertype and the VLAN ID / PPPoE session ID which are used for the flowtable lookups. The flowtable datapath also deals with layer 2 decapsulation.

about wg0 it is never add automatic regardless all settings i try, currently only way is manually add.
list device 'wg0'  or  list network 'wg0'  (both way not add it to flowtable)

brada4, 2024-01-08:
DSA devices (whatever@eth0 and that eth0) do not offload nft flows.
wifi offload prevents roaming as offload states will forward through old devices for 30s
```
net.netfilter.nf_flowtable_tcp_timeout = 30
net.netfilter.nf_flowtable_udp_timeout = 30
```
also sabotages ARP for network teaming, stp or fast hands at the cable.

Down that thread you can split flow pickup between various interfaces, those never pick up never offload anything, their presence in offload list is indifferent.

jow-, 2024-01-08:
Duplicate of #10224

closed 2024-01-08 by jow-