openwrt/openwrt #14367: MediaTek MT7603E: TX power significantly degraded in latest shapshots

csharper2005, 2024-01-08:
### Describe the bug

MediaTek MT7603E TX power significantly degraded in latest snapshots:
23.05.2: 22 dBm
SNAPSHOT: 19 dBm

[image]

### OpenWrt version

r24750-f1de1a090f

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

Beeline SmartBox TURBO+

### Image kind

Official downloaded image

### Steps to reproduce

1. Install OpenWrt 23.05.2. Make sure that `iwinfo` shows normal TX power (20 dBm or more) for MediaTek MT7603E.
2. Install OpenWrt SNAPSHOT. Make sure that `iwinfo` shows low TX power (less than 20 dBm) for MediaTek MT7603E.

### Actual behaviour

MediaTek MT7603E TX power (19 dBm) is below HW and regulatory restrictions.

### Expected behaviour

MediaTek MT7603E TX power is 22 dBm (as well as in OpenWrt 23.05.2).

### Additional info

```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x1fe
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (22.0 dBm)
                        * 2417 MHz [2] (22.0 dBm)
                        * 2422 MHz [3] (22.0 dBm)
                        * 2427 MHz [4] (22.0 dBm)
                        * 2432 MHz [5] (22.0 dBm)
                        * 2437 MHz [6] (22.0 dBm)
                        * 2442 MHz [7] (22.0 dBm)
                        * 2447 MHz [8] (22.0 dBm)
                        * 2452 MHz [9] (22.0 dBm)
                        * 2457 MHz [10] (22.0 dBm)
                        * 2462 MHz [11] (22.0 dBm)
                        * 2467 MHz [12] (22.0 dBm)
                        * 2472 MHz [13] (22.0 dBm)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Supported extended features:
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
                * [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ DEL_IBSS_STA ]: deletion of IBSS station support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```

```
iwinfo
phy0-ap0  ESSID: "OpenWrt-2g"
          Access Point: 14:xx:xx:xx:xx:xx
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 19 dBm  Link Quality: 49/70
          Signal: -61 dBm  Noise: unknown
          Bit Rate: 72.2 MBit/s
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

```
iw reg get
global
country PA: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 36), (N/A)
        (5150 - 5250 @ 80), (N/A, 36), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5470 - 5725 @ 160), (N/A, 30), (N/A)
        (5725 - 5850 @ 80), (N/A, 36), (N/A)
        (57000 - 64000 @ 2160), (N/A, 43), (N/A)
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2024-01-08:
Mine was 20 dBm and now it's 17 dBm on 4A Gigabit with 7603E.

brada4, 2024-01-08:
Router has 3dBm antenna gain programmed in calibration data.
As per link from [wikipedia](https://en.wikipedia.org/wiki/List_of_WLAN_channels#cite_note-34) it is lawful requirement to discount antenna gain and limit signal:
[image]
Probably bug is that 3dBm antenna gain compensation action is not shown anywhere.

brada4, 2024-02-03:
Fixed antenna gain is correctly retracted from tx power to limit radiation. No bug, just improved compliance with regulations.

csharper2005, 2024-02-03:
> Fixed antenna gain is correctly retracted from tx power to limit radiation. No bug, just improved compliance with regulations.

Interesting hypothesis. But it doesn't explain:
1. Why "compliance improved" only for 7603e;
2. Why we have only 19 dBm with PA country.

It was also "not a bug" 😉
https://github.com/openwrt/openwrt/issues/14238

brada4, 2024-02-03:
All EU has same, it is cosmetic problem that caldata adjusment is not presented in iw or luci or anywhere. It is real annoyance losing 1/3 of range with upgrade but law is law.

csharper2005, 2024-02-03:
> All EU has same, it is cosmetic problem that caldata adjusment is not presented in iw or luci or anywhere. It is real annoyance losing 1/3 of range with upgrade but law is law.

PA allows 30 dBm and even more. So, we should get 22 dBm with PA (max HW TX power). Not 19 dBm. Thus, this is a bug.

DragonBluep, 2024-02-03:
The latest mt76 seems to have some fix for this issue.
https://github.com/openwrt/mt76/commit/3e81173d9e2b3345beeb085d1e4abf545ef6c03c

brada4, 2024-02-03:
It is unfair to discount antenna everywhere, but it needs to be noted in upstream regdb if radiation or circuit power gets regulated.

csharper2005, 2024-02-03:
> The latest mt76 seems to have some fix for this issue. [openwrt/mt76@3e81173](https://github.com/openwrt/mt76/commit/3e81173d9e2b3345beeb085d1e4abf545ef6c03c)

Good news. I'm looking forward to this commit in main.

csharper2005, 2024-02-04:
@DragonBluep, indeed. Resolved in `OpenWrt SNAPSHOT, r25000-79c79d9809`.

closed 2024-02-04 by csharper2005