openwrt/openwrt #14371: ASUS RT-AX53U - Kernel Panic at first boot - "not syncing: VFS: Unable to mount root fs on unkno  wn-block(0,0)"

freaked1234, 2024-01-09:
### Describe the bug

Got router first few tries via SSH, always got bootloop
Went TFTP route and tried to install several openwrt versions as per wiki
ASUS Bootloader 1.0.0.1

### OpenWrt version

23.05.0 (and snap) 23.03.5

### OpenWrt release

not sure

### OpenWrt target/subtarget

not sure

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

First install FW via SSH or TFTP on new device

### Additional info

`  *** U-Boot Boot Menu ***

     1. Startup system (Default)
     2. Upgrade firmware
     3. Upgrade bootloader
     4. Upgrade bootloader (advanced mode)
     5. Load image
     0. U-Boot console

  Press UP/DOWN to move, ENTER to select
Reading from 0x1e0000, size 0x100000
Succeeded

RT-AX53U bootloader version: 1.0.0.1
Reading from 0x1e0000, size 0x100000
Succeeded
MAC Address: C8:7F:54:E3:E5:B0
Reading from 0x1e0000, size 0x40000
Succeeded
Check Factory's EEPROM successfully! (r = 17)
Reading from 0x2e0000, size 0x40000
Succeeded
Check Factory2's EEPROM successfully! (r = 17)
Reading from 0x3e0000, size 0x2000
Succeeded
Reading from 0x3e0000, size 0x298a02
Succeeded
   Image Name:   MIPS OpenWrt Linux-5.10.176
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2722242 Bytes = 2.6 MiB
   Load Address: 80001000
   Entry Point:  80001000
   Verifying Checksum ... OK
## Booting kernel from Legacy Image at 83000000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.176
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2722242 Bytes = 2.6 MiB
   Load Address: 80001000
[... 231 lines cut ...]
[    5.140399]  (driver?)
[    5.146925] 1f01             384 mtdblock1
[    5.146929]  (driver?)
[    5.153459] 1f02            1024 mtdblock2
[    5.153464]  (driver?)
[    5.159969] 1f03            1024 mtdblock3
[    5.159973]  (driver?)
[    5.166505] 1f04            1024 mtdblock4
[    5.166510]  (driver?)
[    5.173059] 1f05           51200 mtdblock5
[    5.173064]  (driver?)
[    5.179569] 1f06            4096 mtdblock6
[    5.179573]  (driver?)
[    5.186104] 1f07           47104 mtdblock7
[    5.186109]  (driver?)
[    5.192648] 1f08           51200 mtdblock8
[    5.192653]  (driver?)
[    5.199158] 1f09           16512 mtdblock9
[    5.199162]  (driver?)
[    5.205696] Kernel panic - not syncing: VFS: Unable to mount root fs on unkno  wn-block(0,0)
[    5.213955] Rebooting in 1 seconds..`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-09:
This seems simplest (first in info page is usually easiest)
https://openwrt.org/toh/asus/rt-ax53u#installation_with_mtd-write
Worked for me last summer, and no wan port glitches ever.

freaked1234, 2024-01-09:
> This seems simplest (first in info page is usually easiest) https://openwrt.org/toh/asus/rt-ax53u#installation_with_mtd-write Worked for me last summer, and no wan port glitches ever.

Sorry mate, I don't know if I can follow you. I did everything according to the page you linked, step by step.
At first I flashed via SSH and then via FTP, trying different FW versions, everything resulting in the same boot loop.
Maybe bootloader changed in latest revisions?

rany2, 2024-01-09:
@freaked1234 I checked my RT-AX53U and I can't boot up with 23.05 or SNAPSHOT with the stock u-boot either. I didn't have this issue previously in 22.03.

brada4, 2024-01-09:
Install guide clearly says in step 5 to power off device for first boot.

freaked1234, 2024-01-09:
update:
flashing 22.03.0 via TFTP boots once but only once and then results in the same kernel panic bootloop. Flash --> good boot --> reboot without any changes --> bootloop

flashing 22.03.0, then doing sysupgrade to 22.05 results in kernel panic bootloop too

rany2, 2024-01-09:
@freaked1234 trying to flash via TFTP and run firstboot.

brada4, 2024-01-09:
Just sysupgrade with config reset if you get snything ssh or luci.
22 and 23 has different incompatible switch configuration. You must reset both ways.

brada4, 2024-01-09:
Can you extract /dev/mtd0 when partial openwrt boots and confirm you have same uboot version?

freaked1234, 2024-01-09:
I found this thread:

https://github.com/openwrt/openwrt/issues/12674

which sounds like the exact same issue I am having

So I flashed the factory file from @DragonBluep post via TFTP, which let me boot ONCE again
Then I flashed it again and applied the sysupgrade file included via luci but after successfully upgrading i get send right back into kernel panic bootloop, the same as with all the other versions and combinations I tried
I have no idea what to do with the kernel file thats included?!

Any help dearly appreciated!

@brada4 I will try this now and edit the post within the next few min

brada4, 2024-01-09:
ME:  U-Boot SPL 2018.09 (Feb 24 2021 - 10:45:40 +0800)
YOU: U-Boot SPL 2018.09 (Sep 08 2021 - 14:31:58 +0800)
Seems manufacturing timestamp is set inside uboot. Or it means build dates?
[my-mtd0.gz](https://github.com/openwrt/openwrt/files/13877763/my-mtd0.gz)
Somebody smarter needs to compare images. They are mostly same.

brada4, 2024-01-09:
Try to do factory recovery with asus tool. Then check mtd0 timestamp. 
My installation process was flawless, waited 15 minutes before pulling plug at step 5, no serial cable involved.

brada4, 2024-01-09:
23.05.2 and 22.03.6 should have fix (anything after end of may) as per other linked.

freaked1234, 2024-01-12:
@brada4 yeah but I tried flashing them several times fresh after a asus-firmware-repair, via SSH and TFTP

OK somehow I got it working but I still think someone from the dev-team should look into this issue, since I am very sure that I did everything exactly as it should be. I could try to replicate stuff for you if needed but else I´ll stay away from updates for now.

What I did:

- Asus firmware repair tool
- Flash 22.03.5 via TFTP
- Same bootloop as before
- Flash the kernel file from the prev mentioned github post
- Boots into failsafe mode
- Flash the FW-upgrade from the github post via luci
- works stable for 3 days now, reboots and all

brada4, 2024-01-12:
The problem was with uninitialised flash that gave weird response codes on never touched blocks being initialised by ubifs. problem is fixed with 22.03.**6** and 23.05.**2** (and the single-use firmware provided over there)
Just bad luck programming exactly pre-fix version, and my luck I actually updated factory firmware twice before, so I was much less likely to hit any uninitialised blocks.
Maybe somebody can up example in wiki to start with better version.