openwrt/openwrt #14374: Large MTU AP

buvarbeno, 2024-01-09:
### Describe the bug

I set up an Asus ax53u with multiple SSID on 2.4Ghz. One of the sub AP has MTU setting 1560 Byte to pass trough EoIPV6 tunnels to a Mikrotik wireless client with full 1500 Byte MTU inside the tunnels. When I change some settings on radio, for example change channel, or I turn down and up the radio0 (to schedule auto channel selection when the channel is in Auto), the MTU of the sub AP fall back to 1500 Byte. It can be fixed with reboot, or with "/etc/init.d/network restart", but takes long time.

### OpenWrt version

r24706-4e8c9cebe5

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

/sbin/wifi down radio0
/sbin/wifi up radio0

### Actual behaviour

Tha max MTU trough sub AP with larger then 1500  fall back to 1500

### Expected behaviour

Dont change MTU

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-01-09:
I've faced this issue as well a while back and worked around this by adding the following file into `/etc/hotplug.d/net/01-maxwlanmtu`: https://gist.github.com/rany2/3b0a4c225977813a8847a89358e2f4de

buvarbeno, 2024-01-09:
It seems to work! Thank you!

brada4, 2024-01-09:
Duplicate of https://github.com/openwrt/openwrt/issues/12805
there was other thing to reduce AP MTU to pass correct MSS to "the internet" , but the idea is same - hostapd sets mtu 1500 no matter what is intended otherwise.