openwrt/openwrt #14375: Bug: Multi SSID not working

nightcore500, 2024-01-09:
### Describe the bug

The problem that has been described many times regarding a bug when using a multi ssid setup is unfortunately still present in OpenWRT 23.05.2.
I have a test device here (Banana Pi R3) which has two wifi radio modules (2.4G MT7975N, 5G MT7975P).
According to the current configuration, the device should send out 2 SSIDs per module. It also seems to do this. However, clients only find these SSIDs very randomly.
OpenWRT seems to use the same BSSIDs (MACs) for the SSIDs which leads to this problem.
The current situation is as follows:
2.4G:
SSID: OpenWrt-BPI | Mode: Master | BSSID: D2:98:D9:DB:34:2A
SSID: S2 | Mode: Master | BSSID: D6:98:D9:DB:34:2A
5G:
SSID: OpenWrt-BPI | Mode: Master | BSSID: D6:98:D9:DB:34:2A 
SSID: S2 | Mode: Master | BSSID: D2:98:D9:DB:34:2A

The SSID OpenWrt-BPI in 2.4 G therefore uses the same BSSID as the SSID S2 in 5G.
The same applies to the SSID S2 in 2.4G with the SSID OpenWrt-BPI in 5G.

The result: Clients display the SSIDs very randomly. Sometimes the SSID is missing, sometimes this SSID.
However, the two SSIDs that use the same BSSID are never displayed at the same time.

The tip in the other bug reports to remove all entries with "option macaddr" in the file /etc/config/wireless is unfortunately not applicable here as these entries do not exist in my case.
It's also not really the right solution.

The problem can only be solved by either manually configuring a different MAC for the second SSID "S2" on 2.4G and 5G or by configuring the SSID using "randomly generated".

The problem has already been discussed here:

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

Configure two SSIDs (or more) in the 2.4G and in the 5G module.

### Actual behaviour

2.4G:
SSID: OpenWrt-BPI | Mode: Master | BSSID: D2:98:D9:DB:34:2A
SSID: S2 | Mode: Master | BSSID: D6:98:D9:DB:34:2A
5G:
SSID: OpenWrt-BPI | Mode: Master | BSSID: D6:98:D9:DB:34:2A 
SSID: S2 | Mode: Master | BSSID: D2:98:D9:DB:34:2A

The SSID OpenWrt-BPI in 2.4 G therefore uses the same BSSID as the SSID S2 in 5G.
The same applies to the SSID S2 in 2.4G with the SSID OpenWrt-BPI in 5G.

### Expected behaviour

An unused BSSID (MAC) should be used for each SSID.

### Diffconfig

```text
[... 507 lines cut ...]
CONFIG_PACKAGE_bnx2-firmware=m
CONFIG_PACKAGE_bnx2x-firmware=m
CONFIG_PACKAGE_brcmfmac-firmware-usb=m
CONFIG_PACKAGE_carl9170-firmware=m
CONFIG_PACKAGE_e100-firmware=m
CONFIG_PACKAGE_mwifiex-pcie-firmware=m
CONFIG_PACKAGE_mwl8k-firmware=m
CONFIG_PACKAGE_r8152-firmware=m
CONFIG_PACKAGE_r8169-firmware=m
CONFIG_PACKAGE_rs9113-firmware=m
CONFIG_PACKAGE_rt2800-pci-firmware=m
CONFIG_PACKAGE_rt2800-usb-firmware=m
CONFIG_PACKAGE_rtl8192ce-firmware=m
CONFIG_PACKAGE_rtl8192cu-firmware=m
CONFIG_PACKAGE_rtl8192de-firmware=m
CONFIG_PACKAGE_rtl8192se-firmware=m
CONFIG_PACKAGE_rtl8821ae-firmware=m
CONFIG_PACKAGE_wil6210-firmware=m
CONFIG_PACKAGE_wl12xx-firmware=m
CONFIG_PACKAGE_wl18xx-firmware=m
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

nbd168, 2024-01-09:
Fixed in 38bec08e87b69212daf34eac7b26d643a92d9353 / 4103958e768d8bb0d8c640f48341838d8bf5724a

closed 2024-01-09 by nbd168

nightcore500, 2024-01-09:
@nbd168 Wow, that was fast 👍