openwrt/openwrt #14377: Kernel oops in ledtrig-netdev

manuelm, 2024-01-09:
### Describe the bug

After starting a macvlan (or ipvlan, does not matter) docker container _the second_ time I'm observing the following oops:

```
kern.info kernel: [  819.177307] eth0: renamed from veth106ad63
kern.alert kernel: [  819.213135] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000038
kern.alert kernel: [  819.214006] Mem abort info:
kern.alert kernel: [  819.214284]   ESR = 0x0000000096000006
kern.alert kernel: [  819.214647]   EC = 0x25: DABT (current EL), IL = 32 bits
kern.alert kernel: [  819.215149]   SET = 0, FnV = 0
kern.alert kernel: [  819.215478]   EA = 0, S1PTW = 0
kern.alert kernel: [  819.215793]   FSC = 0x06: level 2 translation fault
kern.alert kernel: [  819.216254] Data abort info:
kern.alert kernel: [  819.216534]   ISV = 0, ISS = 0x00000006
kern.alert kernel: [  819.216900]   CM = 0, WnR = 0
kern.alert kernel: [  819.217190] user pgtable: 4k pages, 48-bit VAs, pgdp=000000000570e000
kern.alert kernel: [  819.217794] [0000000000000038] pgd=0800000006cfb003, p4d=0800000006cfb003, pud=08000000053c4003, pmd=0000000000000000
kern.emerg kernel: [  819.218811] Internal error: Oops: 0000000096000006 [#1] PREEMPT SMP
kern.warn kernel: [  819.219400] Modules linked in: nft_fib_inet nf_flow_table_inet wireguard nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables libchacha20poly1305 ipt_REJECT chacha_neon zstd xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_CT r8169 poly1305_neon nfnetlink nf_reject_ipv4 nf_log_syslog nf_flow_table macvlan lzo_rle lzo libcurve25519_generic libchacha ipvlan iptable_nat iptable_mangle iptable_filter ip_tables br_netfilter ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel veth kpp crypto_acompress gpio_button_hotplug btrfs xor zstd_decompress
kern.warn kernel: [  819.220153]  zstd_compress zstd_common xor_neon raid6_pq lzo_decompress lzo_compress
kern.warn kernel: [  819.228692] CPU: 0 PID: 2290 Comm: dockerd Not tainted 6.1.71 #0
kern.warn kernel: [  819.229265] Hardware name: FriendlyElec NanoPi R5C (DT)
kern.warn kernel: [  819.229758] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
kern.warn kernel: [  819.230415] pc : get_device_state+0x18/0x70
kern.warn kernel: [  819.230839] lr : netdev_trig_notify+0xb0/0x260
kern.warn kernel: [  819.231271] sp : ffff80000a0bb520
kern.warn kernel: [  819.231592] x29: ffff80000a0bb520 x28: dead000000000122 x27: ffff000000718550
kern.warn kernel: [  819.232282] x26: 0000000000000003 x25: ffff80000900cf40 x24: ffff000006c1b0c0
kern.warn kernel: [  819.232968] x23: ffff80000900ce18 x22: ffff000000718000 x21: ffff00000578bf00
kern.warn kernel: [  819.233654] x20: ffff00000578bf78 x19: ffff00000578bf00 x18: 0000000000000000
kern.warn kernel: [  819.234340] x17: 0000000000000000 x16: 0000000000000000 x15: 0000ffffb53489b8
kern.warn kernel: [  819.235027] x14: ffffffffffffffff x13: 0000000000000020 x12: 0101010101010101
kern.warn kernel: [  819.235713] x11: 7f7f7f7f7f7f7f7f x10: 0000000005073fa0 x9 : 0000000000000020
kern.warn kernel: [  819.236399] x8 : 0101010101010101 x7 : 0000000080808080 x6 : 0000000080808080
kern.warn kernel: [  819.237084] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
kern.warn kernel: [  819.237766] x2 : ffff000002f0e400 x1 : 0000000000000000 x0 : 0000000000000000
kern.warn kernel: [  819.238452] Call trace:
kern.warn kernel: [  819.238696]  get_device_state+0x18/0x70
kern.warn kernel: [  819.239077]  netdev_trig_notify+0xb0/0x260
kern.warn kernel: [  819.239480]  raw_notifier_call_chain+0x54/0x74
kern.warn kernel: [  819.239919]  call_netdevice_notifiers_info+0x58/0xa4
kern.warn kernel: [  819.240406]  dev_change_name+0x18c/0x2d0
kern.warn kernel: [  819.240798]  do_setlink+0x704/0xec0
kern.warn kernel: [  819.241149]  rtnl_setlink+0xe0/0x180
kern.warn kernel: [  819.241503]  rtnetlink_rcv_msg+0x11c/0x380
kern.warn kernel: [  819.241904]  netlink_rcv_skb+0x5c/0x130
kern.warn kernel: [  819.242292]  rtnetlink_rcv+0x18/0x24
kern.warn kernel: [  819.242648]  netlink_unicast+0x204/0x310
kern.warn kernel: [  819.243040]  netlink_sendmsg+0x1a0/0x3d4
kern.warn kernel: [  819.243433]  __sys_sendto+0xd8/0x140
kern.warn kernel: [  819.243797]  __arm64_sys_sendto+0x28/0x40
kern.warn kernel: [  819.244197]  invoke_syscall.constprop.0+0x50/0xf0
kern.warn kernel: [  819.244660]  do_el0_svc+0x58/0x170
kern.warn kernel: [  819.245001]  el0_svc+0x18/0x54
kern.warn kernel: [  819.245324]  el0t_64_sync_handler+0xf4/0x120
kern.warn kernel: [  819.245740]  el0t_64_sync+0x174/0x178
kern.emerg kernel: [  819.246114] Code: 910003fd f9000bf3 aa0003f3 f9404c00 (f9401c01)
kern.warn kernel: [  819.246695] ---[ end trace 0000000000000000 ]---
```

After removing the netdev led triggers in `/etc/config/system` the oops is gone.

### OpenWrt version

r24775-04ed95d1f9

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R5C

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-09:
It is upstream code that races with interface rename.
https://github.com/torvalds/linux/blob/master/drivers/leds/trigger/ledtrig-netdev.c

robimarko, 2024-02-04:
This should be fixed with:
https://github.com/openwrt/openwrt/commit/508f2dbfb37e6e23b671f642e5378f9b037fbeb1

manuelm, 2026-11-09:
Took me a while to upgrade my setup but I cannot reproduce this anymore.