openwrt/openwrt #14384: ath9k sets higher voltage on LNA

psyborg55, 2024-01-10:
### Describe the bug

during repair of the device that had burned LNA on CH2 and damaged LNA on CH0 i noticed this version of openwrt sets higher voltage on collector pin of LNA than factory tp-link firmware sets

as it is kernel driver, probably is still present in latest releases

### OpenWrt version

19.07.0 r10860-a3ffeb413b

### OpenWrt release

19.07.0 r10860-a3ffeb413b

### OpenWrt target/subtarget

ar71xx

### Device

TP-Link Archer C7 v1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-11:
¿ [like this burned](https://www.smallnetbuilder.com/wp-content/uploads/2014/08/tplink_archerc7_v2_v1_board_compare.jpg) ?
It is 5 years old version on twice older router. Voltages stay as initialized by factory bootloader. It is too late to try to switch any voltage regulator gpio to lower voltage  now.  Even if there is such control present.

psyborg55, 2024-01-11:
that photos are of too low resolution to see anything on LNA chips. it was burned internally without visible damage. it can be checked by measuring R466 and R472 in circuit - if those are less than 50Kohm and 10Kohm, LNA is damaged.

factory bootloader was used in both cases, and these pins are not initialized during early boot stage

brada4, 2024-01-11:
The v1 circuit in picture looks like been in baking oven with LED pad and 2.4GHz solder points looking well cooked vs fresh V2 board. Both are photographed as received from store kind of.
Code has just magic initvals of clocks and led blinker gpio-s exported, nothing specific to discrete component voltages.
https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/dts/qca9558_tplink_archer-c.dtsi
https://github.com/openwrt/openwrt/blob/main/target/linux/ath79/dts/qca9558_tplink_archer-c7-v1.dts
Only investigtion you can do is measure supplied voltage on last functional AMP and compare stock fw vs old OpenWRT vs latest OpenWRT. 
Approaching 10th year of service - power filter capacitors are about to give up if not already etc. 
And wifi5 card is not original (that is known to crash if present) and may have different power spec (not saying out of spec, just dozen milliA more draw)
? https://openwrt.org/toh/views/toh_available_16128_ax-wifi

psyborg55, 2024-01-11:
dts cannot have anything related as this is wireless driver issue. and in git.kernel there is like 5 changes in last 5 years to the file that probably holds incorrect setting. so the issue is present in latest releases. also not just archer c7 related but any other QCA9558 device

i have replaced the faulty chips, i still get the same performance with tp-link firmware on both bands as it was when i got the device, and on all 3 LNAs openwrt sets higher voltage from XLNABIAS_ pins

brada4, 2024-01-11:
No changes to upstream logic done (check patches/ for each interim releases)
https://github.com/openwrt/openwrt/tree/master/target/linux/ath79
Also upstream ath9k has just arrays of magic values programming radio not changed for long time now.
Probably worth reporting upstream.
Somewhat better images,
https://www.smallnetbuilder.com/wireless/wireless-reviews/tp-link-archer-c7-v2-reviewed/ 
The remaining resistance via blown amp seems like voltage reduction measure, hope you soldered multi-voltage amp for now.