openwrt/openwrt #14395: UA 6GHz band not usable with ath11k

nazar-pc, 2024-01-13:
### Describe the bug

I'm not able to use 6GHz Wi-Fi with QCN9074 card (6GHz only variant) and I think there are multiple reasons.

The first one is that by default 6GHz band is not usable:
```
root@OpenWrt:~# iw reg get
global
country 00: DFS-UNSET
	(755 - 928 @ 2), (N/A, 20), (N/A), PASSIVE-SCAN
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
	(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country US: DFS-FCC
	(2402 - 2472 @ 40), (6, 30), (N/A)
	(5170 - 5250 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5735 - 5895 @ 160), (N/A, 30), (N/A), AUTO-BW
	(5945 - 7125 @ 160), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW
```

When I tried to set it to my country (UA), I've got two results:
```
root@OpenWrt:~# iw reg get
global
country UA: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A), NO-OUTDOOR
	(5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
	(5725 - 5850 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 16), (N/A), NO-OUTDOOR

phy#0 (self-managed)
country UA: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A), NO-HE
	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW, NO-HE
	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW, NO-HE
	(5490 - 5590 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW, NO-HE
	(5590 - 5650 @ 40), (N/A, 24), (600000 ms), DFS, AUTO-BW, NO-HE
	(5650 - 5710 @ 40), (N/A, 24), (0 ms), DFS, AUTO-BW, NO-HE
	(5735 - 5835 @ 80), (N/A, 24), (N/A), AUTO-BW, NO-HE
```

I believe 6GHz is incorrectly missing there (but I have no idea whose fault it is).

Also apparently driver crashed at the same time:
```
[  222.654127] ------------[ cut here ]------------
[  222.656110] WARNING: CPU: 2 PID: 1608 at ath11k_reg_update_chan_list+0x250/0x270 [ath11k]
[  222.656796] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath11k_pci ath11k ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mac80211 lzo cfg80211 slhc r8169 qrtr_mhi qrtr qmi_helpers nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mhi lzo_rle lzo_decompress lzo_compress libcrc32c igc forcedeth e1000e crc_ccitt compat bnx2 i2c_dev dwmac_intel dwmac_generic stmmac_platform stmmac ixgbe e1000 amd_xgbe mdio nls_utf8 pcs_xpcs ena sha512_ssse3 sha512_generic seqiv jitterentropy_rng drbg michael_mic hmac cmac crypto_acompress nls_iso8859_1 nls_cp437 igb vfat fat button_hotplug tg3 realtek phylink mii
[  222.666556] CPU: 2 PID: 1608 Comm: hostapd Not tainted 6.1.71 #0
[  222.667190] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 2023.05-2 09/23/2023
[  222.667941] RIP: 0010:ath11k_reg_update_chan_list+0x250/0x270 [ath11k]
[  222.668630] Code: 01 fe ff ff 41 c7 87 50 51 00 00 00 00 00 00 e9 f1 fd ff ff 49 8d bf 80 06 00 00 be d0 07 00 00 e8 85 4b 31 e1 e9 ee fd ff ff <0f> 0b bb ea ff ff ff eb a3 48 83 cf ff e9 6a fe ff ff bb f4 ff ff
[  222.670418] RSP: 0018:ffffc9000098fb48 EFLAGS: 00010246
[  222.671136] RAX: 0000000000000000 RBX: 0000000000000000 RCX: ffff8881026df0ec
[  222.671853] RDX: 0000000000000001 RSI: ffff8881069b0528 RDI: ffff8881069b2080
[  222.672743] RBP: ffffc9000098fb78 R08: 0000000020000000 R09: 00000000000012b8
[  222.673578] R10: ffff8881040a8840 R11: 0000000000000080 R12: ffff8881069b08e0
[  222.674371] R13: ffff8881069b04f8 R14: ffff8881069b0528 R15: ffff8881069b2080
[  222.675097] FS:  00007fd26ec1eb48(0000) GS:ffff888277d00000(0000) knlGS:0000000000000000
[  222.675862] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  222.676549] CR2: 00007fd26e5be000 CR3: 0000000103730000 CR4: 0000000000350ee0
[  222.677322] Call Trace:
[  222.678012]  <TASK>
[  222.678586]  ? show_regs.part.0+0x1e/0x24
[  222.679151]  ? show_regs.cold+0x8/0xd
[  222.679762]  ? __warn+0x70/0xc0
[  222.680393]  ? ath11k_reg_update_chan_list+0x250/0x270 [ath11k]
[  222.681135]  ? report_bug+0xbb/0x120
[  222.681889]  ? handle_bug+0x45/0x80
[  222.682478]  ? exc_invalid_op+0x18/0x70
[  222.683117]  ? asm_exc_invalid_op+0x1b/0x20
[  222.683753]  ? ath11k_reg_update_chan_list+0x250/0x270 [ath11k]
[  222.684429]  ath11k_mac_tx_mgmt_pending_free+0x3a30/0x7e00 [ath11k]
[  222.685089]  drv_start+0x2e/0x50 [mac80211]
[  222.685663]  ieee80211_do_open+0x32f/0x780 [mac80211]
[  222.686294]  ieee80211_do_open+0x75b/0x780 [mac80211]
[  222.687162]  __dev_open+0xe4/0x1a0
[  222.687720]  __dev_change_flags+0x189/0x1f0
[  222.688260]  ? __check_object_size+0x4c/0x250
[  222.688828]  dev_change_flags+0x22/0x60
[  222.689410]  devinet_ioctl+0x3b2/0x7e0
[  222.694101]  inet_ioctl+0x15d/0x190
[  222.694676]  ? dev_get_by_name_rcu+0x63/0x80
[  222.695248]  ? dev_get_by_name_rcu+0x63/0x80
[  222.695803]  ? dev_ioctl+0x3a5/0x460
[  222.696311]  sock_ioctl+0x1dc/0x320
[  222.696828]  __x64_sys_ioctl+0x42c/0x9b0
[  222.697395]  do_syscall_64+0x42/0x90
[  222.698148]  entry_SYSCALL_64_after_hwframe+0x64/0xce
[  222.698755] RIP: 0033:0x7fd26ebdc5a4
[  222.699302] Code: 63 f6 48 8d 44 24 60 48 89 54 24 30 48 89 44 24 10 48 8d 44 24 20 48 89 44 24 18 b8 10 00 00 00 c7 44 24 08 10 00 00 00 0f 05 <48> 63 f8 e8 c4 fb fd ff 48 83 c4 58 c3 0f be 05 0a 03 04 00 c3 53
[  222.700912] RSP: 002b:00007ffe212e9350 EFLAGS: 00000206 ORIG_RAX: 0000000000000010
[  222.701628] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007fd26ebdc5a4
[  222.702287] RDX: 00007ffe212e93b0 RSI: 0000000000008914 RDI: 0000000000000010
[  222.702929] RBP: 00007fd26e5ad220 R08: 00007fd26e5656e0 R09: 0000000000000004
[  222.703567] R10: 0000000000000002 R11: 0000000000000206 R12: 0000000000000010
[  222.704223] R13: 00007ffe212e93b0 R14: 0000000000000001 R15: 0000000000000000
[  222.704957]  </TASK>
[  222.705447] ---[ end trace 0000000000000000 ]---
```

### OpenWrt version

r24801-b7f9742da8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Additional info

I'm running OpenWRT in a VM just like I did for many-many years prior with ath10k driver and other Wi-Fi 5/6 Qualcomm/Atheros adapters.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-01-13:
> [  222.667941] RIP: 0010:ath11k_reg_update_chan_list+0x250/0x270 [ath11k]

Reminds me https://github.com/openwrt/firmware_qca-wireless/pull/18, so you might be using old BDF and need to update?

brada4, 2024-01-13:
Does it work on pure hardware like off usb stick?

nazar-pc, 2024-01-13:
It is quite challenging to use on bare metal in my case unfortunately, so I didn't try.
I do use latest (as of right now) snapshot, so I think updates mentioned above are already part of the build.

brada4, 2024-01-13:
You need to change power, acpi and irq and apic settings and pci slots in physical server. 
And full dmesg including pci initialisation and ath11k one.
and boot usb at each occassion.
Also try in a debian or fedora vm while in distance.
And lspci -vvv from your card.

reg settings are consistent with upstream:
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n1755
You need to convince them with a copy of law text.

ynezz, 2024-01-13:
> QCN9074 card (6GHz only variant)

It would be nice to know which card its that.

>  I do use latest (as of right now) snapshot, so I think updates mentioned above are already part of the build.

You're likely using some miniPCIe variant, so I would say, that there is some EEPROM on the board and the BDF/calibration data are loaded directly from there, thus the host OS is not involved here. Maybe kernel just needs some compat layer to support this older BDFs? Other option might be to dump the EEPROM, extract the calibration data, update to the new format like and write it back to the EEPROM, profit.

ynezz, 2024-01-13:
I don't know how that exactly works and if there are any checks on the QCN9074 side (like enforcing EEPROM BDF only), but it seems, that host is involved in the BDF loading, card sends EEPROM capability and hosts then uses this during BDF loading, so maybe you can trick the card to take the BDF from the host instead of EEPROM by commenting out following code https://elixir.bootlin.com/linux/v6.1.72/source/drivers/net/wireless/ath/ath11k/qmi.c#L2256 ? And thus you could try to send some fixed BDF (you would need to modify the card ID in that BDF so it matches your card's `board_id 0xa2`.

brada4, 2024-01-14:
The problem is that proxmox or another kvm initialised reg domain on physical host. You need to assign device to virtio stub to avoid any host os interference (besides you will not be able to transmit anything with that card under current laws)

nazar-pc, 2024-01-14:
Will reply to a few questions I have answers for right now.

And full dmesg including pci initialisation and ath11k one.

<details>

```
[    0.000000] Linux version 6.1.71 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24801-b7f9742da8) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Thu Jan 11 09:40:43 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=52674438-8699-578e-bfc5-91be83724702 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000002ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000030000-0x000000000004ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000050000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000005e8ebfff] usable
[    0.000000] BIOS-e820: [mem 0x000000005e8ec000-0x000000005e9ebfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000005e9ec000-0x000000005eaebfff] type 20
[    0.000000] BIOS-e820: [mem 0x000000005eaec000-0x000000005eb6bfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000005eb6c000-0x000000005eb7dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000005eb7e000-0x000000005ebfdfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000005ebfe000-0x000000007effffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007f000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] BIOS-e820: [mem 0x000000fd00000000-0x000000ffffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by Ubuntu distribution of EDK II
[    0.000000] efi: SMBIOS=0x5e9d4000 ACPI=0x5eb7d000 ACPI 2.0=0x5eb7d014 MEMATTR=0x5d2bd118 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 2023.05-2 09/23/2023
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 10269939781 cycles
[    0.000003] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000006] tsc: Detected 2999.998 MHz processor
[    0.000128] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000130] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000136] last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000189] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000197] last_pfn = 0x7f000 max_arch_pfn = 0x400000000
[    0.004542] Using GB pages for direct mapping
[    0.004877] Secure boot disabled
[    0.004878] ACPI: Early table checksum verification disabled
[    0.004880] ACPI: RSDP 0x000000005EB7D014 000024 (v02 BOCHS )
[    0.004884] ACPI: XSDT 0x000000005EB7C0E8 000054 (v01 BOCHS  BXPC     00000001      01000013)
[    0.004890] ACPI: FACP 0x000000005EB78000 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004894] ACPI: DSDT 0x000000005EB79000 002AE9 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004897] ACPI: FACS 0x000000005EBDC000 000040
[    0.004899] ACPI: APIC 0x000000005EB77000 000090 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004901] ACPI: MCFG 0x000000005EB76000 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004903] ACPI: DMAR 0x000000005EB75000 0000E8 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004905] ACPI: WAET 0x000000005EB74000 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.004908] ACPI: BGRT 0x000000005EB73000 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.004909] ACPI: Reserving FACP table memory at [mem 0x5eb78000-0x5eb780f3]
[    0.004911] ACPI: Reserving DSDT table memory at [mem 0x5eb79000-0x5eb7bae8]
[    0.004912] ACPI: Reserving FACS table memory at [mem 0x5ebdc000-0x5ebdc03f]
[    0.004912] ACPI: Reserving APIC table memory at [mem 0x5eb77000-0x5eb7708f]
[    0.004913] ACPI: Reserving MCFG table memory at [mem 0x5eb76000-0x5eb7603b]
[    0.004914] ACPI: Reserving DMAR table memory at [mem 0x5eb75000-0x5eb750e7]
[    0.004914] ACPI: Reserving WAET table memory at [mem 0x5eb74000-0x5eb74027]
[    0.004915] ACPI: Reserving BGRT table memory at [mem 0x5eb73000-0x5eb73037]
[    0.004958] Zone ranges:
[    0.004958]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.004960]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.004961]   Normal   [mem 0x0000000100000000-0x000000027fffffff]
[    0.004962] Movable zone start for each node
[    0.004963] Early memory node ranges
[    0.004963]   node   0: [mem 0x0000000000001000-0x000000000002ffff]
[    0.004964]   node   0: [mem 0x0000000000050000-0x000000000009efff]
[... 685 lines cut ...]
		LnkCap2: Supported Link Speeds: 2.5-8GT/s, Crosslink- Retimer- 2Retimers- DRS-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: unsupported
[... cut ...]
> The problem is that proxmox or another kvm initialised reg domain on physical host. You need to assign device to virtio stub to avoid any host os interference (besides you will not be able to transmit anything with that card under current laws)

I use libvirt wth KVM, there were some difficulties getting it to work in a VM at all, but we found a way around it [on the forum](https://forum.openwrt.org/t/qcn9074-doesnt-initialize-on-x86-64/163288?u=nazar-pc). And even though it seems to work fine without it, I do usse vfio-pci for the card on the host, I've being using VFIO/PCIe passthrough with various devices for almost a decade now.

brada4, 2024-01-14:
-22 means virtualisation did not work right. You have to fix it first.

nazar-pc, 2024-01-14:
> -22 means virtualisation did not work right. You have to fix it first.

I know, if you read the thread in full (I don't expect you to though) you'll see that we have found a solution to that particular problem, so virtualization is a non-issue at this point as you can see from the kernel logs I've attached above.

nazar-pc, 2024-01-14:
Given that 6 GHz band is not in upstream database, I don't think there is much to do downstream. I'm closing this one and thanks everyone for feedback!

closed 2024-01-14 by nazar-pc

brada4, 2024-01-14:
You need to post full bug description here. Once thread confirms problem is in virtualisation thats all news. It will not work in your setup as passed through pci device, nor it will have any practical usage in your place due to radio band not permitted.
You still hide system and logs of host regarding double initialisation of card. You need to prevent that absolutely.