openwrt/openwrt #14396: impossible to get CVG834G binary file, only manifest

lm2lm2, 2024-01-13:
### Describe the bug

I tried several versions since backfire until 21th release of openwrt
none of them was able to output any .bin for CVG834G, even if it appears well within the netgear/brcm category
it's always a .manifest file, never .bin, why?

no error message output on compilation
no disk space problem on the computer, no specific added package, lot of them removed

but impossible to get the CVG834G openwrt's binary, only 14.07 whom doesnt works because of a bug fixed later

how can i do to get 15+th release openwrt's CVG834G file?

thank you

### OpenWrt version

21

### OpenWrt release

21

### OpenWrt target/subtarget

brcm

### Device

CVG834G

### Image kind

Official downloaded image

### Steps to reproduce

try to compile CVG834G on 21th or 22th of openwrt
no bin file, only manifest file

### Expected behaviour

output of CVG834G openwrt .bin file

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lm2lm2, 2024-01-13:
openwrt 21 supported version with CVG834G supported device

brada4, 2024-01-13:
Image is not generated if it is too big for 4MB flash 
v21 is no longer generally supported.

lm2lm2, 2024-01-13:
> Image is not generated if it is too big for 4MB flash v21 is no longer generally supported.

well, even if i disable almost everything, all brcm-based-models are generated **except** that one. Might be a problem somewhere, where lot of others devices whom their images are generated, less than 4MB, are well generated during the same build-in. why only 14th rel of openwrt gets a (wrong) file, where almost all others openwrt releases did not generated that one?
it's very strange and uncommon.. i guess the bug is still here in openwrt23, where i dont see the generated image on openwrt official downloads.. even among 4MB devices where openwrt 23 is downloadable!

brada4, 2024-01-14:
No more advice than in wiki. You can manage swich via ssh maximum.
Try unrelated ddwrt or buy new router, 10 years is a good run for any electronics.
https://openwrt.org/toh/views/toh_available_16128_ax-wifi

lm2lm2, 2024-01-14:
> No more advice than in wiki. You can manage swich via ssh maximum. Try unrelated ddwrt or buy new router, 10 years is a good run for any electronics. https://openwrt.org/toh/views/toh_available_16128_ax-wifi

obviously the goal is not to buy new equipment at all, but to understand why, and if someone were able to output that device binary file... as i see only 14rel proposes it.. it's seems to be very strange.

no one tried with openwrt 23?
device is not supported with ddwrt, but is still since years within the openwrt's official builder, and no one noticed that it's never .bin generated, as only .manifest instead?

brada4, 2024-01-14:
It is too small to be supported with modern releases.
Last time I checked dd-wrt was free, you did not state in this or forum post you want to utilize hardware till it rots away, you just stated image not built in unspecified conditions.

lm2lm2, 2024-01-14:
> It is too small to be supported with modern releases. Last time I checked dd-wrt was free, you did not state in this or forum post you want to utilize hardware till it rots away, you just stated image not built in unspecified conditions.

CVG834G is not supported on ddwrt
even with latest release, some 4MB device are still available
how could i build a CVG834G image? it's just what i want, it's not a crime :)

brada4, 2024-01-14:
You have rougly 150% chance to brick device for good removing squashfs and full image size restrictions in makefiles.
You come too late even for ddwrt.
No pointin continuing, wifi-g is plague of airtime .