openwrt/openwrt #14397: When enable rrm_neighbor_report and rrm_neighbor_report in luci , it delete the option from /etc/config/wireless

permails, 2024-01-13:
### Describe the bug

When enable rrm_neighbor_report and rrm_neighbor_report in luci , it delete the option from /etc/config/wireless，It‘s not excepted. 
But when I disable rrm_neighbor_report and rrm_neighbor_report in luci, it set rrm_neighbor_report=‘0’ and rrm_neighbor_report=’0‘ ，it’s excepted.

### OpenWrt version

23.05.2

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

23.05.2

### Device

TP-LINK TL-XDR6088

### Image kind

Official downloaded image

### Steps to reproduce

1. with WPAD prepared
2. enable 802.11k RRM in the luci page ->network->wireless -> edit ssid ->WLAN roaming
3. enable 802.11k rrm
4. enable rrm_neighbor_report
5. enable rrm_beacon_report
6. save and apply
7. Check the config :/etc/config/wireress there is NO rrm_neighbor_report='1' and rrm_beacon_report='1' option
8. disable rrm_neighbor_report and rrm_beacon_report
9. Check the config :/etc/config/wireress there IS rrm_neighbor_report='0' and rrm_beacon_report='0' option
10. vi the /etc/config/wireress, add the option rrm_neighbor_report='1' and rrm_beacon_report='1' ,restart the wireless radio and sniffer the airlogs, can find neighbor report 
11. disable and enable rrm_neighbor_report enable rrm_beacon_report again ,rrm_neighbor_report='1' and rrm_beacon_report='1' option was deleted ,sniffer the airlogs , the neighbor is non.

### Actual behaviour

when enable rrm_neighbor_report and rrm_beacon_report ,the config of wireless has NO rrm_neighbor_report='1' or rrm_beacon_report='1' 

### Expected behaviour

when enable rrm_neighbor_report and rrm_beacon_report ,the config of wireless will have  rrm_neighbor_report='1' or rrm_beacon_report='1' 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2024-01-13:
It's because ieee80211k implies that rrm_neighbor_report is set. So setting both ieee80211k and rrm_neighbor_report is redundant.

permails, 2024-01-14:
@rany2 
So what you mean is that if I enable 802.11k, neighbor and Beacon reporting will be enabled by default?

rany2, 2024-01-14:
@permails exactly.

permails, 2024-01-14:
So much thx @rany2 
I will check it again

closed 2024-01-14 by permails