openwrt/openwrt #14399: IPQ-WIFI package doesn’t contain my custom board-2 data (ZTE-MF289D)

Kroustalo, 2024-01-14:
### Describe the bug

Good day everyone. I began working on adding support of OpenWrt on my [ZTE MF289D](https://github.com/Kroustalo/OpenWRT-ZTE-MF289D) Since the router is similar to the MF289F [uploaded](https://github.com/openwrt/openwrt/pull/11044) by @stich86  (difference is only in the LTE Modem) I started working on the port. 

I finished adding the configuration and compiles successfully, when I'm booting it does not detect the board-2 data file for the Wi-Fi on 2.4Ghz and 5Ghz. 
After checking the installed file, I noticed that the board-2 file that it has does not match the `bmi-board-id` with one that I created. For example, 0 instead of 31 for the QCA4019.

I checked for the package:
 `./bin/targets/ipq40xx/generic/packages/ipq-wifi-zte_mf289d_2024-01-06-71f45cff-1_arm_cortex-a7_neon-vfpv4.ipk`
but the size of the file from what I researched is way too small for it.

I believe there's some sort of bug that does not allow the system to pass the board data (looks like it's empty) for my router ~~or I messed up the code in my fork~~ 

### OpenWrt version

r24782-4e4cfe230a81

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZTE MF289D

### Image kind

Self-built image

### Steps to reproduce

Download my [fork](https://github.com/Kroustalo/OpenWRT-ZTE-MF289D) and build an image for the ZTE MF289D

### Actual behaviour

No Wi-Fi 2.4ghz and 5ghz bands because the system loads an invalid/empty board-2 file that not the excpected size

### Expected behaviour

Working board-2 file that enables the wireless capabilities of the router.

### Additional info

This is the boot log [mf289d 14012024.txt](https://github.com/openwrt/openwrt/files/13931874/mf289d.14012024.txt)
and here is the part on the Wi-Fi chip.
> [    9.405196] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[    9.405265] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[    9.418374] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
[   11.720758] ath10k_pci 0000:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0046,subsystem-vendor=168c,subsystem-device=cafe,variant=zte,mf289d from ath10k/QCA9984/hw1.0/board-2.bin
[   11.896176] ath10k_pci 0000:01:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9984/hw1.0
[   11.896249] ath10k_pci 0000:01:00.0: failed to fetch board file: -12
[   11.904746] ath10k_pci 0000:01:00.0: could not probe fw (-12)
[   12.721721] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   12.721795] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   12.733698] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   12.796225] ath10k_ahb a000000.wifi: failed to fetch board data for bus=ahb,bmi-chip-id=0,bmi-board-id=31,variant=zte,mf289d from ath10k/QCA4019/hw1.0/board-2.bin
[   12.972485] ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   12.972572] ath10k_ahb a000000.wifi: failed to fetch board file: -12
[   12.980951] ath10k_ahb a000000.wifi: could not probe fw (-12)
[   13.801528] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   13.801601] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   13.813499] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   13.875796] ath10k_ahb a800000.wifi: failed to fetch board data for bus=ahb,bmi-chip-id=0,bmi-board-id=31,variant=zte,mf289d from ath10k/QCA4019/hw1.0/board-2.bin
[   14.052465] ath10k_ahb a800000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   14.052547] ath10k_ahb a800000.wifi: failed to fetch board file: -12
[   14.060941] ath10k_ahb a800000.wifi: could not probe fw (-12)

Afterwards, I attach a ethernet cable and get access through ssh

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_zte_mf289d=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-dashboard=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-proto-qmi=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_luci-theme-openwrt-2020=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

kirdesde, 2024-01-14:
Have you even read this:

https://github.com/Kroustalo/OpenWRT-ZTE-MF289D/blob/6c53de7c2ed5314cd83100f10073230fbaeb1a16/package/firmware/ipq-wifi/Makefile#L125

stich86, 2024-01-14:
@kirdesde i've tried on behalf of @Kroustalo 

here my ipq-wifi folder with "files" folder:

```
openwrt-buildroot:~/4G_Router/OpenWRT-ZTE-MF289D/package/firmware/ipq-wifi$ ls -latR *
-rw-r--r-- 1 xxxx xxxxx 6415 Jan 14 13:36 Makefile

files:
total 32
drwxr-xr-x 2 xxxx xxxx  4096 Jan 14 21:25 .
drwxr-xr-x 3 xxxx xxxx  4096 Jan 14 21:25 ..
-rw-r--r-- 1 xxxx xxxx 12164 Jan 14 13:36 board-zte_mf289d.qca4019
-rw-r--r-- 1 xxxx xxxx 12204 Jan 14 13:36 board-zte_mf289d.qca9984
```

Anyway the package `ipq-wifi-zte_mf289d_2024-01-06-71f45cff-1_arm_cortex-a7_neon-vfpv4.ipk` still missing the board-2.bin file inside

Any other thing that we are missing?

Thanks!

stich86, 2024-01-14:
I've found this issue: https://github.com/openwrt/openwrt/issues/13561

Looks like the behavior to add new board files is changed.

@Ansuel what is the new procedure?

Thanks!

Ansuel, 2024-01-14:
Instruction in the makefile

stich86, 2024-01-14:
> Instruction in the makefile

These one (for testing) are not working:

# To add local files for testing:
# 1. create a new directory in the ipq-wifi package directory called "files".
# 2. place there the needed board files.

I'd to add manually to the got pulled file to get ipq-wifi IPK with correct board-2 files

What I'm doing wrong?

Thanks

kirdesde, 2024-01-15:
your ipq-wifi package directory does not contain a sub-directory "files"

stich86, 2024-01-15:
> your ipq-wifi package directory does not contain a sub-directory "files"

The files subdirectory doesn't seems to work
Only way is to out board file into downloaded git directory for ipq-wifi package

@Ansuel can you confirm this behavior?
Creating a subdirectory "files" into /packages/firmware/ipq-wifi and copy board definition there doesn't produce the expected action (like read me says). It was also exposed in the issue that I've linked

What I'm doing wrong?

Thanks in advance

kirdesde, 2024-01-15:
Another possibility is to fork the qca-wireless.git repo, commit your board files and point the ipq-wifi Makefile to your fork.

stich86, 2024-01-15:
> Another possibility is to fork the qca-wireless.git repo, commit your board files and point the ipq-wifi Makefile to your fork.

And it's ok..
But readme says that's just create a "files" folder on the ipq-wifi and put files there. This currently not know. If it's a know issue I think instructions should be updated accordingly to avoid misunderstanding for new users :)

testuser7, 2024-01-17:
For local testing you can use the following notation:
```
PKG_VERSION:=2024-01-17
PKG_SOURCE:=ipq-wifi-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=file:///home/user/ipq-wifi/
PKG_HASH:=ad2f0bed258aff55d620c23f48fa6467c509f6258081baf81b011866d2d6fb30
```

Kroustalo, 2024-01-19:
Hello, I moved the files and booted succesfully with the wifi enabled.
It seems that I have 2 2.4 GHz radios instead of one, but they seem to function...

[image]

I think that the .dts (in my repo) might be configured wrong but here's the json's and the bootlogs for good measure:

json  (router has 9984 for 5Ghz but seems to work)
[QCA4019_MF289D.json](https://github.com/openwrt/openwrt/files/13992856/QCA4019_MF289D.json)
[QCA9984_MF289D.json](https://github.com/openwrt/openwrt/files/13992904/QCA9984_MF289D.json)

Stock bootlog
[COM PRINT OEM.txt](https://github.com/openwrt/openwrt/files/13992644/COM.PRINT.OEM.txt)

OpenWrt bootlog
[OpenWrt.txt](https://github.com/openwrt/openwrt/files/13992658/OpenWrt.txt)

Stock bootlog After OpenWrt Reset
[OEM COM PRINT after tests.txt](https://github.com/openwrt/openwrt/files/13992807/OEM.COM.PRINT.after.tests.txt)

kirdesde, 2024-01-20:
2 concurrent 2.4 radios in one device is technically impossible.

Why do you have 3 entries in your QCA4019_MF289D.json? According to the stock boot log only board id 20 is used for the ipq4019 wifi.

Drop the other two entries and disable the wifi1 node in the dts and try again.

And please don't post LUCI screenshots but the output of iw or iwconfig

giig982, 2024-04-16:
Hi @Kroustalo did you make some further step on the porting?

Kroustalo, 2024-04-16:
> Hi @Kroustalo did you make some further step on the porting?
Hello, I'm currently on my conscription until December... I'll tackle it when I have time.

Lanchon, 2024-05-06:
@Ansuel @testuser7 

there is a bug in the build system (or its doc), **i can confirm these instructions DO NOT work:**

https://github.com/openwrt/openwrt/blob/064d4a808367295233db5fce981e3b1d4305b014/package/firmware/ipq-wifi/Makefile#L132-L134

the files in `package/firmware/ipq-wifi/files` do not get copied to `$(PKG_BUILD_DIR)`. the resulting device package is totally empty. hence, no files overlay the default `board-2.bin` files, which then end up packaged in the build.

**workaround:**

```diff
---------------------- package/firmware/ipq-wifi/Makefile ----------------------
index 02ed94ae8e..173a98c3f3 100644
@@ -123,7 +123,7 @@ Do not install it for any other device!
   endef
 
   define Package/ipq-wifi-$(1)/install-overlay
-    $$$$(foreach IPQ_WIFI_BOARD_FILE,$$$$(wildcard $(PKG_BUILD_DIR)/board-$(1).*),\
+    $$$$(foreach IPQ_WIFI_BOARD_FILE,$$$$(wildcard $(PKG_BUILD_DIR)/board-$(1).*) $$$$(wildcard files/board-$(1).*),\
       $$$$(call ipq-wifi-install-one,$$$$(IPQ_WIFI_BOARD_FILE),$$(1)))
   endef
```

Lanchon, 2024-05-06:
maybe the files are copied to the build dir first, but then the git repo contents totally overwrite the dir?

Kroustalo, 2024-05-06:
The workaround has worked in my build but I'm still having the dual 2.4Ghz bands on the ipq4019 (with only one entry on my board-2 file)
```
[
    {
        "data": "boardData_1_0_IPQ4019_DK04_2G.bin",
        "names": [
            "bus=ahb,bmi-chip-id=0,bmi-board-id=31,variant=zte,mf289d"
        ]
    }
]
```

testuser7, 2024-05-06:
@Lanchon so maybe you should create a PR?

Kroustalo, 2024-05-06:
@testuser7 I  commented out the _wifi1_ entry and now I have 2 radios.

So I guess that my repo works now...

Lanchon, 2024-05-06:
> @Lanchon so maybe you should create a PR?

i dont like this workaround as a solution (yet):
- we don't know why `files/*` are not being copied. maybe an interaction between [`PKG_SOURCE_*`](https://github.com/openwrt/openwrt/blob/0b0e3e22f8830c6f0224509f25eb0f55d36ba7b5/package/firmware/ipq-wifi/Makefile#L7-L11) and `files/*`? this could affect other makefiles.
- if this gets fixed elsewhere, we might get a bug here. if a fix like this is wanted, i'd argue against using `files` and rather use any other subfolder name or no subfolder.
- a fix like this requires testing:
  - for the case where the subfolder is empty (ie, no BDFs are added).
  - for the case where the subfolder does not exist.
- i think the makefile should explicitly fail if building an empty package, rather than openwrt failing on runtime as it does now.

Kroustalo, 2024-05-06:
Just tried your cases @Lanchon and the compiled successfully without the board files on the build_dir folder. I believe that since it looks the files and you specify the router on the makefile config it will proceed the installation without issues

You could add a check to inform you that it did not found the board files and abort the compile, or add a comment saying that:
"No files = No wifi firmware"

Lanchon, 2024-05-06:
> Just tried your cases @Lanchon and the compiled successfully without the board files

thinking a bit more, i still don't think this is a proper fix.

i was thinking of adding `$$$$(wildcard board-$(1).*)` (because why should the `files` dir be needed?), but no.

one should be able to override an existing similarly named file in the git repo. so files should be copied over to whats on the build dir. adding a glob is not the same. with 2 globs, the order in which they process will determine the contents of the package, which is not good.

the solution is to copy over the files AFTER the repo extraction. and i suggest they SHOULD NOT be copied from `files/` as that folder is already being handled specially by the build system. in fact, no folder seems needed.

but i'll leave this for experts of make.

Insanityforawhile, 2024-08-31:
Could u guys help me, i am currently in root and trying to push the bin file with the sys command:

Thu Jan  1 00:13:04 UTC 1970 upgrade: Image metadata not present
Thu Jan  1 00:13:04 UTC 1970 upgrade: Use sysupgrade -F to override this check when downgrading or flashing to vendor firmware
[  784.691634] ubi: mtd17 is already attached to ubi0
ubiattach: error!: cannot attach mtd17
           error 17 (File exists)
Image check failed.
root@OpenWrt:~# 

I overwrite it with: sysupgrade -F /tmp/openwrt-ipq40xx-generic-zte_mf289d-squashfs-sysupgrade.bin

Seemed to work, router rebooted but now its stuck in tftp with the following error:

ZTE_InitFotaFlashPara: nand_curr_device=1
zte_getHandOffState: read data=0xff from 0x0
Press ESC to abort autoboot in 0 seconds
Creating 1 MTD partitions on "nand1":
0x000001800000-0x000003500000 : "mtd=0"
UBI: attaching mtd2 to ubi0
UBI: physical eraseblock size:   131072 bytes (128 KiB)
UBI: logical eraseblock size:    126976 bytes
UBI: smallest flash I/O unit:    2048
UBI: VID header offset:          2048 (aligned 2048)
UBI: data offset:                4096
UBI: attached mtd2 to ubi0
UBI: MTD device name:            "mtd=0"
UBI: MTD device size:            29 MiB
UBI: number of good PEBs:        232
UBI: number of bad PEBs:         0
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volumes: 1
UBI: number of user volumes:     0
UBI: available PEBs:             226
UBI: total number of reserved PEBs: 6
UBI: number of PEBs reserved for bad PEB handling: 2
UBI: max/mean erase counter: 2/1
zte_set_download_led(0) is called
timer=a8b36c05 ha=8daf1710
ip=192.23.175.141 mac=0x00:0xaa:0xbb:0x17:0xaf:0x8d
MAC0 addr:0:3:7f:ba:db:ad
PHY ID1: 0x4d
PHY ID2: 0xd0b1
ipq40xx_ess_sw_init done
eth0
Warning: eth0 MAC addresses don't match:
Address in SROM is         00:03:7f:ba:db:ad
Address in environment is  00:aa:bb:17:af:8d

eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 up Speed :1000 Full duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
Using eth0 device
TFTP from server 192.0.0.1; our IP address is 192.23.175.141
Filename 'root_uImage'.
Load address: 0x84000000
Loading: *
TFTP error: 'File not found' (1)
Not retrying...
zte_set_download_led(1) is called
blank download is done. send fin to PC...
eth0 PHY0 Down Speed :10 Half duplex
eth0 PHY1 up Speed :1000 Full duplex
eth0 PHY2 Down Speed :10 Half duplex
eth0 PHY3 Down Speed :10 Half duplex
eth0 PHY4 Down Speed :10 Half duplex
Volume kernel not found!
(IPQ40xx) #

Lanchon, 2024-08-31:
@Insanityforawhile, please delete your previous message and seek help in [the forum](https://forum.openwrt.org/).

Insanityforawhile, 2024-09-04:
Had a problem that the second ethernet port was not woking. Fixed it with a edit in /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd9d:65fb:f286::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'lan'  # LAN port
        list ports 'wan'  # Add WAN port to the bridge

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'Simkaart'
        option proto 'qmi'
        option apn 'smartsites.t-mobile'
        option auth 'none'
        option pdptype 'ipv4'
        option device '/dev/cdc-wdm0'

# Remove the LAn2 interface since it duplicates the device

MarcPot, 2025-01-13:
Any plans to merge this repo into main? I tried rebasing to main myself but I sadly do not have the skills to do so...

MarcPot, 2025-01-13:
Been working fine with me actually, only problem I have is that the 4G quits at exactly 0:44 am. Seems to be a ISP thing that can be fixed with modemmanager. But that’s where I’m stuck since opkg is not working anymore.

Kroustalo, 2025-01-13:
I think is due to being a snapshot version and the packages from opkg are not compatable yet. I'll check it in the next days in order to merge it.

Listum, 2025-01-17:
`root@AP:~# dmesg | grep ath10k
[   12.999563] ath10k 6.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   13.000735] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[   13.006701] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   14.560165] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   14.560237] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   14.573312] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7ea63dc5
[   16.866246] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id N/A crc32 eaf00652
[   20.356146] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   20.356216] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   20.437129] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   20.437953] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
[   20.714197] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[   22.154614] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   22.154685] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   22.166505] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   22.234974] ath10k_ahb a000000.wifi: failed to fetch board data for bus=ahb,bmi-chip-id=0,bmi-board-id=20,variant=zte,mf289d from ath10k/QCA4019/hw1.0/board-2.bin
[   22.419007] ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   22.419098] ath10k_ahb a000000.wifi: failed to fetch board file: -12
[   22.427524] ath10k_ahb a000000.wifi: could not probe fw (-12)
[   35.340617] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   35.340688] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   35.421511] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   35.422346] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
[   35.790020] ath10k_pci 0000:01:00.0: rts threshold -1
[   35.790718] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   35.810557] ath10k_pci 0000:01:00.0 phy0-ap0: entered allmulticast mode
[   35.816360] ath10k_pci 0000:01:00.0 phy0-ap0: entered promiscuous mode
[   35.863874] ath10k_pci 0000:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   43.964262] ath10k_pci 0000:01:00.0: wmi: fixing invalid VHT TX rate code 0xff`

In my case. Board file is loading only for qca9984, but not for qca4019.
Why? 
hlep

Listum, 2025-01-17:
Markdown github is not working fine for me, I'll upload as pastebin
https://pastebin.com/BJYxynYW

testuser7, 2025-02-14:
There is a bug: https://github.com/openwrt/openwrt/issues/14399#issuecomment-2095400759

You can try this workaround: https://github.com/openwrt/openwrt/issues/14399#issuecomment-1896639893

nbl530, 2025-02-22:
Thank you for all for the great work you made.
The original firmware of this modem suck a little bit :)
Waiting for the openwrt version

nbl530, 2025-02-22:
> I tested it on my own for a while... Seems to be working, although sometimes the Wi-Fi drops out. Might be a driver issue. You will need to supply a different modem card since the original is not compatable.
> 
> I can look into merging this version but I might need some assistance on the "paperwork"

Thank you for the work you do.
Can you clarify for us where the project is please? Can the beginner like me help you to test or something like?
Openwrt for mf289d will be super..

Kroustalo, 2025-02-22:
Started a pull request.
https://github.com/openwrt/openwrt/pull/18075

[My repo has the bin files  ](https://github.com/Kroustalo/OpenWRT-ZTE-MF289D/tree/main/ZTE%20MF289D%20bin)

momothefox, 2025-05-13:
> the files in package/firmware/ipq-wifi/files do not get copied to $(PKG_BUILD_DIR). the resulting device package is totally empty. hence, no files overlay the default board-2.bin files, which then end up packaged in the build.

Adding this to Makefile copied the board file to the build directory
and it ends up in the correct path in OpenWrt firmware image

```
define Build/Prepare
	$(call Build/Prepare/Default)
	$(CP) ./files/board-* $(PKG_BUILD_DIR)/
endef
```

but it fails to read the variant 
```
[   15.522210] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   15.522254] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   15.525916] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   15.525988] OF: /soc/wifi@a000000: could not find phandle 21
[   15.543455] ath10k_ahb a000000.wifi: Loading BDF type 0
[   15.544111] ath10k_ahb a000000.wifi: failed to fetch board data for bus=ahb,vendor=0000,device=0000,subsystem-vendor=0000,subsystem-device=0000,variant=device_model from ath10k/QCA4019/hw1.0/board-2.bin
[   15.710354] ath10k_ahb a000000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   15.710404] ath10k_ahb a000000.wifi: failed to fetch board file: -12
[   15.710602] ath10k_ahb a000000.wifi: could not probe fw (-12)
[   16.487231] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   16.487273] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   16.491321] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   16.491409] OF: /soc/wifi@a800000: could not find phandle 22
[   16.508878] ath10k_ahb a800000.wifi: Loading BDF type 0
[   16.509189] ath10k_ahb a800000.wifi: failed to fetch board data for bus=ahb,vendor=0000,device=0000,subsystem-vendor=0000,subsystem-device=0000,variant=device_model from ath10k/QCA4019/hw1.0/board-2.bin
[   16.706986] ath10k_ahb a800000.wifi: failed to fetch board-2.bin or board.bin from ath10k/QCA4019/hw1.0
[   16.707022] ath10k_ahb a800000.wifi: failed to fetch board file: -12
[   16.707209] ath10k_ahb a800000.wifi: could not probe fw (-12)

```

i used this with ath10k-bdencoder to combine the 2 board files in one board-2.bin file
`./ath10k-bdencoder -c board-2.json -o board-2.bin`

board-2.json
```
[
    {
        "names": [
            "bus=ahb,bmi-chip-id=0,bmi-board-id=20"
        ],
        "data": "bus=ahb,bmi-chip-id=0,bmi-board-id=20.bin"
    },
    {
        "names": [
            "bus=ahb,bmi-chip-id=0,bmi-board-id=21"
        ],
        "data": "bus=ahb,bmi-chip-id=0,bmi-board-id=21.bin"
    },
    {
        "names": [
            "bus=ahb,bmi-chip-id=0,bmi-board-id=20,variant=device_model"
        ],
        "data": "bus=ahb,bmi-chip-id=0,bmi-board-id=20,variant=device_model.bin"
    },
    {
        "names": [
            "bus=ahb,bmi-chip-id=0,bmi-board-id=21,variant=device_model"
        ],
        "data": "bus=ahb,bmi-chip-id=0,bmi-board-id=21,variant=device_model.bin"
    }
]

```
Any idea?