openwrt/openwrt #14404: The device broadcasts on a channel configured in another

castillofrancodamian, 2024-01-15:
### Describe the bug

The error occurs when trying to broadcast on channel 44 with a channel width of 40 MHz. Channel 48 is selected without the user's intention. Furthermore, there is no agreement between LuCI, the configuration files and iw. This occurs on both devices I have tested.

### OpenWrt version

r24806-735efbfb7c

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ramips/mt7621 and ath79/generic

### Device

MERCUSYS MR70X v1 and TP-Link Archer C60 v1

### Image kind

Self-built image

### Steps to reproduce

1. Go to Edit and set channel 44 and bandwidth to 40 MHz.
2. See radio info in LuCI and iw

### Actual behaviour

archer_c60

[image]

[image]

```
root@archer_c60:~# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:00.0'
        option channel '44'
        option band '5g'
        option htmode 'VHT40'
        option cell_density '0'
        option country 'AR'

config wifi-iface '5g_lag'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'LAG'
        option encryption 'psk2+ccmp'
        option ifname '5g_lag'
        option key 'pass'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/ahb/18100000.wmac'
        option channel '11'
        option band '2g'
        option htmode 'HT20'
        option cell_density '0'
        option country 'AR'

config wifi-iface '2g_lag'
        option device 'radio1'
        option network 'lan'
[... 477 lines cut ...]
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.011543] br-lan: port 7(5g_invitados) entered blocking state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.017614] br-lan: port 7(5g_invitados) entered disabled state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.024443] device 5g_invitados entered promiscuous mode
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.034624] br-lan: port 7(5g_invitados) entered blocking state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.040724] br-lan: port 7(5g_invitados) entered forwarding state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.115755] device 5g_invitados left promiscuous mode
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.121555] br-lan: port 7(5g_invitados) entered disabled state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.189366] br-lan: port 7(5g_invitados) entered blocking state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.195416] br-lan: port 7(5g_invitados) entered disabled state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.201784] device 5g_invitados entered promiscuous mode
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.207548] br-lan: port 7(5g_invitados) entered blocking state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.213509] br-lan: port 7(5g_invitados) entered forwarding state
Mon Jan 15 17:36:43 2024 kern.info kernel: [85759.220374] IPv6: ADDRCONF(NETDEV_CHANGE): 5g_invitados: link becomes ready
Mon Jan 15 17:36:43 2024 daemon.notice netifd: Network device '5g_invitados' link is up
Mon Jan 15 17:36:43 2024 daemon.notice hostapd: 5g_lag: interface state HT_SCAN->ENABLED
Mon Jan 15 17:36:43 2024 daemon.notice hostapd: 5g_lag: AP-ENABLED
```

### Expected behaviour

I hope that the device broadcasts on the channel that has been selected.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

castillofrancodamian, 2024-01-15:
> Invalid Target/Subtarget reported. `ramips/mt7621 and ath79/generic` Is this from a supported device?

Yes.

brada4, 2024-01-16:
N and A and up may choose any sub-channel as control channel. The selection reflects in status. (there is NO issue )

castillofrancodamian, 2024-01-16:
@brada4 But does it always choose channel 48 when selecting 44? Because only in those does this happen.

brada4, 2024-01-16:
Less used sub-channel at start is selected for beacons, in 80MHz you have 4 choices and so on. Some drivers report it back others dont. Since you have[ DFS and AUTO-BW](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n92) it is best for all to do auto channel selection.
Probably  you want so stay off weak channel by tweaking ACS like for 5GHz radio in config/wireless
       `list hostapd_options 'min_tx_power=20'`

This injustice just made 15 years, or since you upgraded past wifi-G and wifi-a....

brada4, 2024-01-16:
> hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes

in your log.