openwrt/openwrt #14407: openwrt 22.03 Kernel panic

zhangsd20, 2024-01-16:
### Describe the bug

Switching ap mode to monitor mode ，and using iwinfo immediately will cause the kernel to crash.  This doesn't happen every time, it happens occasionally.  It will not appear if you do not use iwinfo.

log:

[  628.112228] Kernel bug detected[#1]:
[  628.115927] CPU: 0 PID: 9271 Comm: iwinfo Not tainted 5.10.168 #0
[  628.122217] $ 0   : 00000000 00000001 0000007a 00000000
[  628.127627] $ 4   : 80703b40 80703b40 80771aa0 80c09ae0
[  628.133039] $ 8   : ffffefff 00000000 00000000 ffffffea
[  628.138443] $12   : 0000019c 80c09aec 0000019c 80703ba8
[  628.143854] $16   : 82afa900 830d1200 82b453c0 00000000
[  628.149266] $20   : 00000000 822f2000 00000001 82172003
[  628.154678] $24   : 00000018 80384b1c                  
[  628.160081] $28   : 821b2000 80c09c88 00000000 80428880
[  628.165493] Hi    : 000000d8
[  628.168468] Lo    : 453c0000
[  628.171470] epc   : 80428880 skb_panic+0x5c/0x60
[  628.176241] ra    : 80428880 skb_panic+0x5c/0x60
[  628.181003] Status: 1000fc03 KERNEL EXL IE 
[  628.185332] Cause : 10800024 (ExcCode 09)
[  628.189471] PrId  : 00019374 (MIPS 24Kc)
[  628.193520] Modules linked in: ath9k ath9k_common xt_connlimit vport_vxlan vport_gre vport_geneve pppon
[  628.193873]  nsh geneve udp_tunnel ip6_udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbc
[  628.305219] Process iwinfo (pid: 9271, threadinfo=8a389111, task=ee8de1f5, tls=77ed8df0)
[  628.313567] Stack : 8042dc68 806a8774 80599708 00000118 0000000e 82172000 82171ff5 8217210d
[  628.322212]         82173f00 82170000 8225ee00 804288c8 00000000 80599958 82afab40 00000a20
[  628.330857]         82b453c0 80599708 00000000 82afa900 82b453c0 8042dc94 00000000 00000001
[  628.339504]         8225ee00 830d1200 82b453c0 00000000 00000000 80599b5c 0000dceb 8031d16c
[  628.348151]         00000000 ffffffff 00000001 77ed72b0 77db6180 8225ee00 00000000 830d1200
[  628.356797]         ...
[  628.359325] Call Trace:
[  628.361859] [<80428880>] skb_panic+0x5c/0x60
[  628.366270] [<804288c8>] skb_send_sock_locked+0x0/0x238
[  628.371662] 
[  628.373203] Code: 01003825  0c02f343  24848570 <000c000d> 8c8200a4  8c890054  8c8800a0  00451023  0125 
[  628.383291] 
[  628.384889] ---[ end trace 22abfbbcc27a55b6 ]---
[  628.389658] Kernel panic - not syncing: Fatal exception in interrupt
[  628.396220] Rebooting in 3 seconds..

### OpenWrt version

r20063-55a48b0957

### OpenWrt release

22.03-SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

Qxwlan E600G v2 16M

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-16:
Install 22.03.6 as first effort? Via auc to keep all packages.

brada4, 2024-01-16:
What do those vxlans do? Are they on wifi somehow?

closed 2024-01-22 by zhangsd20