openwrt/openwrt #14421: [wiki] Move `LXC in OpenWrt` to virtualization docs

filimonic, 2024-01-17:
### Describe the bug

Move `https://openwrt.org/lxc_openwrt_host` to `https://openwrt.org/docs/guide-user/virtualization/`

### OpenWrt version

0

### OpenWrt release

0

### OpenWrt target/subtarget

0

### Device

0

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

musashino205, 2024-01-18:
This is not a bug of OpenWrt firmware, please post to the Forum.

filimonic, 2024-01-28:
Created forum post here https://forum.openwrt.org/t/move-lxc-in-openwrt-to-virtualization-docs/185961?u=filimonic

closed 2024-01-28 by filimonic