openwrt/openwrt #14423: MediaTek MT7621 ver:1 eco:4 Bug for WiFi

nitinuae, 2024-01-18:
### Describe the bug

The device hangs (not available on http or ssh) when enabling wifi

### OpenWrt version

OpenWrt 23.05.2 r23630-842932a63d

### OpenWrt release

'OpenWrt 23.05.2 r23630-842932a63d'

### OpenWrt target/subtarget

'ramips/mt7621'

### Device

CUDY X6 v2

### Image kind

Official downloaded image

### Steps to reproduce

Cudy X2 is now using MediaTek MT7621 ver:1 eco:4 processor. Earlier with eco:4 processor, there is no issue after enabling the wifi. But as soon as wifi is enabled on eco:4 processor AND wifi get a connection, the device hangs and cannot be reached till factory reset is done. (No ssh no http). I have set Australia as country.

Reading on web - lots of other users have issue with eco:4. Seems a bug.

### Actual behaviour

Enabling the wifi crashes the device

### Expected behaviour

Should not crash

### Additional info

Other people are reporting wifi cannot be enabled for more than 7db or 3db(its in reddit user reporting for xiami device). In my case (Cudy X6 - it crashes)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-18:
Are you trying v1 firmware on v2? 3..7 looks like missing caldata

nitinuae, 2024-01-18:
> Are you trying v1 firmware on v2? 3..7 looks like missing caldata

I am using v2.
Downloaded from firmware selector
openwrt-23.05.2-ramips-mt7621-cudy_x6-v2-squashfs-sysupgrade

I have all-v1, v2(eco:3) and v2(eco:4). 
The eco:4 that I recvd yesterday is behaving like this. I can crash it everytime (done it 4 times already). Cannot give logs as factory reset only brings it back. 
I also tried reflashing it -though I remembered clearing all settings from Cudy built openwrt to openwrt, I tried refalshing by rewriting everything again and clearing settings - still I can crash it when enabling the wifi. It crashes after a few minutes - as I think when wifi client connects. Surprisingly, wifi client remains connected and can communicate internet (I have complex vlan setup - so not sure on this). But the device says connection refused on both ssh and http.

brada4, 2024-01-18:
Can you boot cudy fw in the meantime and try to figure dts additions they added for v2 eco4 device?
it is important as it is unlikely to tell eco3 eco4 apart from outside.

brada4, 2024-01-18:
like
https://github.com/PabloCastellano/extract-dtb
i just presumed you need wifi and suggested to get it eorking first...

brada4, 2024-01-18:
find and compare
`target/linux/platform/*/dts/*`
or vs /proc/device-tree in running cudy system.

nitinuae, 2024-01-18:
Oh - I have around 10 access points throughout the house - I just replaced one AC with this AX - and found the issue.
I can wait for a fix. I think eco:4 devices are just turning up on shelves -so more people will chime in soon.

nitinuae, 2024-01-18:
> find and compare `target/linux/platform/*/dts/*` or vs /proc/device-tree in running cudy system.

root@CudyAtticV2:/proc/device-tree# ls
#address-cells                 cpc@1fbf0000                   leds                           palmbus@1e000000               sdhci@1e130000
#size-cells                    cpuintc                        mc@1fbf8000                    pcie-phy@1e149000              sysclock
aliases                        cpus                           model                          pcie-phy@1e14a000              xhci@1e1c0000
chosen                         ethernet@1e100000              name                           pcie@1e140000
clkctrl                        interrupt-controller@1fbc0000  nand@1e003000                  pinctrl
compatible                     keys                           nficlock                       rstctrl

Which one to compare?

brada4, 2024-01-18:
find cudy file in git clone. Atom Notepad++ or vscode if you need windowes text editor.
https://github.com/openwrt/openwrt/tree/main/target/linux/ramips/dts

nitinuae, 2024-01-18:
Here is the Cudy file (from their website). They are VERY different.
X6 V2 from cudy-`// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "cudy,x6", "mediatek,mt7621-soc";
	model = "CUDY X6";

	aliases {
		led-boot = &led_internet_blue;
		led-failsafe = &led_internet_blue;
		led-running = &led_internet_blue;
		led-upgrade = &led_internet_blue;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 8 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};

	leds {
		compatible = "gpio-leds";

		led_internet_blue: internet_blue {
			label = "blue:internet";
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
		};

		internet_red {
			label = "red:internet";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};
	};

};

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <50000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x10000>;
				read-only;
			};

			partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0xf80000>;
			};

			partition@fd0000 {
				label = "debug";
				reg = <0xfd0000 0x10000>;
				read-only;
			};

			partition@fe0000 {
				label = "backup";
				reg = <0xfe0000 0x10000>;
				read-only;
			};

			bdinfo: partition@ff0000 {
				label = "bdinfo";
				reg = <0xff0000 0x10000>;
				read-only;
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_bdinfo_de00>;
	nvmem-cell-names = "mac-address";
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "lan1";
		};

		port@1 {
			status = "okay";
			label = "lan2";
		};

		port@2 {
			status = "okay";
			label = "lan3";
		};

		port@3 {
			status = "okay";
			label = "lan4";
		};

		port@4 {
			status = "okay";
			label = "wan";
			nvmem-cells = <&macaddr_bdinfo_de00>;
			nvmem-cell-names = "mac-address";
			mac-address-increment = <1>;
		};
	};
};

&state_default {
	gpio {
		groups = "uart3", "jtag";
		function = "gpio";
	};
};

&bdinfo {
	compatible = "nvmem-cells";
	#address-cells = <1>;
	#size-cells = <1>;

	macaddr_bdinfo_de00: macaddr@de00 {
		reg = <0xde00 0x6>;
	};
};
`

and  [openwrt](https://github.com/openwrt/openwrt/tree/main)/[target](https://github.com/openwrt/openwrt/tree/main/target)/[linux](https://github.com/openwrt/openwrt/tree/main/target/linux)/[ramips](https://github.com/openwrt/openwrt/tree/main/target/linux/ramips)/[dts](https://github.com/openwrt/openwrt/tree/main/target/linux/ramips/dts)
/mt7621_cudy_x6-v2.dts
`// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621_cudy_x6.dtsi"

/ {
	compatible = "cudy,x6-v2", "mediatek,mt7621-soc";
	model = "CUDY X6 v2";
};

&partitions {
	partition@50000 {
		compatible = "denx,uimage";
		label = "firmware";
		reg = <0x50000 0xf80000>;
	};

	partition@fd0000 {
		label = "debug";
		reg = <0xfd0000 0x10000>;
		read-only;
	};

	partition@fe0000 {
		label = "backup";
		reg = <0xfe0000 0x10000>;
		read-only;
	};

	partition@ff0000 {
		label = "bdinfo";
		reg = <0xff0000 0x10000>;
		read-only;

		nvmem-layout {
			compatible = "fixed-layout";
			#address-cells = <1>;
			#size-cells = <1>;

			macaddr_bdinfo_de00: macaddr@de00 {
				compatible = "mac-base";
				reg = <0xde00 0x6>;
				#nvmem-cell-cells = <1>;
			};
		};
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_bdinfo_de00 0>;
	nvmem-cell-names = "mac-address";
};

&gmac1 {
	nvmem-cells = <&macaddr_bdinfo_de00 1>;
	nvmem-cell-names = "mac-address";
};

&wifi {
	nvmem-cells = <&macaddr_bdinfo_de00 0>;
	nvmem-cell-names = "mac-address";
};`

nitinuae, 2024-01-18:
Formatting is  fxxed. And there is no way to correct it.
I did use the insert tag - and still the forum software made changes to it.

brada4, 2024-01-18:
Use three backticks before and after code blocks
\`\`\`

nitinuae, 2024-01-18:
Cudy Factory Openwrt Image
///
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "cudy,x6", "mediatek,mt7621-soc";
	model = "CUDY X6";

	aliases {
		led-boot = &led_internet_blue;
		led-failsafe = &led_internet_blue;
		led-running = &led_internet_blue;
		led-upgrade = &led_internet_blue;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 8 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};

	leds {
		compatible = "gpio-leds";

		led_internet_blue: internet_blue {
			label = "blue:internet";
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
		};

		internet_red {
			label = "red:internet";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};
	};

};

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <50000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x10000>;
				read-only;
			};

			partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0xf80000>;
			};

			partition@fd0000 {
				label = "debug";
				reg = <0xfd0000 0x10000>;
				read-only;
			};

			partition@fe0000 {
				label = "backup";
				reg = <0xfe0000 0x10000>;
				read-only;
			};

			bdinfo: partition@ff0000 {
				label = "bdinfo";
				reg = <0xff0000 0x10000>;
				read-only;
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_bdinfo_de00>;
	nvmem-cell-names = "mac-address";
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "lan1";
		};

		port@1 {
			status = "okay";
			label = "lan2";
		};

		port@2 {
			status = "okay";
			label = "lan3";
		};

		port@3 {
			status = "okay";
			label = "lan4";
		};

		port@4 {
			status = "okay";
			label = "wan";
			nvmem-cells = <&macaddr_bdinfo_de00>;
			nvmem-cell-names = "mac-address";
			mac-address-increment = <1>;
		};
	};
};

&state_default {
	gpio {
		groups = "uart3", "jtag";
		function = "gpio";
	};
};

&bdinfo {
	compatible = "nvmem-cells";
	#address-cells = <1>;
	#size-cells = <1>;

	macaddr_bdinfo_de00: macaddr@de00 {
		reg = <0xde00 0x6>;
	};
};
///

Cudy DTS file as seen at github -
///
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621_cudy_x6.dtsi"

/ {
	compatible = "cudy,x6-v2", "mediatek,mt7621-soc";
	model = "CUDY X6 v2";
};

&partitions {
	partition@50000 {
		compatible = "denx,uimage";
		label = "firmware";
		reg = <0x50000 0xf80000>;
	};

	partition@fd0000 {
		label = "debug";
		reg = <0xfd0000 0x10000>;
		read-only;
	};

	partition@fe0000 {
		label = "backup";
		reg = <0xfe0000 0x10000>;
		read-only;
	};

	partition@ff0000 {
		label = "bdinfo";
		reg = <0xff0000 0x10000>;
		read-only;

		nvmem-layout {
			compatible = "fixed-layout";
			#address-cells = <1>;
			#size-cells = <1>;

			macaddr_bdinfo_de00: macaddr@de00 {
				compatible = "mac-base";
				reg = <0xde00 0x6>;
				#nvmem-cell-cells = <1>;
			};
		};
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_bdinfo_de00 0>;
	nvmem-cell-names = "mac-address";
};

&gmac1 {
	nvmem-cells = <&macaddr_bdinfo_de00 1>;
	nvmem-cell-names = "mac-address";
};

&wifi {
	nvmem-cells = <&macaddr_bdinfo_de00 0>;
	nvmem-cell-names = "mac-address";
};
///

nitinuae, 2024-01-18:
I am comparing the V1 and V2 file as provided by UART.
I can see that spi chip is changed,(in V2) and partition layout is also changed a LOT.

V1
``// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/input/input.h>
#include <dt-bindings/gpio/gpio.h>

/ {
	compatible = "cudy-x6", "mediatek,mt7621-soc";
	model = "CUDY X6";

	memory@0 {
		device_type = "memory";
		reg = <0x0 0x10000000>;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 8 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};

	leds {
		compatible = "gpio-leds";

		led_blue: blue {
			label = "blue";
			gpios = <&gpio 15 GPIO_ACTIVE_HIGH>;
		};

		led_red: red {
			label = "red";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};
	};

};

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <80000000>;
		m25p,fast-read;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x10000>;
			};

			debug: partition@1fd0000 {
				label = "debug";
				reg = <0x1fd0000 0x10000>;
			};

			backup: partition@1fe0000 {
				label = "backup";
				reg = <0x1fe0000 0x10000>;
			};

			bdinfo: partition@ff0000 {
				label = "bdinfo";
				reg = <0x1ff0000 0x10000>;
				read-only;
			};

			firmware: partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0x1f80000>;
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&gmac0 {
	mtd-mac-address = <&bdinfo 0xDE00>;
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "lan1";
		};

		port@1 {
			status = "okay";
			label = "lan2";
		};

		port@2 {
			status = "okay";
			label = "lan3";
		};

		port@3 {
			status = "okay";
			label = "lan4";
		};

		port@4 {
			status = "okay";
			label = "wan";
			mtd-mac-address = <&factory 0xA>;
		};
	};
};

&state_default {
	gpio {
		groups = "i2c", "wdt";
		function = "gpio";
	};
};
``

and V2

``
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "mt7621.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
	compatible = "cudy,x6", "mediatek,mt7621-soc";
	model = "CUDY X6";

	aliases {
		led-boot = &led_internet_blue;
		led-failsafe = &led_internet_blue;
		led-running = &led_internet_blue;
		led-upgrade = &led_internet_blue;
		label-mac-device = &gmac0;
	};

	chosen {
		bootargs = "console=ttyS0,115200";
	};

	keys {
		compatible = "gpio-keys";

		reset {
			label = "reset";
			gpios = <&gpio 8 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_RESTART>;
		};

		wps {
			label = "wps";
			gpios = <&gpio 7 GPIO_ACTIVE_LOW>;
			linux,code = <KEY_WPS_BUTTON>;
		};
	};

	leds {
		compatible = "gpio-leds";

		led_internet_blue: internet_blue {
			label = "blue:internet";
			gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
		};

		internet_red {
			label = "red:internet";
			gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
		};
	};

};

&spi0 {
	status = "okay";

	flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <50000000>;

		partitions {
			compatible = "fixed-partitions";
			#address-cells = <1>;
			#size-cells = <1>;

			partition@0 {
				label = "u-boot";
				reg = <0x0 0x30000>;
				read-only;
			};

			partition@30000 {
				label = "u-boot-env";
				reg = <0x30000 0x10000>;
				read-only;
			};

			factory: partition@40000 {
				label = "factory";
				reg = <0x40000 0x10000>;
				read-only;
			};

			partition@50000 {
				compatible = "denx,uimage";
				label = "firmware";
				reg = <0x50000 0xf80000>;
			};

			partition@fd0000 {
				label = "debug";
				reg = <0xfd0000 0x10000>;
				read-only;
			};

			partition@fe0000 {
				label = "backup";
				reg = <0xfe0000 0x10000>;
				read-only;
			};

			bdinfo: partition@ff0000 {
				label = "bdinfo";
				reg = <0xff0000 0x10000>;
				read-only;
			};
		};
	};
};

&pcie {
	status = "okay";
};

&pcie1 {
	wifi@0,0 {
		compatible = "mediatek,mt76";
		reg = <0x0000 0 0 0 0>;
		mediatek,mtd-eeprom = <&factory 0x0000>;
	};
};

&gmac0 {
	nvmem-cells = <&macaddr_bdinfo_de00>;
	nvmem-cell-names = "mac-address";
};

&switch0 {
	ports {
		port@0 {
			status = "okay";
			label = "lan1";
		};

		port@1 {
			status = "okay";
			label = "lan2";
		};

		port@2 {
			status = "okay";
			label = "lan3";
		};

		port@3 {
			status = "okay";
			label = "lan4";
		};

		port@4 {
			status = "okay";
			label = "wan";
			nvmem-cells = <&macaddr_bdinfo_de00>;
			nvmem-cell-names = "mac-address";
			mac-address-increment = <1>;
		};
	};
};

&state_default {
	gpio {
		groups = "uart3", "jtag";
		function = "gpio";
	};
};

&bdinfo {
	compatible = "nvmem-cells";
	#address-cells = <1>;
	#size-cells = <1>;

	macaddr_bdinfo_de00: macaddr@de00 {
		reg = <0xde00 0x6>;
	};
};
``

nitinuae, 2024-01-19:
Any update on this - it needs to be tagged to MT7621 maintainer...

brada4, 2024-01-19:
Not mt7621 but CUDY X6 v2 maintainer , ie one that added initial support.

Ostring, 2024-05-02:
I confirm, my router have the same problem, (LYNKSYS E5600 MediaTek MT7621 ver:1 eco:3 ramips/mt7621), Some times get stuck 23.05.3