openwrt/openwrt #14427: Incorrect opkg flags in rootfs image

jck112, 2024-01-18:
### Describe the bug

When creating the rootfs image, the default packages are installed using `opkg` which [adds](https://github.com/openwrt/opkg-lede/blob/d038e5b6d155784575f62a66a8bb7e874173e92e/libopkg/opkg_install.c#L1198) the `user` flag for most packages[^1].

The `user` flag  should only be applied to packages have been installed by the user and thus should not be applied to the default packages that are part of the system.

For example here are the initial package flags (truncated) from a new SNAPSHOT image on a NanoPi R4S:
```
root@OpenWrt:~# opkg status | awk '/Package:|Status:|^$/1'
Package: libc
Status: install hold installed

Package: mkf2fs
Status: install user installed

Package: opkg
Status: install user installed
...
Package: ppp-mod-pppoe
Status: install user installed
```

[^1]: Packages, like `libc`, which [specify flags](https://github.com/openwrt/openwrt/blob/90644fb2d004112c3d877abd94cd5528d7587ee1/package/libs/toolchain/Makefile#L15), like `hold`, do not get the `user` flag added by default.

### OpenWrt version

r24855-3891355015

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-18:
But they are installed via opkg, unlike libc or kernel meta.

jck112, 2024-01-18:
All packages (including `libc` and `kernel`) are [installed](https://github.com/openwrt/openwrt/blob/90644fb2d004112c3d877abd94cd5528d7587ee1/package/Makefile#L72) by the host using `opkg`.

Some packages, such as `libc`, specify `PKG_FLAGS` which are written into the package's "control" metadata embedded in its ipk file. Due to a quirk in `opkg` the `user` flag is not applied to these packages on install.

The `user` flag is intended to be automatically applied by `opkg` to all packages installed by the user. This makes it easy for users to differentiate what packages were installed as part of the system or as dependencies for other packages.

Thus I don't believe it is correct for the packages installed as part of the image build to have the `user` flag applied.

commit be9023ed (2024-02-16): build: fix opkg flags in rootfs

By default opkg sets the "user" flag when a package is installed,
which resulted in most packages in the rootfs having this flag
set incorrectly. This patch removes the "user" flag from all
installed packages when preparing the rootfs image.

Fixes: #14427
closed 2024-02-16 by openwrt-bot