openwrt/openwrt #14437: Zyxel EX5601-T0 Corrupted Flow Control on OpenWRT

NovaPixell, 2024-01-19:
### Describe the bug

I am writing this issue after several desperate attempts to figure out what happened to the test Zyxel EX5601 (OpenWrt), which I own.

I have a 2.5Gbps connection FTTH with Zyxel EX5601 Router (master router runs OpenWRT, which has no problems). So there are two identical Zyxel, the test unit has a "switch/extend" mode configuration of the master LAN.

MYTH ISSUE:
The WAN port (blue) has severe bandwidth management problems, particularly during upload traffic. I experience this problem only with this port and also by negotiating 1Gbps.

> Link to SpeedTests: https://www.speedtest.net/result/c/ce546dc5-2dc4-4381-8abb-39b8235c0e0e
> https://www.speedtest.net/result/c/14c66f38-65ce-4ce7-a85b-9acb1eca2fc9

Very often coming up with results such as: 2000Mbps DL/10Mbps UL.

> So I wonder what it could be? I have changed any type of cable but nothing..
> This problem appeared, perhaps coincidentally as soon as I tried an OpenWRT [firmware that forcibly disabled Flow Control ](https://github.com/openwrt/openwrt/issues/13122#issuecomment-1699259726)(because of the tests related to the mediatek switch crash... but that's another story 😂)
> 

Clearly after ascertaining the inuity of a firmware without Flow Control (since it was truncating the upload), I put a clean OpenWRT back in, and from the logs I can see that the port negotiates 2.5Gbps correctly with Flow Control... so I have no idea if it could have been this affair that "cracked" the port.

Do you guys have any ideas about what it could be? I think it's a fixable problem, but very strange. Could it be a hardware problem?

### OpenWrt version

r24801-b7f9742da8

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

mediatek/filogic

### Device

Zyxel EX5601-T0 ubootmod

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-20:
Show negotiated link:
`ethtool wan`
Show pause status
`ethtool -a wan`
Disable pauses
`ethtool -A auto off tx off rx off`
Check it is disabled. 

(either of 2  ethtool packages will do)

brada4, 2024-01-20:
you need 1-5% reverse bandwidth for tcp acks.

No, it is a software problem. No need for pauses either.

examine dmesg.
run mii-tool monitor if some renegotiation happens.

For workaround to have device serve you-
If you have free gigabit port you could bypass around fast ports to have it working in kind of smaller router mode.

NerdyProjects, 2024-01-20:
Hi,

there is no way that the "force disable flow control" patch did something permanent to your hardware. Unfortunately, I don't know enough to help you any further, I am sorry.

NovaPixell, 2024-01-20:
> Show negotiated link: `ethtool wan` Show pause status `ethtool -a wan` Disable pauses `ethtool -A auto off tx off rx off` Check it is disabled.
> 
> (either of 2 ethtool packages will do)

Current situation: 
 OpenWrt SNAPSHOT, r24870-ced3fbcda1 (Updated to latest firmware)

```
We have 2 ports (2.5 Gbps) - 1 WAN (eth1) and LAN (working correctly, same chip):
[    4.009656] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 0x886F (release)
[    4.019002] Maxlinear Ethernet GPY211C mdio-bus:06: Firmware Version: 0x886F (release)
[    4.225903] mt7530-mdio mdio-bus:1f lan1 (uninitialized): PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
[   41.761758] mtk_soc_eth 15100000.ethernet eth1: PHY [mdio-bus:06] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
```

**eth1 port is WAN.**

```
root@Zyxel-TEST:/# ethtool eth1
Settings for eth1:
        Supported ports: [ ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: Twisted Pair
        PHYAD: 6
        Transceiver: external
        Auto-negotiation: on
        MDI-X: Unknown
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes
```

```
root@Zyxel-TEST:/# ethtool -a eth1
Pause parameters for eth1:
Cannot get device pause settings: Not supported
```

```
root@Zyxel-TEST:/# ethtool -A auto off tx off rx off
ethtool: bad command line argument(s)
For more information run ethtool -h
```

I thank you in advance for the support, let's figure out what the problem is. Since it is really absurd

NovaPixell, 2024-01-20:
Actual SpeedTest, using eth1 (WAN):  https://www.speedtest.net/result/c/26ca535c-cf7c-433a-a8e9-0a9d5fff2190

brada4, 2024-01-20:
mtk-soc-eth supports ethtool -a/-A on DSA ports, not on direct mii connection. 
Since you are dealing with GigE link partner you may remove unused LAN port from LAN and transfer WAN config to it.
Kind of unfair that cheaper routers work better.

Better speedtest not leaking location https://www.waveform.com/tools/bufferbloat

Suspect duplicate: https://github.com/openwrt/openwrt/issues/14192

NovaPixell, 2024-01-20:
> mtk-soc-eth supports ethtool -a/-A on DSA ports, not on direct mii connection. Since you are dealing with GigE link partner you may remove unused LAN port from LAN and transfer WAN config to it. [...]

If I force the link to negotiate 100Mbps, I have no flow control problems.
With respect to the issue you linked, it seems to be related the problem.

Is it possible that it cannot be solved?

> Since you are dealing with GigE link partner you may remove unused LAN port from LAN and transfer WAN config to it.

What do you mean? You mean to create a custom DTS?

brada4, 2024-01-20:
workaround: remove physical port from br-lan then move all WAN configuration to it. In total ignore misbehaving port.

If you can survive with router disconnected then when port is removed from switch you may loop cat6 cable and check how uncontrollable port reacts to ethtool -a/-A on other side.

Pause can be disabled via DTS (and should be IMO), but other issue seems link flopping in addition.

\@nbd168 is among authors of the driver, and it gets improved often.

NovaPixell, 2024-01-20:
> workaround: remove physical port from br-lan then move all WAN configuration to it. In total ignore misbehaving port.
>
> If you can survive with router disconnected then when port is removed from [...]

Yes, I also tried using the WAN port (eth1) separate from the br-lan and the problem persists.

brada4, 2024-01-20:
Check which port has ethtool -a control permitted. There you can disable pause frames aka flow control.

NovaPixell, 2024-01-20:
> Check which port has ethtool -a control permitted. There you can disable pause frames aka flow control.

I can do this on all lan ports: lan1 (2.5 Gbps), lan2, lan3, lan4

brada4, 2024-01-20:
Ahh understood, eth0 is parent DSA port. It has to be physical port.

NovaPixell, 2024-01-20:
> Ahh understood, eth0 is parent DSA port. It has to be physical port.

eth0 is the entire mtk switch

NovaPixell, 2024-01-20:
Now I tried the 6.1 kernel and the problem still persists...

brada4, 2024-01-20:
You need to disable pause on WAN port and LAN port you are testing.

mrkiko, 2024-01-21:
On a side note / curiosity:
on my gl-mt2500 I can see
[    0.954161] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.111 (0x886F)
[    2.059749] mtk_soc_eth 15100000.ethernet eth0: PHY [mdio-bus:05] driver [Maxlinear Ethernet GPY211C] (irq=POLL)
...

but i remember seeing the printe posted in this thread at the beginnig, like:
[    4.019002] Maxlinear Ethernet GPY211C mdio-bus:06: Firmware Version: 0x886F (release)

was wondering from where the difference in firmware version come from and on what it can depend.

Thanks!

mrkiko, 2024-01-21:
Auto-answering myself: comes from commit
d523f2eb1dadb74d365365eb3bd921ad8b6b2abb in kernel tree.

brada4, 2024-01-21:
@mrkiko do you have flow control negotiated and major performance issues?

NovaPixell, 2024-01-21:
> On a side note / curiosity: on my gl-mt2500 I can see [ 0.954161] Maxlinear Ethernet GPY211C mdio-bus:05: Firmware Version: 8.111 (0x886F) [ 2.059749] mtk_soc_eth 15100000.ethernet eth0: PHY [...]

But the problem arose when I flashed a firmware that forced flow control. Initially I had no problems with it

NovaPixell, 2024-01-21:
> You need to disable pause on WAN port and LAN port you are testing. look at `ip link` dsa ports take from of `lan1@dsa` or `wan@eth0`

I'll test it later. What exactly should I do? I tried all combinations. What I want to do, is try to do a thorough debug on the port's PHY.

brada4, 2024-01-21:
First use `ethtool -a interface` to determine if you can disable flow control
Then add one such interface as wan and other add into br-lan. In principle it is so tragic that you can keep wifi in LAN side and observe it works better than router local stack.