openwrt/openwrt #14438: Trouble with Relay Mode and NextDNS CLI:

viniribeirossa, 2024-01-20:
### Describe the bug

Friends, I'm playing around with the relay mode and authenticated my PPPoE internet on my router. I activated the relay mode on the "wan6" and "guest" interfaces. However, I use NextDNS CLI on both the lan and guest interfaces. Unfortunately, when I activated the relay mode on the "guest" interface, it overrode the NextDNS DNS settings.

When I activate the relay mode, the guest interface only recognizes the DNS from the wan6 interface. However, as I mentioned before, I use NextDNS CLI, and I want to continue using it on the guest interface alongside the relay mode. I try using 'dhcp-option: 6,172.172.172.1' (IP address of my guest interface), but it has no effect. The NextDNS CLI doesn't recognize it unless I deactivate the relay mode on the guest interface.

The question is: How can I use relay mode and NextDNS CLI together?

Sorry, but it's impossible to receive help in the fórum

### OpenWrt version

DISTRIB_ID='OpenWrt' DISTRIB_RELEASE='22.03.6' DISTRIB_REVISION='r20265-f85a79bcb4' DISTRIB_TARGET='ramips/mt7621' DISTRIB_ARCH='mipsel_24kc' DISTRIB_DESCRIPTION='OpenWrt 22.03.6 r20265-f85a79bcb4'

### OpenWrt release

22.03.06

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Link DIR-878 A1

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-20:
You need to add DNS dnsmasq forwardings for domains you want to resolve without DoH proxy, like
/ntp.org/1.1.1.1
There is no macro for DHCP-acquired DNS server, so well known servers are easier choice.

viniribeirossa, 2024-01-20:
I think you didn't understand it correctly!

I have a WAN interface that authenticates the connection with PPPoE. On my router, I use NextDNS CLI on the GUEST and LAN interfaces, and everything works fine. However, when I activate relay mode on the WAN6 and GUEST interfaces, the NextDNS CLI settings no longer work for the GUEST internet; the DNS from relay mode takes precedence.

viniribeirossa, 2024-01-20:
See
[image]
[image]
[image]
[image]
[image]

brada4, 2024-01-20:
DHCP and DNS -> General Settings -> DNS forwardings
Add 172.x.y.z as DNS server (or 127.1$12345 if you really use NextDNS CLI proxy.
-> Resolv and hosts files -> Ignore resolv file

brada4, 2024-01-20:
You are asking in very tangled way, principal setup is like:
https://openwrt.org/docs/guide-user/services/dns/stubby

viniribeirossa, 2024-01-20:
It doesn't seem to work with relay mode active on the interface 

[image]
[image]
[image]
[image]

viniribeirossa, 2024-01-20:
Problem solved!

For anyone who ever has this problem, the fix is ​​simple!

When we use relay mode, as the name suggests, it retransmits everything to the destination interface, if you want to use another DNS other than the one relayed by relay mode, it's simple...

Open the target interface, in my case it is guest

_Guest (interface) > Edit > DHCP Server >  IPv6 Settings_, you will see the option **RA-Service** leave it in **server mode** and disable the option **Local IPv6 DNS server**, the rest of the options change to relay mode

@brada4 Thank you for your effort in trying to help me!

brada4, 2024-01-20:
You can add dnat rule on lan side to port 53 nat to gw port 53 (both tcp and udp)

viniribeirossa, 2024-01-20:
Could you show me how to make the rules? Commands?

brada4, 2024-01-20:
```
fw4 print
...
    define lan_devices = { "br-lan", "br-iot"}
...
       chain dstnat {
                type nat hook prerouting priority dstnat; policy accept;
        }
```
/etc/nftables.d/dnsnat.nft
```
chain capturedns {
    type nat hook prerouting priority dstnat; policy accept;
    iifname $lan_devices meta l4proto {tcp,udp} dport 53 counter dnat to 192.168.1.1:53
}
```

v22 cannot syntax check include tables, likely you need to correct something.