openwrt/openwrt #14440: realtek: cpu port unable to send or receive any traffic when tagged

jwh, 2024-01-20:
### Describe the bug

After a376508216440178184fb3ab71faf87eea637109 and/or e691e2b302d98d4239ffbfced0759384592ea995, when the config is for example:

```
config bridge-vlan
        option device 'switch'
        option vlan '10'
        list ports 'lan1:u*'
```

Traffic will not be passed in either direction, it will only work with the default config which is PVID 1, however reverting either of the above commits will also break that.

Reverting both commits allows previous configuration to work:

```
config bridge-vlan
        option device 'switch'
        option vlan '10'
        list ports 'lan1:u*'

config interface 'lan'
        option device 'switch.10'
        option proto 'dhcp'
```

On their own the commits seem innocuous but it seems that either the assumptions are wrong or the chip behaves in a way that other DSA switches do not...

Have a bunch of GS1900s, however I've only upgraded a new GS1900 24 v2 and as such, this is the only one showing this problem but as the switches are the same as smaller models I expect if I upgrade those they will also break, but they're in use so I cannot realistically do that (yet).

### OpenWrt version

ced3fbcda18e200315cb077f7e2b362ae52ee065

### OpenWrt release

ced3fbcda18e200315cb077f7e2b362ae52ee065

### OpenWrt target/subtarget

. /etc/openwrt_release && echo $DISTRIB_TARGE

### Device

ZyXEL GS1900-24 v1 (actually a v2 but seems identical)

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jwh, 2024-01-20:
> Invalid Target/Subtarget reported. `. /etc/openwrt_release && echo $DISTRIB_TARGE` Is this from a supported device?

Copy paste fail, can someone remove the invalid label?

closed 2024-01-26 by jwh

Neustradamus, 2024-12-22:
Good news, there is a new merged PR linked to Zyxel GS1900:
- https://github.com/openwrt/openwrt/pull/16442