openwrt/openwrt #14441: Can we switch the default ZRam swap algorithm to zstd?

Wabuo, 2024-01-20:
### Describe the bug

It seems to be the standard in most desktop distributions because of better overall performance.

Is there any reason that doesn't apply on embedded devices? (If so I'd love a one line why for curiosity sakes but feel fre to just close the issue)
As far as I understand the topic, zstd is the better choice except if you are massively cpu locked.

Or is the default of lzo-rle just an outdated artefact in the menuconfig that got forgotten?

I'm happy to write a PR, but need to be pointed at some documentation on how to influence menuconfig defaults.

### OpenWrt version

23.05.3

### OpenWrt release

23.05.3

### OpenWrt target/subtarget

-

### Device

R4S

### Image kind

Self-built image

### Steps to reproduce

-

### Actual behaviour

-

### Expected behaviour

-

### Diffconfig

```text
-
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-20:
zstd needs more cpu time on already slow CPU.

rany2, 2024-01-20:
>It seems to be the standard in most desktop distributions because of better overall performance.

`zstd` is not better for performance, it will probably have a higher compression ratio but it will be slower. Most devices that need zswap/zram probably have a really weak CPU as well, so it's probably better to stick to a more simple algorithm like OpenWRT does right now.

brada4, 2024-01-20:
Either way it saves the day whem you try to stuff 64MB of processes in 64MB RAM.

https://www.reddit.com/r/Fedora/comments/mzun99/new_zram_tuning_benchmarks/

rany2, 2024-01-20:
It could be worth setting the default compression algorithm based on architecture, but that seems like a lot of work for something that the user could change themselves if the default compression algorithm causes issues for them.

Djfe, 2024-02-16:
zstd can be quite fast on decompression while having a way better compression ratio than lz4.
On Desktop it's definitely worth it, especially since it requires less cycles copying data from zram to cpu (to ram) since the data is smaller. that might be different for embedded devices.

brada4, 2024-02-16:
The script forces lzo by default in place of upstream default (faster) lzo-rle btw.

reukiodo, 2024-03-01:
According to that post https://www.reddit.com/r/Fedora/comments/mzun99/new_zram_tuning_benchmarks/ it seems that lz4 is superior to lzo-rle on any page cluster size (what page cluster size does OpenWrt use?), so I attempted to change this to lz4 in the /etc/config/system:
```
root@U35WF:/etc/init.d# cat /etc/config/system 

config system
        option hostname 'U35WF'
        option ttylogin '0'
        option log_size '64'
        option urandom_seed '0'
        option description 'KIMAX Blueendless U35WF'
        option notes 'single 12TB 3.5" hard drive NAS'
        option zonename 'UTC'
        option log_proto 'udp'
        option conloglevel '8'
        option cronloglevel '5'
        option zram_comp_algo 'lz4'
        option zram_size_mb '128'

config timeserver 'ntp'
        list server '0.openwrt.pool.ntp.org'
        list server '1.openwrt.pool.ntp.org'
        list server '2.openwrt.pool.ntp.org'
        list server '3.openwrt.pool.ntp.org'

config led 'led_eth'
        option name 'ETH'
        option sysfs 'green:eth'
        option trigger 'netdev'
        option mode 'link tx rx'
        option dev 'eth0'

config led 'led_wifi_led'
        option name 'wifi'
        option sysfs 'blue:wifi'
        option trigger 'netdev'
        option mode 'link tx rx'
        option dev 'wlan0'

root@U35WF:/etc/init.d#
```
and upon reboot it seems to set it ? (what is the sh write error?)
```
Thu Feb 29 17:42:05 2024 daemon.debug zram_start: activating '/dev/zram0' for swapping (128 MiB)
Thu Feb 29 17:42:05 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_start: activating '/dev/zram0' for swapping (128 MiB)
Thu Feb 29 17:42:05 2024 daemon.debug zram_reset: enforcing defaults via /sys/block/zram0/reset
Thu Feb 29 17:42:05 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_reset: enforcing defaults via /sys/block/zram0/reset
Thu Feb 29 17:42:05 2024 daemon.debug zram_comp_algo: set compression algorithm 'lz4' for zram '/dev/zram0'
Thu Feb 29 17:42:05 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_comp_algo: set compression algorithm 'lz4' for zram '/dev/zram0'
Thu Feb 29 17:42:06 2024 daemon.notice procd: /etc/rc.d/S15zram: sh: write error: Invalid argument
Thu Feb 29 17:42:06 2024 kern.info kernel: [   56.566951] zram0: detected capacity change from 0 to 262144
Thu Feb 29 17:42:06 2024 daemon.notice procd: /etc/rc.d/S15zram: Setting up swapspace version 1, size = 134213632 bytes
Thu Feb 29 17:42:06 2024 kern.info kernel: [   56.694292] Adding 131068k swap on /dev/zram0.  Priority:100 extents:1 across:131068k SSDsc
```
but when I check it still shows `[lzo-rle]`:
```
root@U35WF:~# /etc/init.d/zram status

Gathering stats info for zram device "zram0"

ZRAM
----
Block device              - /sys/block/zram0
Device size               - 128 MiB
Compression algo          - lzo [lzo-rle] lz4 lz4hc 842 zstd 

DATA
----
Original data size        - 0.00 MiB
Compressed data size      - 0.00 MiB
Compress ratio            - 78.77 

MEMORY
------
Memory used, total        - 0.00 MiB
Allocator overhead        - 0.00 MiB
Allocator efficiency      - 1.27 %
Maximum memory ever used  - 0.00 MiB
Memory limit              - 0.00 MiB

PAGES
-----
Same pages count          - 0
Pages compacted           - 0
Free pages discarded      - 32767
root@U35WF:~# 
```
Is this a bug?

reukiodo, 2024-03-01:
I think it is the line which attempts to echo the new setting into the `/dev/block/zram0/comp_algorithm`:
```
root@U35WF:~# echo lz4 > "/sys/block/$( basename /dev/zram0 )/comp_algorithm"
ash: write error: Invalid argument
root@U35WF:~# echo 'lz4' > "/sys/block/$( basename /dev/zram0 )/comp_algorithm"
ash: write error: Invalid argument
root@U35WF:~# echo "lz4" > "/sys/block/$( basename /dev/zram0 )/comp_algorithm"
ash: write error: Invalid argument
root@U35WF:~# 
```

TalsLabs, 2024-03-07:
@reukiodo  I also faced the same problem, I searched around and saw [another user on OpenWRT forum solve it](https://forum.openwrt.org/t/version-23-05-0-rc1-does-not-work-zram-swap/162921). We just have to  `opkg install kmod-lib-lz4 kmod-lib-zstd` according to which compression algorithm you want. 

Thanks for letting me know about these settings and command too, I just found out in the new OpenWRT build, that there is a setting now under the **System** tab, that can do the same thing.
>option zram_comp_algo 'lz4'
  option zram_size_mb '128'
  
> /etc/init.d/zram status

[image]

reukiodo, 2024-03-11:
That was the missing piece! I am glad it's a simple fix and even more happy that it's already there in LuCI to make it easier!

```
Thu Feb 29 17:42:09 2024 daemon.debug zram_start: activating '/dev/zram0' for swapping (128 MiB)
Thu Feb 29 17:42:09 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_start: activating '/dev/zram0' for swapping (128 MiB)
Thu Feb 29 17:42:09 2024 daemon.debug zram_reset: enforcing defaults via /sys/block/zram0/reset
Thu Feb 29 17:42:09 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_reset: enforcing defaults via /sys/block/zram0/reset
Thu Feb 29 17:42:10 2024 daemon.debug zram_comp_algo: set compression algorithm 'lz4' for zram '/dev/zram0'
Thu Feb 29 17:42:10 2024 daemon.notice procd: /etc/rc.d/S15zram: zram_comp_algo: set compression algorithm 'lz4' for zram '/dev/zram0'
Thu Feb 29 17:42:10 2024 kern.info kernel: [   99.110343] zram0: detected capacity change from 0 to 262144
Thu Feb 29 17:42:10 2024 daemon.notice procd: /etc/rc.d/S15zram: Setting up swapspace version 1, size = 134213632 bytes
Thu Feb 29 17:42:10 2024 kern.info kernel: [   99.229428] Adding 131068k swap on /dev/zram0.  Priority:100 extents:1 across:131068k SSDsc
```

reukiodo, 2024-03-31:
As far as OpenWrt is concerned, the default being lzo somewhat makes sense as it takes the least amount of storage space, though lz4 makes much more sense when storage space can spare an extra ~50KB or so.

reukiodo, 2025-02-09:
As I understand, the higher priority is utilized first, and once it is full, then uses the lower priority. Therefor the higher priority one should use fastest compression and lower priority should use strongest compression. The zstd compression can be either fast (level=1) or strong (level=22) so it could be used for both.

Since storage size (compression library) is usually a major concern, it might make more sense to use zstd then for both lower (stronger level 22) and higher (faster level 1) priority. This would use the same library, which means kmod-lib-lzo wouldn't be necessary, saving ~5K. Not much, but still...

It would actually be really cool if the swapping algorithm could automatically adjust the compression levels based on the memory's usage frequency.

reukiodo, 2025-02-21:
I'd love to see a secondary dropdown in the GUI to select compression level when zstd is selected. It would be nice to set much higher compression (9) on devices that have the CPU grunt for it, and select lower (1) compression on weaker CPUs.