openwrt/openwrt #14443: Saner defaults for NanoPI R4S (4GB) and other powerful, big flash/sd-card driven devices. [discussion]

Wabuo, 2024-01-20:
### Describe the bug

I'm happy to write up a PR, or well, probably a few, but want to narrow things down first and feel out opinions, instead of doing just something and probably wasting my and other people's time.

Specifically, I would love to change the following things:
(preferably in the next minor release)
1. Change root file system to 1GB (or more?)
-- seems to be a sane default for heaps of space for packages, while leaving lots of space on a 32GB SD for an extra partition for eg. /opt/ or /srv/. This will force fewer people to use the image builder and come on if you are using a SD-card smaller than 2GB than you are the niche case and will have to use the image builder.
-- currently, it's fairly easy to run in to size constraints using the firmware selector on the download page, which is a bit ironic with this kind of device.
2. Have a default ZRam with 2GB zstd swap, seems a good default, and it can easily be changed in LUCI.
-- having some swap on this kind of device makes a lot of sense, and a zram based one will increase performance over no swap and putting one on the SD card is well just stupid.
3. Enable every kernel setting needed to pass https://github.com/moby/moby/blob/master/contrib/check-config.sh for container support from the kernel. Preferably including AppArmor & SELinux support.
-- this will mean Docker, Podman and LXC will work out of the box and won't require a custom kernel any more.
-- useful for things like a unify controller, IP camera motion tracking or a local Prometheus/Grafana setup to monitor the local net or ...
-- this seems to be a much sought after use case for this device, and is probably the main reason for heaps of custom-builds, and we definitely have the space and resources on this device. 
-- are there negative, real live security implications, worth considering?
4. Probably the most controversial: increase, the CPU clock rate to the one advertised and used by the vendor, eg. 2.0GHz.
-- Yes, the device is run and advertised slight overclocked by the vendor, I see no reason not to follow that.
-- after all, that's how they designed, sell and warrant their product.
-- https://wiki.friendlyelec.com/wiki/index.php/NanoPi_R4S#Hardware_Spec

There are other things like making `zoneinfo-all` and maybe `htop` part of the defaults for this device, but all that can easily be dealt with by using the firmware selector on the download page.

So what are people's thoughts about me working towards integrating these changes?

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-20:
Think about supported 400MHz MIPS device with 8MB flash and 64MB RAM.

jow-, 2024-01-20:
There's a strong sentiment against per-device feature package selections (beyond preselecting stuff like required kmods). Any change done has to cater to the lowest common denominator of the entire subtarget, no device specific kernel configs or non-essential package selections. 

1) Change rootfs size - probably okay, though effort is likely better spent in implementing automatic online-resizing
2) Automatic zram - non-standard feature deviating from OpenWrt baseline
3) Kernel with required container features - probably okay as well, if all supported devices of this subtarget can handle the increased kernel image size
4) No opinion on that, but likely fine if the vendor firmware does the same

brada4, 2024-01-21:
4. you get cpufreq adjusted frequency from /proc/cpuinfo. Be happy you have that kind of power saving.

mrkiko, 2024-01-21:
Having per-device packages and defaults would be something I am open for discussion.
Otherwise I wouldn't implement these changes in upstream.
Just my opinion.

mrkiko, 2024-01-21:
And it would ben ice to see if there are more parts that can be set dynamically instead of statically - e.g.: tune some sysctl depending on RAM size and so on.

Wabuo, 2024-01-22:
> Think about supported 400MHz MIPS device with 8MB flash and 64MB RAM.

Oh I am, but I think it makes no sense to run either with a setup optimised for the other.

So I'm trying to find a decent way to cater for both.

Wabuo, 2024-01-22:
> Having per-device packages and defaults would be something I am open for discussion. Otherwise I wouldn't implement these changes in upstream. Just my opinion.

I like the idea, but it could turn in a maintenance hell as in every release somebody needs to make sure these defaults are still sane.
- As I said, I'm all for sane defaults and love the idea of per device sane baseline defaults, but I'm super new to the OpenWrt community and can't even phantom the possible maintenance overhead this might create!
- It also kinda sounds like it could end up being a debugging mess, but I guess in reality a lot of devices run with a bunch of none default packages and even kernels and those bugs that show up because of it are still valid bugs.

So if those are none or manageable concerns, I'd love to do the legwork to get it done for the R4S.

What other options are there? Because currently the majority of users with this and similar devices are running 3dparty custom-builds or their own, which isn't ideal.

Wabuo, 2024-01-22:
> There's a strong sentiment against per-device feature package selections (beyond preselecting stuff like required kmods). Any change done has to cater to the lowest common denominator of the entire subtarget, no device specific kernel configs or non-essential package selections.

Fair enough, I can kinda see why, I assume that doesn't apply to the rootfs size.

> 1. Change rootfs size - probably okay, though effort is likely better spent in implementing automatic online-resizing

I actually really like having a 3d partition that gets left alone while flashing a new image through `sysupgrade`, if the rootfs size dint change, especially because there's no easy way to back up something like 20+GB in the regular update process.
But that's kinda the standard case now, with 32GB SD-cards now being the average smallest available standard size.
So I'm more in favour of an automatic 3d partition mounted at lets say /opt/ with automatic online-resizing (excluding a few hundred MB at the end to be compatible with slight differences in SD size and restoring backup images of the whole SD).
Which could be combined with auto online-resizing of rootfs up to a sane limit of, I don't know, 1GB or 2GB.
Quite frankly, I don't see a use case for a 30 something GB rootfs on the front of having room for packages, but using that space for container images volumes and or websites seems useful.
Also how would an auto-resizing rootfs play along with the firmware selector on the download page.
It also easily allows compressing that 3d partition without it directly affecting the "main system", for example mine is currently mounted with `rw,compress_algorithm=zstd:6,compress_chksum,atgc,gc_merge,lazytime` 
- A lot of this needs to be discussed In sub issues, as soon as things get more concrete, but I wanted to put the general thoughts on the table.

> 2. Automatic zram - non-standard feature deviating from OpenWrt baseline, unlikely to go in - at least not in an uncoordinated manner as it would lead to a chaotic proliferation of device specific default package selections and features

So it basically boils down to implement a canonical change with overall sane defaults that can autoconfigure.
Or to just leave it as an after install customisation, which again could include an automatic sane setup (to be implemented)
> 3. Kernel with required container features - probably okay as well, if all supported devices of this subtarget can handle the increased kernel image size

They most likely can, but guess I'll go do some digging.
- On a sidenote, some of the needed Kernel features, well more some of the nice to have ones, aren't exposed through menueconfig can they just be added or is this for some reason a touchy subject or even a no-go?
-- specifically (will be edited in, in a sec)

> 4. No opinion on that, but likely fine if the vendor firmware does the same

Sweet, this results in a device specific patch overriding the subtarget CPU freq but an otherwise identical kernel.

Wabuo, 2024-01-22:
> 4. you get cpufreq adjusted frequency from /proc/cpuinfo. Be happy you have that kind of power saving.

I don't think power saving is relevant, as in the CPU governor can reduce clock speed anyways and more importantly the device takes roughly as much energy as a charging phone while competing with x86 machines.

So I personally would love to even overclock it to the same point as ImortalWRT because any bit of "umpf" I can squeeze out means power saved by not having to run an x86 machine. But that's kinda totally besides the point.

In my opinion, stock OpenWRT should follow stock device vendor.
I'm pretty sure that's the rule anyway.

And sure the ability for online over and under -clocking would be awesome as, but again kinda out of the scope of this discussion.

brada4, 2024-01-22:
You added that to the scope. Openwrt follows mainline kernel , adds some hardware support, then vendor or immortalwrt does same. Downstreams are usually nice enough to point to upstreams' defects.
If you can find immortalwrt commit/pr about overclocking or disabling cpufreq it may describe valid reasons to do so to share here.

Wabuo, 2024-01-22:
I think the commint that introduce it is here https://github.com/immortalwrt/immortalwrt/commit/4c139c05881e760185002de234cae8b32e66e7af

Its still the same patch today:
-- 23.05 https://github.com/immortalwrt/immortalwrt/blob/openwrt-23.05/target/linux/rockchip/patches-5.15/992-rockchip-rk3399-overclock-to-2.2-1.8-GHz.patch
-- master https://github.com/immortalwrt/immortalwrt/blob/master/target/linux/rockchip/patches-6.1/992-rockchip-rk3399-overclock-to-2.2-1.8-GHz.patch

Couldt find any explanation except the one given in the patch itself "It's stable enough to overclock cpu frequency to 2.2/1.8 GHz, and for better performance.", but maybe your github fuu is better than mine ... 

Given that the pach is 3 years old thers ither no ImortalWRT users with an R4S who botherd to complain or they are happy.

brada4, 2024-01-22:
> enough
> quite often

Stable operation is paramount.