openwrt/openwrt #14444: bump hostapd version

ascendbeing, 2024-01-20:
### Describe the bug

https://github.com/openwrt/openwrt/commit/56d7887917102877ed2f03414f7ed812a29d6b39

this was done but version wasn't bumped. since it's a .c file it's not something end user can easily install. they would have to recompile manually. if you can version bump hostapd it will get distributed further and people will be able to try dfs with perhaps more reliability, bc people using snapshot will be able to upgrade the hostapd to get the fix

### OpenWrt version

master 

### OpenWrt release

master 

### OpenWrt target/subtarget

all

### Device

any

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2024-01-20:
This isn't invalid, it's just globally affecting. it's not invalid, it affects every single sub target and target and at least everything on master

kirdesde, 2024-01-20:
> This isn't invalid, it's just globally affecting. it's not invalid, it affects every single sub target and target and at least everything on master

No, it's not. It's a hostapd patch, it's automatically applied during the hostapd package build.
No version bump needed.

rany2, 2024-01-20:
@kirdesde the version bump is needed to force a rebuild if you aren't doing a clean build.

brada4, 2024-01-20:
You have to post accurate version tags to confirm you are not a robot....