openwrt/openwrt #14445: Failure to bring up radio with syntax error in netifd

wpisarski, 2024-01-20:
### Describe the bug

I did a fresh install (openwrt-23.05.2-x86-64-generic-ext4-combined-efi.img), set up the basics, installed kmod-mt7921e and tried to set up WiFi in client mode. I can scan the networks and join my wireless network, but resulting device and interface fail to get IP and work. Updating all packages and rebooting did not help

Wireless overview says for radio0:
```
AMD RZ608 802.11ac/ax/b/g/n
Channel: 1 (? GHz) | Bitrate: ? Mbit/s
```
Network created when joining my wireless network says my SSID and in the next line:
```
Wireless is not associated
```

Created wwan interface in Interfaces screen:
```
Protocol: DHCP client
RX: 0 B (0 Pkts.)
TX: 0 B (0 Pkts.)
Error: Network device is not present
```

Relevant part of system log (syntax error part is repeated 3 times, omitted for readability):
```
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265): WARNING: Variable 'data' does not exist or is not an array/object
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265): Command failed: Not found
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265): Command failed: Not found
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265): Syntax error: Unable to compile module '/usr/share/hostap/common.uc':
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   | Syntax error: Unable to resolve path for module 'nl80211'
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   | In /usr/share/hostap/common.uc, line 1, byte 35:
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |  `import * as nl80211 from "nl80211";`
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |   Near here ------------------------^
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   | Syntax error: Unable to resolve path for module 'rtnl'
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   | In /usr/share/hostap/common.uc, line 2, byte 29:
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |  `import * as rtnl from "rtnl";`
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   |   Near here ------------------^
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265): In /usr/share/hostap/wdev.uc, line 3, byte 137:
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):  `import { vlist_new, is_equal, wdev_create, wdev_set_mesh_params, wdev_remove, wdev_set_up, phy_open } from "/usr/share/hostap/common.uc";`
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):   Near here ------------------------------------------------------------------------------------------------------------------------------^
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):
Sat Jan 20 23:39:09 2024 daemon.notice netifd: radio0 (2265):

...

Sat Jan 20 23:39:09 2024 daemon.notice netifd: Wireless device 'radio0' is now up
```

Wireless device:
```
03:00.0 Network controller [0280]: MEDIATEK Corp. RZ608 Wi-Fi 6E 80MHz [14c3:0608]
	Subsystem: MEDIATEK Corp. RZ608 Wi-Fi 6E 80MHz [14c3:0608]
	Flags: bus master, fast devsel, latency 0, IRQ 47, IOMMU group 2
	Memory at d0800000 (64-bit, prefetchable) [size=1M]
	Memory at d0900000 (64-bit, prefetchable) [size=16K]
	Memory at d0904000 (64-bit, prefetchable) [size=4K]
	Capabilities: [80] Express Endpoint, MSI 00
	Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Kernel driver in use: mt7921e
```

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

x86/64

### Device

HP HP t730 Thin Client

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-20:
It says you are missing or mismatched ucode-mod-rtnl

wpisarski, 2024-01-20:
> It says you are missing or mismatched ucode-mod-rtnl

I've installed this package and rebooted, but there's no change in behavior or logs.

jow-, 2024-01-20:
You'll require `ucode-mod-nl80211` too.

wpisarski, 2024-01-20:
Installing that second package reduced the log, but no change elsewhere and no connectivity:

```
Sun Jan 21 00:52:45 2024 daemon.notice netifd: radio0 (2505): WARNING: Variable 'data' does not exist or is not an array/object
...
Sun Jan 21 00:52:45 2024 daemon.notice netifd: radio0 (2505): Command failed: Not found
Sun Jan 21 00:52:45 2024 daemon.notice netifd: radio0 (2505): Command failed: Not found
...
Sun Jan 21 00:52:45 2024 daemon.notice netifd: Wireless device 'radio0' is now up
```

wpisarski, 2024-01-21:
I think I found something. If I scan and join the network, then a new wireless network and interface are created. But if I go to device configuration for this network, then I get `WPA-Encryption requires wpa_supplicant (for client mode) or hostapd (for AP and ad-hoc mode) to be installed.` in `Wireless Security` tab and the only option in the dropdown is `No Encryption (open network)`

But then trying to install `hostapd` results in:
```
# opkg install hostapd
Installing hostapd (2023-09-08-e5ccbfc6-6) to root...
Downloading https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/base/hostapd_2023-09-08-e5ccbfc6-6_x86_64.ipk
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package hostapd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package hostapd.
```

This happens every time when I try to install it. I've tried `hostapd-mini` and `hostapd-openssl`, they can be installed, but don't change this situation.

brada4, 2024-01-21:
you need supplicant for wwan, or wpad for both.

wpisarski, 2024-01-21:
After installing `wpa-supplicant` and `hostapd-basic-openssl` I was indeed able to connect to WiFi. So I guess that is solved. Would be nice if LuCI failed at the stage of joining a secured network.

A side issue is why hostapd fails to install because of checksum mismatch. Do you think I should report this separately?

The package in question, downloaded from https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/base/hostapd_2023-09-08-e5ccbfc6-6_x86_64.ipk
```
# sha256sum hostapd_2023-09-08-e5ccbfc6-6_x86_64.ipk 
b690664d1842beb3d7b6ebc00aa2c6999f7487ba7b9c9d081af45954735c58eb  hostapd_2023-09-08-e5ccbfc6-6_x86_64.ip
```

brada4, 2024-01-21:
checksum fixes over time, run opkg update like in hour and check again.

closed 2024-01-21 by wpisarski

ynezz, 2024-01-23:
>  * opkg_install_pkg: Checksum or size mismatch for package hostapd. Either the opkg or the package index are corrupt. 
> checksum fixes over time, run opkg update like in hour and check again.

Perhaps some weird CDN issues? Would it be possible to [check it next time](https://docs.fastly.com/en/guides/checking-cache#using-a-fastly-debug-header-with-curl) and provide output of following command:

```
curl -svo /dev/null -H "Fastly-Debug:1" https://downloads.openwrt.org/releases/23.05.2/packages/x86_64/base/hostapd_2023-09-08-e5ccbfc6-6_x86_64.ipk
```

Cc: @aparcar

vgdh, 2024-08-28:
I used these packages and it finally works as access point
```
rt2800-usb-firmware
kmod-rt2800-usb
ucode-mod-nl80211
wpad 
```

brada4, 2024-08-28:
Wifi support is absent from x86 images, probably somebody should write a guide how to "install wifi" like wpad and identify driver package(s) from lspci/lsusb.

oliv99, 2024-09-30:
thanks for this post which save my life :-)

I have installed a generic x86-64 in a proxmox virtual machine
I have Alpha AWUS036ACM with MT7612U Chipset as USB passthrough 

I have install the following;
opkg install ucode-mod-nl80211 ucode-mod-rtnl hostapd

[image]

Everything is working as expected !