openwrt/openwrt #14449: MAP-T not making use of full port range allocated

davesbell, 2024-01-21:
### Describe the bug

I'm attempting to get a working MAP-T configuration working on a Raspberry Pi. I've had some success.

My BNG is sending the DHCP Option 95 to the CPE with what appears to be a valid configuration.

```
rule=type=map-t,ealen=10,prefix4len=28,prefix6len=38,ipv4prefix=192.0.2.0,ipv6prefix=2001:db8:4000::,dmr=2001:db8::/64,
RULE_1_FMR=0
RULE_1_EALEN=10
RULE_1_PSIDLEN=6
RULE_1_OFFSET=6
RULE_1_PREFIX4LEN=28
RULE_1_PREFIX6LEN=38
RULE_1_IPV4PREFIX=192.0.2.0
RULE_1_IPV6PREFIX=2001:db8:4000::
RULE_1_IPV6PD=2001:db8:4011::
RULE_1_PD6LEN=48
RULE_1_PD6IFACE=wan6
RULE_1_IPV6ADDR=2001:db8:4011::c00:200:11
RULE_BMR=1
RULE_1_IPV4ADDR=192.0.2.0
RULE_1_ADDR4LEN=32
RULE_1_PORTSETS='1296-1311 2320-2335 3344-3359 4368-4383 5392-5407 6416-6431 7440-7455 8464-8479 9488-9503 10512-10527 11536-11551 12560-12575 13584-13599 14608-14623 15632-15647 16656-16671 17680-17695 18704-18719 19728-19743 20752-20767 21776-21791 22800-22815 23824-23839 24848-24863 25872-25887 26896-26911 27920-27935 28944-28959 29968-29983 30992-31007 32016-32031 33040-33055 34064-34079 35088-35103 36112-36127 37136-37151 38160-38175 39184-39199 40208-40223 41232-41247 42256-42271 43280-43295 44304-44319 45328-45343 46352-46367 47376-47391 48400-48415 49424-49439 50448-50463 51472-51487 52496-52511 53520-53535 54544-54559 55568-55583 56592-56607 57616-57631 58640-58655 59664-59679 60688-60703 61712-61727 62736-62751 63760-63775 64784-64799 '
RULE_1_DMR=2001:db8::/64
RULE_COUNT=1
```

I have no IPv4 being issued over the WAN, and my routing table looks good.

```
root@OpenWrt:~# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         0.0.0.0         0.0.0.0         U     0      0        0 map-wan6_4
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 br_int
192.168.2.0     0.0.0.0         255.255.255.0   U     0      0        0 phy0-ap0
```

Doing a wget from the command line, I get a working connection

```
root@OpenWrt:~# wget -O /dev/null -4 http://bbc.co.uk
Downloading 'http://bbc.co.uk'
Connecting to 151.101.0.81:80
Redirected to / on bbc.co.uk
Redirected to / on www.bbc.co.uk
Writing to '/dev/null'
/dev/null            100% |*******************************|   574k  0:00:00 ETA
Download completed (588050 bytes)
```

There appears to be an issue with the port set utilisation. If I make multiple requests to the same URL, eventually it stops being able to reach it.

```
root@OpenWrt:~# wget -O /dev/null -4 http://google.com
Downloading 'http://google.com'
Connecting to 216.58.201.110:80
Redirected to / on www.google.com
Writing to '/dev/null'

Download completed (18103 bytes)

root@OpenWrt:~# wget -O /dev/null -4 http://google.com
Downloading 'http://google.com'
Failed to send request: Operation not permitted
```

Looking at the nf_conntrack table I can see only the first block of ports is utilised. Once these entries time out, the connections will work again

```
root@OpenWrt:~# cat /proc/net/nf_conntrack | grep 216.58.201.110  | awk '{print $16}'
dport=1309
dport=1311
dport=1298
dport=1305
dport=1303
dport=1310
dport=1297
dport=1306
dport=1301
dport=1308
dport=1299
dport=1296
dport=1300
dport=1307
dport=1304
dport=1302
```

From the output above you can see these are destination ports from the first block of ports. There are none for any of the other ports.

RULE_1_PORTSETS='1296-1311 2320-2335 ....'

I've replicated this on a Fritzbox 7530 running 23.05.0, and a raspberry pi b 3+ also running 23.05.0.

I've got the map package version 7 installed.

### OpenWrt version

r23497-6637af95aa

### OpenWrt release

23.05.0

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530

### Image kind

Official downloaded image

### Steps to reproduce

Request a URL from a single IP repeatedly.

### Actual behaviour

Eventually the website stops responding when the first port block allocation is filled.

### Expected behaviour

Either ports are randomly used from the entire port block allocation, or once the first set of ports is exhausted the next block of ports is used.

### Diffconfig

```text
N/A
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

davesbell, 2024-01-21:
Just to add, I've confirmed that this behaviour exists for hosts behind the router, as well as for the router itself.

rany2, 2024-01-21:
I'm not very familiar with this technology but I think the `nft list ruleset` output will be helpful in this case.

brada4, 2024-01-21:
From 23.05.2 preferably

jow-, 2024-01-21:
Likely dupe of #11972 - would appreciate if someone could extract a proper fix from its various referenced scripts and hacks

brada4, 2024-01-21:
It isimpotant to know that mark/connmark are not partially submerged under xt nft compat  bridge.

davesbell, 2024-01-21:
As requested.

```
root@OpenWrt:~# nft list ruleset
table inet fw4 {
	chain input {
		type filter hook input priority filter; policy drop;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		iifname { "lan1", "192.0.2.0-wan6_4" } jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
		jump handle_reject
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		iifname { "lan1", "192.0.2.0-wan6_4" } jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
		jump handle_reject
	}

	chain output {
		type filter hook output priority filter; policy accept;
		oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
		ct state established,related accept comment "!fw4: Allow outbound established and related flows"
		oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
		oifname { "lan1", "192.0.2.0-wan6_4" } jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
	}

	chain prerouting {
		type filter hook prerouting priority filter; policy accept;
		iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
	}

	chain handle_reject {
		meta l4proto tcp reject with tcp reset comment "!fw4: Reject TCP traffic"
		reject comment "!fw4: Reject any other traffic"
	}

	chain syn_flood {
		limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
		drop comment "!fw4: Drop excess packets"
	}

	chain input_lan {
		jump accept_from_lan
	}

	chain output_lan {
		jump accept_to_lan
	}

	chain forward_lan {
		jump accept_to_wan comment "!fw4: Accept lan to wan forwarding"
		jump accept_to_lan
	}

	chain helper_lan {
	}

	chain accept_from_lan {
		iifname "br-lan" counter packets 165 bytes 12131 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

	chain accept_to_lan {
		oifname "br-lan" counter packets 40 bytes 4440 accept comment "!fw4: accept lan IPv4/IPv6 traffic"
	}

[... 188 lines cut ...]
	}

	chain mangle_postrouting {
		type filter hook postrouting priority mangle; policy accept;
	}

	chain mangle_input {
		type filter hook input priority mangle; policy accept;
	}

	chain mangle_output {
		type route hook output priority mangle; policy accept;
	}

	chain mangle_forward {
		type filter hook forward priority mangle; policy accept;
		iifname { "lan1", "192.0.2.0-wan6_4" } tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 ingress MTU fixing"
		oifname { "lan1", "192.0.2.0-wan6_4" } tcp flags syn tcp option maxseg size set rt mtu comment "!fw4: Zone wan IPv4/IPv6 egress MTU fixing"
	}
}
```

Sorry, bit of an openwrt newbie :-) I'm unsure how to update if there is no official package listed.

@jow- I did have a good hunt through the bug list, and did come across that one. I couldn't figure out what was going on though.

rany2, 2024-01-21:
@davesbell The issue is more obvious now with your ruleset, thanks

rany2, 2024-01-21:
@jow- The issue I think is that `connlimit_ports` is not implemented in firewall4.

jow-, 2024-01-21:
So the idea of the original rules is to only allow a certain maximum amount of flows for a given port range, then switch to the next range? So no fixed mapping but random distribution? If so, it can probably remodeled using the nft numgen facility (https://serverfault.com/a/972166)

brada4, 2024-01-21:
That would be neat addition.

brada4, 2024-01-21:
Or just rule with all ranges together?

rany2, 2024-01-21:
> So the idea of the original rules is to only allow a certain maximum amount of flows for a given port range, then switch to the next range? So no fixed mapping but random distribution? If so, it can probably remodeled using the nft numgen facility (https://serverfault.com/a/972166)

One of the modifications in #11972 actually made use of numgen, though they were calling nft directly instead of using firewall4.

jow-, 2024-01-21:
Ah, good. So it needs to be factored out into a clean *.nft script which we can dynamically include on demand from the MAP-T handler.

brada4, 2024-01-21:
Any portspec is multiport in nftables
```
iptables-translate -A FLOP -p tcp -m multiport --dports 1,2:3,4:10
nft add rule ip filter FLOP ip protocol tcp tcp dport { 1,2-3,4-10} counter
```

jow-, 2024-01-21:
Hm, so we won't need any explicit distribution logic at all? That would simplify the solution a lot. Only problem is that fw4 does not understand multiple ranges in a single rule but the MAP-T handler could write the appropriate rule on-the-fly into some /var/run/whatever.nft and register it dynamically via ubus.

rany2, 2024-01-21:
> Ah, good. So it needs to be factored out into a clean *.nft script which we can dynamically include on demand from the MAP-T handler. Could you link me the specific comment/script?

There you go: https://github.com/fakemanhk/openwrt-jp-ipoe/blob/main/map.sh.new

brada4, 2024-01-21:
Looks like it to me

jow-, 2024-01-21:
Thanks @rany2 - so this script is halfway there, but it needs some improvements. Instead of manually calling the nft cli multiple times it should generate a nft script and load it as include. I'm also not entirely happy with the additional nft table it spawns but the numgen based approach appears sound.

brada4, 2024-01-21:
Indeed stuffing port range needs fw3 changed.

rany2, 2024-01-21:
@jow- So you mean, generating a nft file in `/usr/share/nftables.d/ruleset-post/` and reloading the firewall?

jow-, 2024-01-21:
@rany2 - almost, the nft ruleset file should be generated in /tmp or /var and the config include section added via ubus, like how the rules for fw3 are added via those json_* calls.

jow-, 2024-01-22:
@davesbell - would it be possible for you to produce a .pcap file containing the DHCP offer you receive? I'd like to setup a custom DHCPv6 server here which announces MAP-T rules to simplify testing the proto handler and resulting firewall rulesets.

davesbell, 2024-01-22:
@jow- I certainly can, or alternatively you can use this tool to generate the offer for you: https://map46.cisco.com/

This should produce the options in my initial post.
```
# For ISC DHCPD dhcpd6.conf (ensure the lines are copied in full!)
option dhcp6.map-option code 95 = string;
option dhcp6.map-option 00:59:00:15:00:0a:1c:c0:00:02:00:26:20:01:0d:b8:40:00:5d:00:04:06:00:00:00:00:5b:00:09:40:2a:09:80:41:00:00:00:00;
```

If you still want a PCAP, let me know and I will capture it later.

ldir-EDB0, 2025-05-26:
May I offer this (completely untested in a proper environment) lump of code as an attempt to crawl closer to the finish line.  I haven't yet worked out how to do the json 'add this nft file to the firewall' thing.  It's totally based on fakemanhk's modifications.  It generates mapet.nft as something that can be passed to nft.
`RULE_0_IPV4ADDR="2.212.4.22"
RULE_0_PORTSETS="0-10 1025-1036 2025-2036 3025-3036"
k=0
cfg=kdb

DONT_SNAT_TO="0"
foo(){
            local portcount=0                                                    
            local allports=""                                                    
            for portset in $(eval "echo \$RULE_${k}_PORTSETS"); do               
                local startport=${portset%-*}                                    
                local endport=${portset#*-}                                      
                                                                                 
                for x in $(seq "$startport" "$endport"); do                      
                    case " $DONT_SNAT_TO " in                                    
                        *" $x "*)                                                
                            # Skip this port                                     
                            ;;                                                   
                        *)                                                       
                            allports="${allports}${portcount} : ${x}, "         
                            portcount=$((portcount + 1))
                            ;;                                                   
                    esac                                                         
                done                                                             
            done                                                                 
            allports=${allports%??}

cat << EOF >/tmp/mapet.nft
#!/usr/sbin/nft -f

table inet mapet
delete table inet mapet

table inet mapet {
	chain srcnat {
		type nat hook postrouting priority 0; policy accept;
		ip protocol {tcp, udp, icmp} oifname map-$cfg snat ip to $(eval echo \$RULE_${k}_IPV4ADDR) : numgen inc mod $portcount map { $allports }
	}
}
EOF

}

foo
`