openwrt/openwrt #14451: Configuration abnormality when adding client interface to wifi

zhangsd20, 2024-01-22:
### Describe the bug

When my motherboard supports two wireless network cards, I only enable one of the network cards, set it to ap/mesh mode, and add an interface to client mode. It was found that the network card was working abnormally and ap/mesh could not work.

Below is my wireless configuration
wireless:
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/a000000.wifi'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/a800000.wifi'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option disabled '1'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'sta'
        option ssid 'OpenWrt12'
        option encryption 'none'
        option wds '1'
        option macaddr 'random'

This is the iwinfo output information after the configuration is completed. At this time, the Tx power and HT mode are incorrect.
iwinfo:
phy0-ap0  ESSID: "OpenWrt"
          Access Point: DC:EB:53:13:C8:5D
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -102 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0-sta0 ESSID: unknown
          Access Point: B6:9E:8F:34:D7:96
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -102 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

According to the above configuration, I only need to enable another network card for this network card to work normally.

wireless:
   config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/soc/a000000.wifi'
        option channel '1'
        option band '2g'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/a800000.wifi'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'
        option disabled '0'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'sta'
        option ssid 'OpenWrt12'
        option encryption 'none'
        option wds '1'
        option macaddr 'random'

iwinfo:
phy0-ap0  ESSID: "OpenWrt"
          Access Point: DC:EB:53:13:C8:5D
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 30 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -103 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy0-sta0 ESSID: unknown
          Access Point: 7E:03:8B:C4:57:CA
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -103 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "OpenWrt"
          Access Point: DC:EB:53:13:C8:5C
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

At this time, the information seen by iwinfo is correct.
In turn, I also tested the situation with another network card. It also requires both network cards to be turned on at the same time to work properly. I believe this problem also exists on other motherboards. How to solve this problem?

### OpenWrt version

r23314-7efec0acca

### OpenWrt release

23.05-SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Qxwlan E2600AC c1

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

zhangsd20, 2024-01-25:
No, you're not right. First of all, there will be a default country for no configured country, and I worked fine with 5G before I added the "client" interface. It is because I added the "client" interface that the 5G did not work。

Below is my wireless configuration
config wifi-device 'radio0'
option type 'mac80211'
option path 'platform/soc/a000000.wifi'
option channel '1'
option band '2g'
option htmode 'HT20'
option cell_density '0'

config wifi-iface 'default_radio0'
option device 'radio0'
option network 'lan'
option mode 'ap'
option wds '1'
option ssid 'OpenWrt'
option encryption 'none'

config wifi-device 'radio1'
option type 'mac80211'
option path 'platform/soc/a800000.wifi'
option channel '36'
option band '5g'
option htmode 'VHT80'
option cell_density '0'

config wifi-iface 'default_radio1'
option device 'radio1'
option network 'lan'
option mode 'ap'
option ssid 'OpenWrt'
option encryption 'none'

iwinfo:
wlan0     ESSID: "OpenWrt"
          Access Point: DC:EB:53:13:C8:5D
          Mode: Master  Channel: 1 (2.412 GHz)  HT Mode: HT20
          Center Channel 1: 1 2: unknown
          Tx-Power: 26 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -102 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:003C 168C:4019 [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "OpenWrt15"
          Access Point: DC:EB:53:13:C8:5C
          Mode: Master  Channel: 36 (5.180 GHz)  HT Mode: VHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 27 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -94 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 168C:003C 168C:4019 [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

brada4, 2024-01-25:
You need to set AP channel to auto to let it match STA channel that is implied once you are connected to it.
No need to throw more unformatted conf files. Please post logread -e hostapd from the moment you start AP on top of STA.

zhangsd20, 2024-01-26:
According to your description, setting chan to auto is the same problem.
If you have any device that supports openwrt23.05.2 and you can try configuration testing, you should have the same problem

config wifi-device 'radio1'    
        option type 'mac80211'    
        option path 'platform/soc/a800000.wifi'
        option channel 'auto'                  
        option band '5g'                       
        option htmode 'VHT80'                  
        option cell_density '0' 
                                
config wifi-iface 'default_radio1'
        option device 'radio1'    
        option network 'lan'      
        option mode 'ap'          
        option ssid 'OpenWrt'   
        option encryption 'none'
                                
config wifi-iface 'wifinet3'    
        option device 'radio1'  
        option mode 'sta'     
        option ssid 'OpenWrt123'
        option encryption 'none'

root@OpenWrt:/# iwinfo 
phy1-ap0  ESSID: "OpenWrt"
          Access Point: DC:EB:53:13:C8:5C
          Mode: Master  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -93 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

phy1-sta0 ESSID: unknown
          Access Point: DE:EB:53:13:C8:5C
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -93 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

brada4, 2024-01-26:
Please take care of formatting your code blocks with three backticks.
I see you removed sta from any networks now?
Since your tx power is above legal limits it is not expected to work.

zhangsd20, 2024-01-26:
Please look at the configuration I sent, I did not delete the "sta", only adding the "client" interface will cause problems, if delete the "client" interface ap can work.

I did not configure the power, it is the default power, so it is not possible that the power you said is too large

zhangsd20, 2024-01-26:
I suggest you test it for yourself. You'll see the problem very clearly

brada4, 2024-01-26:
I have ramips wds working (RE200 extender)

zhangsd20, 2024-01-26:
See what happens with just one device!

Your test method is AP+extender, with an ap interface and a sta interface on the extender, so that after the extender is connected to the AP, the extender works properly.

But, have you tried what happens when the extender doesn't have an AP attached?

I want to configure the ap interface and sta interface for only one device, which can be configured in previous openwrt versions.

zhangsd20, 2024-01-26:
Testing test1 as follows will not work, test2 works fine.
The first device cannot work properly if sta interfaces are added.

test1:
device1( sta+ap) <---->device2(sta+ap)

test2:
device1( ap) <---->device2(sta+ap)

brada4, 2024-01-26:
Particular wifi chip supports just one 4-address WDS AP or STA, you can try mesh network, but in principle for 2 devices just one WDS link is needed.

zhangsd20, 2024-01-27:
In case you didn't get what I meant, let me restate it
test1:
device1( sta<ssid: 123> +ap <ssid: 124>) <---->device2(sta<ssid: 124>+ap<ssid: 125>)
in principle for 2 devices just one WDS link is needed.

Each device is configured with one ap interface and one sta interface, but the ssid is different. This problem was not present in older versions of openwrt

brada4, 2024-01-27:
No wds there, just wwan. Remove all wds settings and there you go.

zhangsd20, 2024-01-27:
No, it's still not working. It is recommended that you test to open only one device or configure another ssid

zhangsd20, 2024-01-27:
You never understood what I meant when I said from the beginning that only one device just added a sta interface and it wouldn't work.

You can turn off your main AP and quota the RE200 extender to ap+sta to see if it works. Or you can add a sta interface directly to your main AP.

brada4, 2024-01-27:
Yes. the extender works as ap+sta and i can restart ao without ruining sta

zhangsd20, 2024-01-30:
Did you just open the extender? And it's not connected to any ap?

brada4, 2024-01-30:
Only this, worked right away.
https://openwrt.org/docs/guide-user/network/wifi/atheroswds

zhangsd20, 2024-01-31:
Please show me your configuration and "iwinfo" information.

If I change the 2.4 g equipment, and use (https://openwrt.org/docs/guide-user/network/wifi/atheroswds), the configuration will not be able to succeed here.

brada4, 2024-01-31:
It is imperative to try before complaining

zhangsd20, 2024-01-31:
Please show me your configuration and "iwinfo" information.

brada4, 2024-01-31:
It is exactly as in the guide. Internet-connected thing is mt76 and the extender is ath79/ath10k/ath9k
Additionally I made extender br-lan dhcp client over wds link so it is flat network more or less.

zhangsd20, 2024-01-31:
I just want to see your configuration or iwinfo information

brada4, 2024-01-31:
This section is supposed to have WDS enabled. Yes, there is additional AP entry in iwinfo from WDS relay connecting.
config wifi-iface 'wifinet3'
option device 'radio1'
option mode 'sta'
option ssid 'OpenWrt123'
option encryption 'none'

zhangsd20, 2024-02-01:
Please show your "iwinfo" information.

zhangsd20, 2024-02-01:
I have tested many devices with the same configuration with the same results, if you can't provide iwinfo information, I still can't believe it

brada4, 2024-02-01:
Im fine with it. I dont have powers to follow simple written guidance in your place.

zhangsd20, 2024-02-01:
So it's not a bug in your opinion?

zhangsd20, 2024-02-01:
I followed the "Wireless Repeater" area exactly, and the end result was the same; And I showed you my configuration and "iwinfo" information, which proves the point. I don't understand why you don't want to show your "iwinfo" information and configuration?

varkey, 2024-09-17:
@zhangsd20 I was seeing a similar issue (followed the same Wireless Repeater guide) and as @brada4 suggested earlier, I added the country code for the 5 GHz radio and the AP started working. 

> AP will not work if you do not set country code on 5GHz radio