openwrt/openwrt #14454: Building libselinux for x64 image using CentOS 9 will fail

Halry, 2024-01-22:
### Describe the bug

Compiler OS: CentOS Stream 9, installed all packages and set up environments following guidelines.
Target : x86_64 (no problem when building arm(filogic, ipq806x) images)

 `
[halry@localhost x64]$ make V=s package/libselinux/compile
make[2]: Entering directory '/home/halry/x64/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/halry/x64/scripts/config'
make[1]: Entering directory '/home/halry/x64'
make[2]: Entering directory '/home/halry/x64/package/libs/toolchain'
Makefile:749: WARNING: skipping libgomp -- package has no install section
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.prepared_7a8501b3aad99800af2af246a4456e96_6664517399ebbbc92a37c5bb081b5c53_check
mkdir -p /home/halry/x64/build_dir/target-x86_64_musl/toolchain
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.prepared_7a8501b3aad99800af2af246a4456e96_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.configured_*
rm -f /home/halry/x64/staging_dir/target-x86_64_musl/stamp/.toolchain_installed
(cd /home/halry/x64/build_dir/target-x86_64_musl/toolchain/./; if [ -x ./configure ]; then find /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r chmod u+w; find /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ -name config.guess | xargs -r -n1 cp --remove-destination /home/halry/x64/scripts/config.guess; find /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r chmod u+w; find /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ -name config.sub | xargs -r -n1 cp --remove-destination /home/halry/x64/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/halry/x64/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD="x86_64-openwrt-linux-musl-ld.bfd" NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/halry/x64/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/halry/x64/build_dir/target-x86_64_musl/toolchain=toolchain -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -I/home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify " LDFLAGS="-L/home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -fuse-ld=bfd -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info   ; fi; )
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.built
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.built_check
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libgcc_s.so.1 /home/halry/x64/build_dir/target-x86_64_musl/toolchain/
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.built
rm -rf /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc
mkdir -p /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc
install -d -m0755 /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/usr/bin
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/ld-musl-*.so* /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib/
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libc.so* /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib/
ln -sf ../../lib/libc.so /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/usr/bin/ldd
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libatomic.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libc.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libcrypt.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libdl.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libitm.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libm.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libpthread.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libquadmath.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libresolv.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/librt.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libssp_nonshared.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libstdc++.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libstdc++fs.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libsupc++.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libutil.a /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libxnet.a /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib/
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/libgcc_pic.a /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib/libgcc_s_pic.a; cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/libgcc.map /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/lib/libgcc_s_pic.map
touch /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc.installed
mkdir -p /home/halry/x64/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/halry/x64/tmp/.root-copy.flock -c 'cp -fpR /home/halry/x64/build_dir/target-x86_64_musl/toolchain/.pkgdir/libc/. /home/halry/x64/staging_dir/target-x86_64_musl/root-x86/'
touch /home/halry/x64/staging_dir/target-x86_64_musl/root-x86/stamp/.libc_installed
mkdir -p /home/halry/x64/bin/targets/x86/64/packages /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL /home/halry/x64/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib
cp -fpR /home/halry/x64/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/libgcc_s.so.* /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/
find /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/halry/x64/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/halry/x64/scripts/strip-kmod.sh" PATCHELF="/home/halry/x64/staging_dir/host/bin/patchelf" /home/halry/x64/scripts/rstrip.sh /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc
rstrip.sh: /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/lib/libgcc_s.so.1: shared object
(cd /home/halry/x64/build_dir/target-x86_64_musl/toolchain/ipkg-x86_64/libgcc/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions [... cut ...]
### Steps to reproduce

1: clone openwrt to local disk
2: select x86_64 target
3: enable selinux and select libselinux
4: compile image "make V=s"

### Actual behaviour

build will fail.

### Expected behaviour

Able to build successfully.

### Additional info

You can build libselinux and image successfully by running ` export LD_LIBRARY_PATH= ` before building. It seems CentOS environments interfere the compiling process.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-22:
Could you , please, re-format code block with triple backticks?

Halry, 2024-01-22:
> Could you , please, re-format code block with triple backticks?

Sorry, just forgot how to use markdown, right now it should be okay.

brada4, 2024-01-22:
Please check if /usr/lib .... libfts.so.* does not get in the way (and lib names in musl-fts directory)

Halry, 2024-01-22:
> Please check if /usr/lib .... libfts.so.* does not get in the way (and lib names in musl-fts directory)

No, there isn't any file named include *libfts* and that "musl-fts" directory.

```
[halry@localhost usr]$ find /usr -iname *libfts*
find: ‘/usr/share/polkit-1/rules.d’: Permission denied
find: ‘/usr/share/empty.sshd’: Permission denied
find: ‘/usr/libexec/initscripts/legacy-actions/auditd’: Permission denied
```

Halry, 2024-01-22:
My LD_LIBRARY_PATH environment setting seems fine

```
[halry@localhost usr]$ $LD_LIBRARY_PATH
-bash: /usr/local/lib:/usr/local/bin/python2.7:/usr/local/lib:/usr/local/bin/python2.7:: No such file or directory
```

Halry, 2024-01-27:
Okay, I see, thanks.

closed 2024-01-27 by Halry