openwrt/openwrt #14458: hostapd: wap3 client can authenticate with random password

roshii, 2024-01-22:
### Describe the bug

After an initial successful authentication to an OpenWrt AP, a WAP3 client can reconnect to the AP with any random password provided MAC address remains the same. A malicious network observer could just read MAC and use it to gain access to network bypassing authentication.

### OpenWrt version

r23630-842932a63d

### OpenWrt release

23.05.2

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C6 v2 (EU/RU/JP)

### Image kind

Official downloaded image

### Steps to reproduce

1. Setup an AP with `/etc/config/wireless` as follows.
```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'pci0000:00/0000:00:00.0'
	option band '5g'
	option country 'FR'
	option cell_density '0'
	option htmode 'VHT40'
	option channel '165'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/ahb/18100000.wmac'
	option channel '1'
	option band '2g'
	option country 'FR'
	option cell_density '0'
	option ht_coex '1'
	option htmode 'HT20'
	option txpower '14'

config wifi-iface 'rocinante1'
	option device 'radio1'
	option mode 'ap'
	option ssid 'rocinante'
	option encryption 'sae'
	option multicast_to_unicast_all '1'
	option key 'secret127'
	option network 'home'
```

2. Connect with a client using valid sae password
3. Disconnect client
4. Connect to AP with any random password ensuring MAC address remains the same

### Actual behaviour

The client can connect using an invalid password with a MAC address that was authenticated once.

A logread on the AP show the following.
When connecting with valid password:
```
Mon Jan 22 10:52:09 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 IEEE 802.11: associated (aid 1)
Mon Jan 22 10:52:09 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 92:64:f7:49:a5:16 auth_alg=sae
Mon Jan 22 10:52:09 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 RADIUS: starting accounting session EC4B4EA41D28B67C
Mon Jan 22 10:52:09 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 WPA: pairwise key handshake completed (RSN)
Mon Jan 22 10:52:09 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 92:64:f7:49:a5:16
```

Disconnecting:
```
Mon Jan 22 10:52:17 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 92:64:f7:49:a5:16
Mon Jan 22 10:52:17 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 IEEE 802.11: disassociated
Mon Jan 22 10:52:18 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

Reconnecting with a wrong password
```
Mon Jan 22 10:52:30 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 IEEE 802.11: authenticated
Mon Jan 22 10:52:30 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 IEEE 802.11: associated (aid 1)
Mon Jan 22 10:52:30 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 92:64:f7:49:a5:16 auth_alg=open
Mon Jan 22 10:52:30 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 RADIUS: starting accounting session 045919528FDA63F5
Mon Jan 22 10:52:30 2024 daemon.info hostapd: phy1-ap0: STA 92:64:f7:49:a5:16 WPA: pairwise key handshake completed (RSN)
Mon Jan 22 10:52:30 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 92:64:f7:49:a5:16
```

### Expected behaviour

The client should not be able to connect using an invalid password.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2024-01-22:
Can you `logread -e prune` (it should show faileed auth sessions)

roshii, 2024-01-22:
> Can you `logread -e prune` (it should show faileed auth sessions)

It produces no output at all.

brada4, 2024-01-22:
Does it "fix" to replace wpad-basic-mbedtls with wpad-mbedtls (or hostapd-mbedtls if running out of space) i.e compile configuration problems.
PS I see auth_alg=open where it should be only eap folowed by good auth ("good" like in your report, but I cannot re-trigger)

roshii, 2024-01-23:
I am running an image with hostpad-mbedtls built through attended-sysupgrade and OpenWrt servers.
I did consider it as an official image when writing this issue, please correct me if wrong.

nbd168, 2024-02-09:
@roshii so how did you arrive at the conclusion that it has something to do with the MAC address? Did you try this with a different client (one that hasn't seen the valid password) by changing the MAC address? Or did you just try to connect again with the same client, but a different password?
Did you rule out the possibility that the client may simply have cached the credentials and used them with a WPA2 handshake?

roshii, 2024-02-16:
> @roshii so how did you arrive at the conclusion that it has something to do with the MAC address? Did you try this with a different client (one that hasn't seen the valid password) by changing the [...]

I tried with the same client, running Android 14 and which and the option to randomize (or not) MAC address. If MAC address is kept constant, client may connect with a wrong password after a successful authentication. On the other hand, if MAC address is randomized on each connection (including after changing password), the trick does not work and client can only authenticate with a valid password. Rebooting AP also lead to failed re-authentication with invalid password regardless of the conditions.
Based on that I assumed AP is maintaining some cache based on MAC address. I will try to connect with a different device spoofing MAC address to make sure client is not caching connection based on his own randomized MAC address. Reporting back as soon as done.

brada4, 2024-02-16:
Conf is pure sae.

rany2, 2024-02-16:
After giving this a second try, I could reproduce this when `auth_cache` is set to 1 however while it did "authenticate" I wasn't able to pass any traffic so I don't think it matters anyway.

roshii, 2024-02-25:
TLDR: Android client is caching connection authentication. Nothing to do with OpenWrt.

Following some test, I was not able to reproduce and authenticate with an invalid password with a Linux/nmcli client. 
Furthermore, using the same MAC address on both clients, I was not able to gain network access with an invalid password with any clients after successful authentication by the other client. 
To finish, forgetting connection on Android client does not allow to reconnect with wrong password.