openwrt/openwrt #14462: MF287 (ipq40xx) boot loop when no SIM card is present

mb555444, 2024-01-22:
### Describe the bug

I updated to the lastest development snapshot and reset all settings.
As a result, the router went into a boot loop. It was working fine with my custom settings.

I could SSH to 192.168.1.1 for 1-2 seconds before it reboots, so I was able to repeatedly do this, each time running some commands to diagnose the issue.

I ran `ps` and noticed that the last activity usually was setting up the LTE connection:
```
...
  398 root         0 IW<  [ath10k_aux_wq]
  399 root         0 IW<  [ath10k_tx_compl]
  488 root         0 IW<  [ath10k_wq]
  489 root         0 IW<  [ath10k_aux_wq]
  490 root         0 IW<  [ath10k_tx_compl]
  711 logd      1460 S    /sbin/logd -S 64
  917 root       856 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p
 1021 root      2064 S    {hostapd} /sbin/ujail -t 5 -n hostapd -U network -G
 1022 root      2064 S    {wpa_supplicant} /sbin/ujail -t 5 -n wpa_supplicant
 1067 network   2820 S    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 1068 network   2780 S    /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_suppl
 1086 root      1604 S    /sbin/netifd
 1197 root      1112 S    /usr/sbin/odhcpd
 1422 root      1320 S    {qmi.sh} /bin/sh ./qmi.sh qmi setup wan {"device":"/
 1455 root      2064 S    {ntpd} /sbin/ujail -t 5 -n ntpd -U ntp -G ntp -C /et
 1497 ntp       1096 S    /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.
 1754 root      2064 S    {dnsmasq} /sbin/ujail -t 5 -n dnsmasq -u -l -r /bin/
 1784 dnsmasq   1196 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c
 1831 root      1468 R    {dnsmasq} /bin/sh /etc/rc.common /etc/init.d/dnsmasq
 1890 root       876 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p
 1909 root       920 S    uqmi -s -d /dev/cdc-wdm0 -t 1000 --uim-get-sim-state
 1937 root      1104 R    ps
```
I also saw in the system log:
```
Mon Jan 22 11:32:55 2024 daemon.notice netifd: wan (1406): Waiting for SIM initialization
Mon Jan 22 11:32:55 2024 daemon.info procd: - init complete -
Mon Jan 22 11:32:55 2024 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Mon Jan 22 11:32:55 2024 daemon.info urandom_seed[1511]: Seed saved (/etc/urandom.seed)
Mon Jan 22 11:32:56 2024 daemon.notice netifd: wan (1406): SIM in illegal state - Power-cycling SIM
```
I normally use this router as an access point and gigabit switch, the LTE modem is unused. Therefore, no SIM card is present.
I inserted an old, inactive SIM card. After doing this, the boot loop no longer occurred.

The reason why this occurred after the factory reset seems to be that in my typical config, the WAN interface is deleted, likely LTE initialization is therefore skipped.

### OpenWrt version

r24884-fd46680966

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZTE MF287

### Image kind

Official downloaded image

### Steps to reproduce

0. remove SIM card
1. install latest snapshot
2. reset to factory defaults
3. reboot router
4. observe the reboot loop

### Actual behaviour

OpenWRT tries to initialise the WAN interface. When no SIM card is present, for some reason this causes a reboot.

### Expected behaviour

Missing SIM card should not cause the router to reboot

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mb555444, 2024-01-23:
FYI @andyboeh

andyboeh, 2024-01-23:
Currently, I don't have the device at hand, but it could be related to this commit: https://github.com/openwrt/openwrt/commit/696c9f723d3b2fe1577da849bab97a93b9ad5d68

It's actually easy to test, what happens if you run
```
uqmi -d /dev/cdc-wdm0 --uim-power-off --uim-slot 1
# --- wait a few seconds ---
uqmi -d /dev/cdc-wdm0 --uim-power-on --uim-slot 1
```
with a SIM inserted and without a SIM? You might be able to remove WAN in recovery mode.

It is possible that the modem reacts with a modem reset instead, which also causes a device reset! I did not find a GPIO to block the reset signal (as on the MF287Pro and MF282Plus and similar devices).

How do the top LEDs (the ring) behave when the reboot happens?

mb555444, 2024-01-23:
Thanks for the quick response.

None of the following commands causes a restart with or without SIM:
```
uqmi -d /dev/cdc-wdm0 --uim-power-off --uim-slot 1
uqmi -d /dev/cdc-wdm0 --uim-power-on --uim-slot 1
uqmi -s -d /dev/cdc-wdm0 -t 1000 --uim-get-sim-state
```

The following causes a restart without SIM but not with SIM:
```
a) /etc/init.d/network restart
b) /etc/init.d/network stop ; sleep 30; /etc/init.d/network start
```
In the above cases, the top of the device shows a red light (all 4 segments on, then 3, 2, 1, 0), then (I assume) the reboot happens. This is the same light sequence that also can be observed during the restart loop reported earlier.
In case b, the restart happens later, so it seems to be bringing the network up without SIM that is the trigger.

The following causes a restart always:
```
echo 1 > /sys/class/gpio/modem-reset/value
```

Unfortunately, I am not familiar with LTE modem commands (never used them), so I could not do any further testing as to what exactly might happen during "/etc/init.d/network start".

mb555444, 2024-01-23:
Correction to the above. The following command causes the router to reboot when no SIM card is present:
```
uqmi -d /dev/cdc-wdm0 --uim-power-off --uim-slot 1
```

Leo-PL, 2024-01-28:
This reminds me to check current builds on MF286* series too. Perhaps the restart GPIO shall be dropped, or the power button blocker engaged.

andyboeh, 2024-01-28:
@blocktrron Since this is probably related to your commit, could you comment on this please?

mb555444, 2024-01-29:
@andyboeh 
When no SIM is inserted, I get the following response:
```
# uqmi -s -d /dev/cdc-wdm0 -t 1000 --uim-get-sim-state
{}
```
When a SIM is present or in the first few seconds after removing it, the response is:
```
# uqmi -s -d /dev/cdc-wdm0 -t 1000 --uim-get-sim-state
{"card_application_state":"puk-required","pin1_status":"blocked","pin1_verify_tries":0,"pin1_unlock_tries":10,"pin2_status":"not_verified","pin2_verify_tries":3,"pin2_unlock_tries":10}
```

Hasi123, 2025-09-14:
I can also confirm this issue. I tried to initially install OpenWRT on my MF287 (without a SIM) and was quite frustrated, because it always rebooted after a minute. The reboot also happens with the initramfs firmware. After some tinkering I found out, that 23.05 worked fine and am currently using it.