openwrt/openwrt #14463: ubox (2024-01-15) update introduces a lot of noisy kmodloader logs

DragonBluep, 2024-01-23:
### Describe the bug

A lot of noisy logs, but all kernel modules are working properly. Reverting https://github.com/openwrt/openwrt/commit/ced3fbcda18e200315cb077f7e2b362ae52ee065 can silence them.

Removing `kmod-ath9k` and its dependence can reduce the amount of these logs.

cc @Ansuel

```
[   17.853233] do_page_fault(): sending SIGSEGV to modprobe for invalid read access from 77eee000
[   17.862012] epc = 0040218d in kmodloader[400000+3000]
[   17.867148] ra  = 00402239 in kmodloader[400000+3000]
```
<details>
<summary>Full dmesg</summary>

```
[    0.000000] Linux version 5.15.147 (runner@fv-az1498-458) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r0-e15a69b) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Tue Jan 23 01:20:16 2024
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR2543N/ND
[    0.000000] SoC: Atheros AR7242 rev 1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=0002ffe0
[    0.000000] Readback ErrCtl register=0002ffe0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55924K/65536K available (5667K kernel code, 585K rwdata, 1244K rodata, 1200K init, 206K bss, 9612K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 400.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 9556302233 ns
[    0.000002] sched_clock: 32 bits at 200MHz, resolution 5ns, wraps every 10737418237ns
[    0.008086] Calibrating delay loop... 265.42 BogoMIPS (lpj=1327104)
[    0.094378] pid_max: default: 32768 minimum: 301
[    0.100648] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.108028] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.126035] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.135989] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.143201] pinctrl core: initialized pinctrl subsystem
[    0.151212] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.183709] clocksource: Switched to clocksource MIPS
[    0.191044] NET: Registered PF_INET protocol family
[    0.196451] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.204905] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.213478] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.221353] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.229110] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236252] TCP: Hash tables configured (established 1024 bind 1024)
[    0.242899] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.249585] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.257273] NET: Registered PF_UNIX/PF_LOCAL protocol family
[    0.263045] PCI: CLS 0 bytes, default 32
[    0.272316] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.288224] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.294190] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.308266] pinctrl-single 18040028.pinmux: 64 pins, size 8
[... 169 lines cut ...]
</details>

### OpenWrt version

r0-65fb97b

### OpenWrt release

SNAPSHOT

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR2543N/ND

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2024-01-23:
Can you retry with https://github.com/openwrt/openwrt/commit/4d2a82a73c79aea41f14feaaeea319a965041f63 ?

DragonBluep, 2024-01-23:
> Can you retry with [4d2a82a](https://github.com/openwrt/openwrt/commit/4d2a82a73c79aea41f14feaaeea319a965041f63) ?

Hi! It's not helpful, this report is based on today's master branch https://github.com/openwrt/openwrt/commit/65fb97b4507bcff3f98739975e9cc7df53137b12.
```
root@OpenWrt:~# opkg list |grep ubox
libubox202312041 - 2023-12-04.1-ca3f6d0c-1
ubox - 2024-01-22-6cf7d837-1
```

guidosarducci, 2024-01-23:
> this report is based on today's master branch 65fb97b<https://github.com/openwrt/openwrt/commit/65fb97b4507bcff3f98739975e9cc7df53137b12>.
>
> Sorry, I've missed this part, thanks for double checking!
>
> Reverting ced3fbc<https://github.com/openwrt/openwrt/commit/ced3fbcda18e200315cb077f7e2b362ae52ee065> can silence them.
>
> Ok, those changes were introduced by @guidosarducci<https://github.com/guidosarducci> in openwrt/ubox#1<https://github.com/openwrt/ubox/pull/1>
>
> BTW I've noticed as well the same issue<https://ynezz.gitlab.io/-/openwrt-device-runtime-testing/-/jobs/5988706415/artifacts/dmesg-zyxel-gs1900-initramfs.txt> on ZyXEL GS1900-8, running r24887-65fb97b450 as well.
>
> [   19.103873] do_page_fault(): sending SIGSEGV to modprobe for invalid read access from 77ee6000
> [   19.103940] epc = 0040218d in kmodloader[400000+3000]
> [   19.104013] ra  = 00402239 in kmodloader[400000+3000]

I doubt it's just noise. Might be an old bug getting exposed with recent changes, although my build of this master ran without errors.

I did find a potential SIGSEGV yesterday and about to PR. Try cherry-picking https://github.com/guidosarducci/ubox/commit/9fb3c1bb19397d3b69263981643d14ec22703b4f .

DragonBluep, 2024-01-24:
@guidosarducci Thanks! I can confirm this patch works.

closed 2024-01-24 by openwrt-bot